genodelabs/genode #3329: Image creation tools missing for sel4 and okl4 in empty build directory

skalk, 2019-05-07:
When invoking a run-script that only uses depot packages within an empty run-script it may happen that the okl4 specific `elfweaver` or the sel4 specific `gen_boot_image` tools are missing in the build directory. Until now, they were build as a side-effect of building either the kernel or syscall library. Now, that we have a lot more depot package run-scripts only, this phenomena regularily happens in our nightly tests. To first circumvent the problem, we shoul expliucitly build the tools when the are missing.

skalk, 2019-05-07:
@chelmuth if you are ok with it, please apply those two commits 2039e18 and 573596d to staging

skalk, 2019-05-07:
@chelmuth I rebased to hit current staging, so commit ids are ec6faa2 and 27e29c3

nfeske, 2019-05-08:
This looks like a duplicate of #2918. Thanks @skalk for addressing this long-standing glitch.

commit 5621d5d0 (2019-05-08): sel4: build elfloader on demand (fix #3329)
closed 2019-05-29 by chelmuth