genodelabs/genode #3332: intel_gpu_drv: compile error with GCC 8.3.0

cproc, 2019-05-07:
When building 'make drivers/gpu' with GCC 8.3.0, the following error occurs:

```
    COMPILE  main.o
In file included from /.../repos/os/src/drivers/gpu/intel/main.cc:37:
/.../repos/os/src/drivers/gpu/intel/mmio.h: In member function ‘void Igd::Mmio::update_context_status_pointer()’:
/.../repos/os/src/drivers/gpu/intel/mmio.h:1224:48: error: unused variable ‘rp’ [-Werror=unused-variable]
    RCS_RING_CONTEXT_STATUS_PTR::access_t const rp = read<RCS_RING_CONTEXT_STATUS_PTR::Read_pointer>();
                                                ^~
```

I'm not sure if the line can be removed or if the 'read<>()' call is still needed. @cnuke?

cnuke, 2019-05-09:
@cproc to be honest I am not sure if it is needed, it looks superfluous - just remove it.

cproc, 2019-05-14:
Fixed in c3dd082

commit e3743d39 (2019-05-14): intel_gpu_drv: fix compile error with GCC 8.3.0

Fixes #3332
commit 92c314d3 (2019-05-27): intel_gpu_drv: fix compile error with GCC 8.3.0

Fixes #3332
closed 2019-05-29 by chelmuth