genodelabs/genode #3333: base: add copy constructor to 'Genode::Session_label'

cproc, 2019-05-07:
When building 'make init' with GCC 8.3.0, the following errors occur:

```
    COMPILE  child.o
/.../repos/os/src/init/child.cc: In member function ‘virtual Genode::Child_policy::Route Init::Child::resolve_session_request(const Name&, const Genode::Session_label&)’:
/.../repos/os/src/init/child.cc:549:48: error: could not convert ‘target_label’ from ‘const Label’ {aka ‘const Genode::String<160>’} to ‘const Label’ {aka ‘const Genode::Session_label’}
                      target_label, target_diag };
                                                ^
/.../repos/os/src/init/child.cc:564:48: error: could not convert ‘target_label’ from ‘const Label’ {aka ‘const Genode::String<160>’} to ‘const Label’ {aka ‘const Genode::Session_label’}
                      target_label, target_diag };
                                                ^
/.../repos/os/src/init/child.cc:578:48: error: could not convert ‘target_label’ from ‘const Label’ {aka ‘const Genode::String<160>’} to ‘const Label’ {aka ‘const Genode::Session_label’}
                      target_label, target_diag };
                                                ^
```

Apparently, GCC 8 disregards derived copy constructors as candidate.

cproc, 2019-05-07:
The fix works with GCC 8.3.0, but triggers a compile error with the current tool chain:

```
In file included from /.../repos/base/include/session/session.h:18:0,
                 from /.../repos/base/include/cpu_session/cpu_session.h:21,
                 from /.../repos/base/include/vm_session/vm_session.h:21,
                 from /.../repos/os/src/init/child.cc:14:
/.../repos/base/include/base/session_label.h: In member function ‘Genode::Session_label Genode::Session_label::prefix() const’:
/.../repos/base/include/base/session_label.h:69:26: error: no matching function for call to ‘Genode::Session_label::Session_label()’
     return Session_label();
                          ^
/.../repos/base/include/base/session_label.h:34:17: note: candidate: template<class T, class ... TAIL> Genode::Session_label::Session_label(const T&, TAIL&& ...)
   using String::String;
                 ^~~~~~
/.../repos/base/include/base/session_label.h:34:17: note:   template argument deduction/substitution failed:
/.../repos/base/include/base/session_label.h:69:26: note:   candidate expects at least 1 argument, 0 provided
     return Session_label();
                          ^
/.../repos/base/include/base/session_label.h:34:17: note: candidate: Genode::Session_label::Session_label(const char*)
   using String::String;
                 ^~~~~~
/.../repos/base/include/base/session_label.h:34:17: note:   candidate expects 1 argument, 0 provided
/.../repos/base/include/base/session_label.h:34:17: note: candidate: template<unsigned int N> Genode::Session_label::Session_label(const Genode::String<N>&)
/.../repos/base/include/base/session_label.h:34:17: note:   template argument deduction/substitution failed:
/.../repos/base/include/base/session_label.h:69:26: note:   candidate expects 1 argument, 0 provided
     return Session_label();
                          ^
/.../repos/base/include/base/session_label.h:43:3: note: candidate: template<long unsigned int N> Genode::Session_label::Session_label(const Genode::String<OTHER_CAPACITY>&)
   Session_label(Genode::String<N> const &other)
   ^~~~~~~~~~~~~
/.../repos/base/include/base/session_label.h:43:3: note:   template argument deduction/substitution failed:
/.../repos/base/include/base/session_label.h:69:26: note:   candidate expects 1 argument, 0 provided
     return Session_label();
                          ^
/.../repos/base/include/base/session_label.h:24:16: note: candidate: constexpr Genode::Session_label::Session_label(const Genode::Session_label&)
 struct Genode::Session_label : String<160>
                ^~~~~~~~~~~~~
/.../repos/base/include/base/session_label.h:24:16: note:   candidate expects 1 argument, 0 provided
/.../repos/base/include/base/session_label.h:24:16: note: candidate: constexpr Genode::Session_label::Session_label(Genode::Session_label&&)
/.../repos/base/include/base/session_label.h:24:16: note:   candidate expects 1 argument, 0 provided
```

I'm not sure if this is a problem with the code or with the current tool chain.

nfeske, 2019-05-08:
Let's just replace the code along with the tool chain update. The struggle to craft an intermediate version that works with both tool chains - just to be binned once we switched to the new one - is not worth it.

nfeske, 2019-05-08:
BTW, I'm relabeling several of the GCC-update-related issues from "bug" to "cleanup", except for cases where the update revealed an actual bug.

ttcoder, 2019-05-13:
Would it make sense to feed that particular source file (child.cc) to llvm/clang to get a "peer perspective" beside GCC's perspective? I seem to recall there was a ticket mentionning clang in this tracker, not sure.

nfeske, 2019-05-14:
An convenient way would be feeding the creation of the binary archive for a base platform through the static analyzer as described in https://genode.org/documentation/release-notes/18.11#Static_code_analysis. Since the static analyzer is based in LLVM, this would implicitly give us the "peer perspective".

```
tool/depot/static_analyze tool/depot/create <user>/pkg/...
```

commit 35c31abb (2019-05-20): base: add copy constructor to 'Genode::Session_label'

Fixes #3333
closed 2019-05-29 by chelmuth