genodelabs/genode #3349: Planning of Hack'n'Hike 2020

m-stein, 2019-05-13:
The Hack'n'Hike 2019 is over and I had the feeling everybody enjoyed it! The participants came from many different countries - Switzerland, Slovakia, Poland, Netherlands, USA, and Germany - and equally versatile were their hacking projects. While some were working at seamless integration of virtual machines or porting the Hatari emulator, others were dealing with network support on RaspberryPi 3 or ARM 64bit support in base-hw. Further topics were porting Qemu, updating RiscV support, the Exec channel of SSH, and polishing the packaging tools. Besides the hacking, a lively exchange of ideas and experiences was going on and we had two beautiful hikes to the fortress of Königstein and the Gamrig rock near Rathen.

I'd like to encourage the participants to present their personal HnH19 project in the context of this issue or - even better - as article on [genodians.org](http://genodians.org/)! Please don't hesitate to share your story - even if your idea is still work in progress!

Of course, this issue shall serve as forum for planning the next Hack'n'Hike gathering in the year 2020. So, stay tuned and give us feedback if you have suggestions for improvement or if you just like it as it is!

m-stein, 2019-08-12:
We have a date and a location for the Hack'n'Hike 2020!

## When?
__Start:__ 3 pm Friday 8th May 2020
__End:__ 12 am Monday 11th May 2020

## Where?
__Villa Lückendorf__, Kammstraße 34, 02797 Oybin
- [Homepage](http://preview.bluepage.eu/vl/cms/en/5/Villa)
- [Google Maps](https://goo.gl/maps/x5qGTdrMGfW3g24V6)

## What is provided?
* Lodging with Bath, Bedding and Towels
* Seminar room with LAN, WLAN, Internet
* Catering for breakfast and dinner (__please tell us if you have special requirements!__)
* Parking place

## How to register?
* The event is at no cost, lunch not included
* Please register via mail to martin.stein@genode-labs.com __till end of February 2020__

Let's now try to find some interesting outdoor activities for the event in the context of this issue! Afterwards, we will post schedule and travel information as usual. 

If you have further questions regarding the event, please don't hesitate to ask in this issue or via mail to martin.stein@genode-labs.com !

nfeske, 2019-08-12:
Fantastic! Thanks Martin for arranging this perfect location for us.

m-stein, 2020-03-09:
Hi Genodians,
Please keep in mind that the Hack'n'Hike takes place in 2 months! If you'd like to join us this year, please [send me a mail](mailto:martin.stein@genode-labs.com)!

m-stein, 2020-04-02:
# Important notice! #
Hi Genodians,

I hope you are doing all well?

Due to the current situation with COVID-19 in Germany and other countries around the world, we have decided to __cancel the Hack'n'Hike 2020__. It is quite unlikely that things will get back to normal until 8th of May and we'd like to prevent uncertainty about the event. Please also note that we're __not__ planning to provide an alternative date this year, so, next Hack'n'Hike will probably be held in April/May 2021.

Best regards and stay healthy!

m-stein, 2021-02-26:
This issue isn't of relevance anymore.

closed 2021-02-26 by m-stein