genodelabs/genode #3351: base-nova: kernel compile error with GCC 8.3.0

cproc, 2019-05-14:
When building 'make KERNEL=nova' with GCC 8.3.0, the following error occurs:

```
    COMPILE  acpi.o
cc1plus: error: -Wabi won't warn about anything [-Werror=abi]
cc1plus: note: -Wabi warns about differences from the most up-to-date ABI, which is also used by default
cc1plus: note: use e.g. -Wabi=11 to warn about changes from GCC 7
```

I'm not sure if we need this warning option at all, and if we do, which ABI number we should state there. There's some documentation of the option with a list of the available ABI numbers at [1].

[1] https://gcc.gnu.org/onlinedocs/gcc/C_002b_002b-Dialect-Options.html

alex-ab, 2019-05-15:
Read a bit through the documentation. From my understanding you can remove this option.

cproc, 2019-05-15:
Fixed in fe1dd16.

commit 4f12de31 (2019-05-15): nova: remove '-Wabi' compiler option

Fixes #3351
commit 382371d9 (2019-05-27): nova: remove '-Wabi' compiler option

Fixes #3351
closed 2019-05-29 by chelmuth