genodelabs/genode #3352: seoul: string function declaration mismatch errors with GCC 8.3.0

cproc, 2019-05-14:
When building 'make app/seoul' with GCC 8.3.0, the following errors occur:

```
    COMPILE  component.o
In file included from /.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/include/nul/bus.h:24,
                 from /.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/include/nul/vcpu.h:22,
                 from /.../repos/ports/src/app/seoul/component.cc:47:
/.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/genode/include/service/string.h:47:14: error: declaration of ‘const char* strchr(const char*, int)’ conflicts with built-in declaration ‘char* strchr(const char*, int)’ [-Werror=builtin-declaration-mismatch]
 const char * strchr(const char *s, int c);
              ^~~~~~
/.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/genode/include/service/string.h:51:5: error: declaration of ‘int strspn(const char*, const char*)’ conflicts with built-in declaration ‘long unsigned int strspn(const char*, const char*)’ [-Werror=builtin-declaration-mismatch]
 int strspn(const char *s, const char *accept);
     ^~~~~~
/.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/genode/include/service/string.h:52:5: error: declaration of ‘int strcspn(const char*, const char*)’ conflicts with built-in declaration ‘long unsigned int strcspn(const char*, const char*)’ [-Werror=builtin-declaration-mismatch]
 int strcspn(const char *s, const char *reject);
     ^~~~~~~

cproc, 2019-05-14:
@alex-ab: fixed in https://github.com/cproc/seoul/commit/aacb3cfe6d9f5ddaef62dea7f5dbd013953cb90b

cproc, 2019-05-14:
More compile errors:

```
    COMPILE  model/pit8254.o
/.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/model/pit8254.cc: In member function ‘PitCounter& PitCounter::operator=(const PitCounter&)’:
/.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/model/pit8254.cc:365:39: error: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘class PitCounter’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
     memcpy(this, &other, sizeof(*this));
                                       ^
```
```
    COMPILE  model/vbios.o
/.../contrib/seoul-dd8fd8837e8047c6261811c46047fc1003169b04/src/app/seoul/model/vbios.cc:199:5: error: expected ‘(’ before ‘volatile’
 asm volatile ("; .code16"
     ^~~~~~~~
     (
```

cproc, 2019-05-14:
@alex-ab: fixed in [1], [2] and [3]

[1] https://github.com/cproc/seoul/commit/aacb3cfe6d9f5ddaef62dea7f5dbd013953cb90b
[2] https://github.com/cproc/seoul/commit/cefb2d8cc7f07e301993c50d40552f390ec4238e
[3] https://github.com/cproc/seoul/commit/c3442d7bba1ea2e65c532d34abd79801b9c57e51

cproc, 2019-05-14:
These errors occur in the ports repository when C++17 is enabled:

```
    COMPILE  user_env.o
/.../repos/ports/src/app/seoul/user_env.cc:138:6: error: the program should also define ‘void operator delete(void*, long unsigned int)’ [-Werror=sized-deallocation]
 void operator delete (void * ptr)
      ^~~~~~~~
/.../repos/ports/src/app/seoul/user_env.cc:132:6: error: the program should also define ‘void operator delete [](void*, long unsigned int)’ [-Werror=sized-deallocation]
 void operator delete[](void *ptr)
      ^~~~~~~~
```

alex-ab, 2019-05-15:
@cproc: thanks. I added your commits to the seoul branch and created a genode commit with you as author - e694cd92ecae0ba228d1ecc797a370e58673d0fa

cproc, 2019-05-15:
@alex-ab, @chelmuth: the 'genreg.patch' commit just contains the change that was created during the seoul port preparation. To add my commits, I had to either revert this change or create a commit for it. It does not belong to this issue. Also, @alex-ab's Genode commit could be squashed with my "fix compile errors" commit above.

alex-ab, 2019-05-15:
Ok, removed the genreg.patch - fixup by 63915ce23273d5ce175be20be7f92895bf4ecd10.

commit cd682df3 (2019-05-16): ports: fix 'seoul' compile errors with GCC 8.3.0

Fixes #3352
commit 9923a1bf (2019-05-27): ports: fix 'seoul' compile errors with GCC 8.3.0

Fixes #3352
closed 2019-05-29 by chelmuth