genodelabs/genode #3353: tool/depot: improve handling of missing ports

m-stein, 2019-05-15:
Currently the depot/create and depot/extract tools stop whenever a port is missing and report to the user only this one missing port. This can be quite annoying when building, for instance, sculpt with dependencies to more than 20 ports. There is also no possibility to tell the tools to automatically prepare missing ports. However, the latter shouldn't be set by default as this might imply downloads and file operations (outside the depot directory) that the user has not explicitely approved by calling these tools.

m-stein, 2019-05-15:
Fixed:
df29a8a tool/depot: improve handling of missing ports

nfeske, 2019-05-15:
Thanks for the fantastic feature!

I'm deeply impressed that you endured the magic of make as put on display in these tools. Admit it: You enjoyed it a little, in a strange, perverse way! :-)

m-stein, 2019-05-15:
Nothing to hide here :) it was pleasantly challenging to understand the code and play with it.
But I also got the feeling that, besides its elegant main concept, 'make' also provides some horrifying flexibilities paired with bits of cryptographic syntax to the developer.

m-stein, 2019-05-17:
There's a fixup for a bug that occured when some ports were already prepared:
34d21caf9e Fixup "tool/depot: improve handling of missing ports" (content_env_missing_ports.mk)

m-stein, 2019-05-23:
@chelmuth @nfeske This should fix recent noux_tool_chain_auto issues: 
 6cdf293 Fixup "tool/depot: improve handling of missing ports" (check exactly those archives that'll get extracted)

nfeske, 2019-05-23:
Thanks @m-stein. I merged your commit, along with another minor fixup db3d7f2e529bd19fb77c0640ee12cb0ad49d35d0 (spelling).

m-stein, 2019-05-23:
@nfeske Thanks for the review and merge!

commit d9ef1ce0 (2019-05-22): tool/depot: improve handling of missing ports

* The extract tool determines and reports all missing ports at once.
* The extract tool automatically prepares all missing ports if PREPARE_PORTS=1.
* The missing_ports tool prints a list of missing ports for given archives.

Fixes #3353
closed 2019-05-29 by chelmuth