genodelabs/genode #3355: base: save FPU registers in '_jmp_slot' function

cproc, 2019-05-15:
GCC 8.3.0 generates more code which uses FPU registers and it happened with the 'log' test on x86_64 that the floating point number argument for the 'Genode::log()' function got overwritten in 'Genode::Cancelable_lock::lock()', which got called during the execution of the '_jmp_slot' linker function.

I have a fix for x86_64 so far and we might need one for x86_32 and ARM, too, or not. Saving the FPU state on the stack requires proper stack alignment on x86, so some components need to be adapted.

chelmuth, 2019-05-29:
Is there consent on the state of this issue? It's the last one blocking #3307 and 19.05 is already out ;-)

cproc, 2019-06-13:
@chelmuth: there didn't appear to be problems related to not saving the FPU state on x86_32 and ARM32 so far and there are no plans to implement it in the near future, so I guess the issue could be closed for now.

chelmuth, 2019-06-14:
Thanks, closing it now.

closed 2019-06-14 by chelmuth