genodelabs/genode #3361: tool chain: integrate 'ali2dep' tool

cproc, 2019-05-16:
I'd like to integrate the 'ali2dep' tool (issue #3144) with the 19.05 tool chain, but I'm not sure what exactly needs to be done.

So far, I checked out the 'ali2dep' repository and typed 'make', which printed the following error message:

```
build.gpr:5:08: undefined attribute "create_missing_dirs"
gprbuild: "build/build" processing failed
Makefile:16: recipe for target 'all' failed
make: *** [all] Error 4
```

@senier, @jklmnn: can you help here?

jklmnn, 2019-05-16:
You need the GNAT Community Toolchain to use the `create_missing_dirs` directive in gpr files. If you only use the Debian GNAT you can just create a directory named `build` in the repository and comment the line in the gpr file out.

cproc, 2019-05-16:
I commented the line out and now the following error occurs:

```
gnatgcc -c -gnatwI osint.adb
osint.adb:2597:17: "Null_FD" is undefined
```

This is on Ubuntu 16.04 with GNAT 5.4.0.

jklmnn, 2019-05-16:
I'm afraid we don't support anything older that GNAT 6.3.0 (which doesn't support the `create_missing_dirs` directive. Anything newer than that should work also.

cproc, 2019-05-16:
Ok, then I'll try to build it with the bootstrap tool chain which is built by the 'tool_chain' script.

cproc, 2019-05-17:
Fixed in c6c352a.

With GCC 6.3.0 the same 'Null_FD' error occured, so this commit depends on #3307.

commit 2a954419 (2019-05-20): tool_chain: integrate 'ali2dep' tool

Fixes #3361
closed 2019-05-29 by chelmuth