genodelabs/genode #3363: Create package for linux_nic_drv

trimpim, 2019-05-17:


skalk, 2019-05-20:
@trimpim: what is the difference of this package compared to the existent `drivers_nic-linux`, or did you missed that one?

trimpim, 2019-05-20:
@skalk I just missed the `drivers_nic-linux` package (as the naming schema is different as foe all other nic dirvers).

closed 2019-05-20 by trimpim

skalk, 2019-05-20:
@trimpim actually it isn't using a different scheme. All `pkg` (meta-)packages, which contain driver environments, like `drivers_interactive-*` and `drivers_nic-*` use the same leading prefix, but platform-dependent suffixes.
In contrast to that all binary names and their derived `src` package names are indeed using it vice versa, taking first the platform-specific part.