genodelabs/genode #3365: hw: triple faults on x86_64 when built with GCC 8.3.0

cproc, 2019-05-20:
GCC 8.3.0 generates more code which uses FPU instructions and this causes triple fault exceptions with 'bootstrap' and 'core' on hw/x86_64.

When running QEMU with the '-d int,cpu_reset --no-reboot' options, the first reported exception is:

```
...
Bender: Hello World.
check_exception old: 0xffffffff new 0x6
     0: v=06 e=0000 i=0 cpl=0 IP=0008:0000000000212745 pc=0000000000212745 SP=0018:0000000000233db8 env->regs[R_EAX]=00000000fec00000
RAX=00000000fec00000 RBX=0000000000230000 RCX=00000000fee00000 RDX=0000000000230318
RSI=0000000000100000 RDI=0000000000230000 RBP=0000000000230108 RSP=0000000000233db8
R8 =fffffffffffffea0 R9 =0000000000000000 R10=0000000000000000 R11=0000000000018110
R12=00000000002303a0 R13=0000000000230360 R14=000000000022a760 R15=000000000022a760
RIP=0000000000212745 RFL=00200046 [---Z-P-] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0018 0000000000000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
CS =0008 0000000000000000 00000000 00209800 DPL=0 CS64 [---]
SS =0018 0000000000000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
DS =0018 0000000000000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
FS =0000 0000000000000000 00000000 00000000
GS =0000 0000000000000000 00000000 00000000
LDT=0000 0000000000000000 0000ffff 00008200 DPL=0 LDT
TR =0000 0000000000000000 0000ffff 00008b00 DPL=0 TSS64-busy
GDT=     0000000000200140 00000037
IDT=     0000000000000000 00000000
CR0=80010011 CR2=0000000000000000 CR3=0000000000223000 CR4=00000020
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
CCS=0000000000230318 CCD=0000000000000000 CCO=SUBQ    
EFER=0000000000000d00
```

So it seems to be an 'invalid opcode' (#UD) exception which leads to the triple fault. The instruction causing the exception is

```
 212745:       66 0f ef c0             pxor   %xmm0,%xmm0
```

It's not clear yet why this causes an exception. I've seen it mentioned that 'CR4.OSFXSR' (bit 9) = 0 can cause #UD, but it might be something else.

The kernel boots with workaround commit [1], but it compiles a part of the base library with -O0, which might not be desired. When using the '-mno-sse' option instead, the 'log' test fails to print the floating point number correctly.

[1] https://github.com/cproc/genode/commit/e9b2292f48e689b456bf6e444b6bbbd11a914e09

ssumpf, 2019-05-20:
@cproc: As I stated today, I do not think the FPU is in a well defined state at this point. For example, the `fninit` function has not been called (I assume) at this point. There are also flags in CR0 and CR4, while the compiler seems to be free to use SSE2 instructions, since the are mandatory on amd64, at any given time. It might just be an exception (#GP or #floating point) in this early stage. Have a look at Nova and Hw for FPU initialization code, it is there.

ssumpf, 2019-05-21:
@chelmuth: a6ea901 fixes the issue.

ssumpf, 2019-05-23:
1663c2e and 366b7f6 fix the floating point issues on KVM and VirtualBox with SMP (gcc 8.3.0) and x201 (gcc 6.3.0). I will verify this with 8.3.0 on x201 tomorrow. Additionally, SMP support on Broadwell (x250) seems to be broken, even on master with gcc 6.3.0 the kernel does not come up when executing `run/log`.

nfeske, 2019-05-23:
Thanks a lot @ssumpf for investigating, the thorough testing, and for the two commits! I merged them to staging to see what the tests a saying. @skalk could you please have a look?

skalk, 2019-05-24:
@nfeske I already used to be @ssumpf's sidekick yesterday, but also will continue to investigate.

ssumpf, 2019-05-24:
775818a applies FPU changes to core-hw-muen

skalk, 2019-05-24:
@nfeske @ssumpf I've misunderstood the remark about: "have a look". Now, I have again reviewed the commits and apart from the one comment I made, I'm fine with the commits.

ssumpf, 2019-05-24:
As discussed: 45f4902 and 7095b7b. Separate the stack alignment commit.

ssumpf, 2019-05-24:
We also fixed the x250 in issue #3376.

commit 5680ba95 (2019-05-21): hw: enable FPU during CPU startup on x86

Also disable TS (task switch) flag in cr0 during kernel initialization,
so FPU faults are not raised. This became necessary since GCC lately
aggressively generates FPU instructions at arbitrary places and also at
early kernel-bootstrapping stages.

fixes #3365
commit e9c39980 (2019-05-22): hw: enable FPU during CPU startup on x86

Also disable TS (task switch) flag in cr0 during kernel initialization,
so FPU faults are not raised. This became necessary since GCC lately
aggressively generates FPU instructions at arbitrary places and also at
early kernel-bootstrapping stages.

fixes #3365
closed 2019-05-29 by chelmuth