genodelabs/genode #3369: nvme_drv does not work with QEMU 3.1.0

cnuke, 2019-05-22:
It looks like interrupts are not working anymore (the driver queries the device by polling, I/O is done with interrupts).

chelmuth, 2019-05-22:
Starting point for investigation could be the log of a traced Qemu run.
```
qemu-system-x86_64 -no-kvm -display sdl -cpu core2duo -display sdl -nographic -m 512 -drive id=nvme0,file=build/x86_64/bin/nvme.raw,format=raw,if=none -device nvme,drive=nvme0,serial=fnord,id=nvme0n1 -serial mon:stdio -cdrom build/x86_64/var/run/nvme.iso -machine q35 -d guest_errors,trace:nvme\*

24005@1558526041.014882:nvme_mmio_cfg wrote MMIO, config controller config=0x0
24005@1558526041.015453:nvme_mmio_aqattr wrote MMIO, admin queue attributes=0xff003f
24005@1558526041.015505:nvme_mmio_asqaddr wrote MMIO, admin submission queue address=0x1ffdd000
...
[init -> platform_drv] 00:03.0 uses IRQ, vector 0xb, non-maskable
[init -> nvme_drv] NVMe PCIe controller found (0x8086:0x5845)
24005@1558526045.625111:nvme_mmio_intm_set wrote MMIO, interrupt mask set, data=0xffffffff, new_mask=0xffffffff
24005@1558526045.625365:nvme_mmio_cfg wrote MMIO, config controller config=0x0
24005@1558526045.625504:nvme_mmio_stopped cleared controller enable bit
[init -> nvme_drv] Warning: ignore mpsmax:4
24005@1558526045.629016:nvme_mmio_cfg wrote MMIO, config controller config=0x0
24005@1558526045.629128:nvme_mmio_cfg wrote MMIO, config controller config=0x400000
24005@1558526045.629198:nvme_mmio_cfg wrote MMIO, config controller config=0x460000
24005@1558526045.647511:nvme_mmio_aqattr wrote MMIO, admin queue attributes=0x7f003f
24005@1558526045.658128:nvme_mmio_acqaddr wrote MMIO, admin completion queue address=0x50f000
24005@1558526045.674101:nvme_mmio_aqattr wrote MMIO, admin queue attributes=0x7f007f
24005@1558526045.674258:nvme_mmio_asqaddr wrote MMIO, admin submission queue address=0x5c8000
24005@1558526045.674310:nvme_mmio_cfg wrote MMIO, config controller config=0x460001
24005@1558526045.674385:nvme_mmio_start_success setting controller enable bit succeeded
24005@1558526045.692308:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1000, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1000, returning 0
24005@1558526045.692750:nvme_identify_ctrl identify controller
24005@1558526045.692846:nvme_dma_read DMA read, prp1=0x577000 prp2=0x0
24005@1558526045.693200:nvme_irq_pin pulsing IRQ pin
24005@1558526045.813978:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1004, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1004, returning 0
24005@1558526046.894946:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1000, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1000, returning 0
24005@1558526046.895296:nvme_identify_ns identify namespace, nsid=1
24005@1558526046.895400:nvme_dma_read DMA read, prp1=0x5ca000 prp2=0x0
24005@1558526046.895651:nvme_irq_pin pulsing IRQ pin
24005@1558526047.013642:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1004, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1004, returning 0
24005@1558526048.111315:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1000, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1000, returning 0
24005@1558526048.111677:nvme_create_cq create completion queue, addr=0x5cb000, cqid=1, vector=0, qsize=127, qflags=3, ien=1
24005@1558526048.111845:nvme_irq_pin pulsing IRQ pin
24005@1558526048.229502:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1004, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1004, returning 0
24005@1558526049.287525:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1000, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1000, returning 0
24005@1558526049.288636:nvme_create_sq create submission queue, addr=0x5ce000, sqid=1, cqid=1, qsize=127, qflags=1
24005@1558526049.288849:nvme_irq_pin pulsing IRQ pin
24005@1558526049.406930:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1004, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1004, returning 0
24005@1558526050.460328:nvme_mmio_intm_clr wrote MMIO, interrupt mask clr, data=0x1, new_mask=0xfffffffe
[init -> nvme_drv] NVMe:1.2.0 serial:'fnord' model:'QEMU NVMe Ctrl' frev:'1.0     '
[init -> nvme_drv] Block size:512 count:67108864
24005@1558526050.580720:nvme_ub_mmiord_invalid_ofs MMIO read beyond last register, offset=0x1008, returning 0
nvme_ub_mmiord_invalid_ofs in nvme_mmio_read: MMIO read beyond last register, offset=0x1008, returning 0
24005@1558526050.581000:nvme_rw read 128 blocks (65536 bytes) from LBA 0
24005@1558526050.582688:nvme_irq_pin pulsing IRQ pin
```

cnuke, 2019-05-22:
Thanks for the trace, I'll look into it after the relase.

cnuke, 2019-05-31:
FWIW, the invalid MMIO read requests are *false-positives* AFAICT. They appear to be caused by our `MMIO` framework which will read the registers before it writes them and QEMU does not anticipate reads in those regions.

m-stein, 2019-06-03:
@cnuke Just as a side note: The MMIO framework can be used also write-only when either writing a whole register at once (no bit-fields) or writing a register that has ``STRICT_WRITE = true`` (including its bit-fields).

cnuke, 2019-06-03:
@m-stein thanks for the clarification. I admit not phrasing my statement about the `MMIO` framework well, after all the `nvme_drv` instructs the framework to behave the way it does. I'll keep adding the write hint in mind when switching the driver to the `Block::Request_stream` API.

commit 32dd677d (2019-06-03): nvme_drv: only set the first INTMS bit on reset

Since we only set the first INTMC bit a check in QEMU will fail
and we don't get interrupts.

Fixes #3369.
commit 147c52c5 (2019-06-11): nvme_drv: only set the first INTMS bit on reset

Since we only set the first INTMC bit a check in QEMU will fail
and we don't get interrupts.

Fixes #3369.
commit 737e9580 (2019-06-13): nvme_drv: only set the first INTMS bit on reset

Since we only set the first INTMC bit a check in QEMU will fail
and we don't get interrupts.

Fixes #3369.
closed 2019-06-13 by chelmuth