genodelabs/genode #3377: Add [[fallthrough]] annotations where needed

cproc, 2019-05-24:
With the updated tool chain, the '-Wextra' compiler option requires '[[fallthrough]]' annotations for fallthrough situations in switch statements.

Currently, the following files (there might be more) are reported by the compiler and should probably get annotations if the fallthrough is actually intended (ideally checked by someone who is familiar with the respective files):

repos:

- base-hw/src/include/hw/spec/riscv/page_table.h
- base-hw/src/include/hw/spec/arm/lpae.h
- base-hw/src/include/hw/spec/arm/page_table.h
- base/src/lib/base/console.cc
- gems/src/app/gpt_write/util.h
- libports/src/lib/vfs/lwip/vfs.cc
- os/src/server/nic_router/dhcp_client.cc
- os/src/test/vfs_stress/main.cc

contrib:

- seoul: executor/vbios_keyboard.cc
- seoul: model/rtc146818.cc

There are more files affected in contrib sources which are not maintained by Genode Labs. Not sure if we want to patch these as well or build them with the '-Wno-implicit-fallthrough' compiler option instead.

- binutils
- gcc
- gdb
- stb

ssumpf, 2019-05-27:
- [x] base-hw/src/include/hw/spec/* (63049cd)

ssumpf, 2019-05-27:
- [x] os/src/server/nic_router/dhcp_client.cc (d5dc3d1)
- [x] base/src/lib/base/console.cc seems to work already

closed 2020-04-23 by nfeske