genodelabs/genode #3378: Remove libc_terminal plugin

ehmry, 2019-05-24:
The libc_terminal plugin is obsolete and can be replaced by the terminal VFS plugin in all cases (just vbox by now).

ehmry, 2019-05-24:
Fix at 1823d379560d07d6eb804f87de97e471d8bbbd82, can wait until after 19.05.

chelmuth, 2019-05-29:
I'm wondering if the removal of libc_terminal from the vbox run script is sufficient. Was it unused?

chelmuth, 2019-05-29:
Also...
```
> grep -rn libc_terminal repos/
repos/ports/recipes/src/vbox5/content.mk:39:                        lib/mk/libc_terminal.mk \
repos/ports/recipes/src/vbox5/content.mk:40:                        src/lib/libc_terminal \
```

ehmry, 2019-05-29:
As I understand it, the plugin only provided a "/dev/terminal" file that can be easily replaced by vfs terminal. Fixup at 34b562fd1414f75c557ea2239daa53f547ce7fcc.

chelmuth, 2019-05-29:
Then it's used in the following configurations, which means the respective run scripts must be adapted.
* https://github.com/genodelabs/genode/blob/master/repos/ports/run/vm_genode_usb_hid.vbox#L128
* https://github.com/genodelabs/genode/blob/master/repos/ports/run/virtualbox_nic_router.vbox#L66
* https://github.com/genodelabs/genode/blob/master/repos/ports/run/test.vbox#L66

ehmry, 2019-05-29:
Strange, I see the VFS terminal in already in use? https://github.com/genodelabs/genode/blob/24a3eea89e46c9effdd3ec2fbc5f85cf14ba9942/repos/ports/run/vbox5_genode_usb_hid.run#L208

chelmuth, 2019-05-29:
Then just ignore my ignorant code search results ;-)

ehmry, 2019-06-05:
The VFS terminal plugin does not work with Vbox because of missing `ioctl`s (unless libc_terminal plugin is present, which will handle the `ioctl` calls). @cnuke is this a good time to move forward on enriching the terminal plugin?

cnuke, 2019-06-05:
@ehmry thats fine from my side - all the needed ioctl definitions should already be in place in the VFS. You should get away with merely implementing the `File_io_server::ioctl` method. At some point we will remove this method but I don't see that happen before we have switched the VFS plugins to use some other mechanism for dealing with `ioctl` (see #3033).

commit 2027d9b4 (2019-05-29): Remove libc_terminal

Fix #3378
commit b883f5c9 (2019-06-11): Remove libc_terminal

Fix #3378
commit dc8a2ec5 (2019-06-13): Remove libc_terminal

Fix #3378
closed 2019-06-13 by chelmuth