genodelabs/genode #3383: Possible filesystem-related 'bitrot' in old code

ttcoder, 2019-05-27:
Took me a while, but I'm now successfully reading BFS, NTFS ..etc files from within Genode. In the process, I came across several places that needed to be tweaked (mostly one-liner tweaks), mainly in genode but also possibly in genode-world and contrib.

The code path mainly involves genode/os and fuse. I understand Sculpt has moved away from fuse (to Rump), which might explain some drifting away from the old code was not detected: it still compiles, but does not run. At my end, however, I do need FUSE, in order to get NTFS and BFS support (neither of which seem to be supported by Rump). Hopefully some of the below 18.11 "patches" can be upstreamed (when time permits! after 19.05 release).

***

fatfs_fs:
The FS variant needs to set non-zero inodes, so that opendir() and readdir() will return directory listings -- similar to the libc variant, which already has the fix: libports/src/lib/libc_fatfs/plugin.cc already contains this line:
			dirent->d_ino = 1; /* libc's readdir() wants an inode number */

libports/src/lib/libc/vfs_plugin.cc:942 : this gets triggered, and returns EINVAL, by e.g. ntfs-3g, resulting in a bail out instead of mounting the NTFS partition. I remed out the fcnt(F_SETLK) in unix_io.c in contrib, makes more sense than tweaking vfs_plugin.cc I think?

libports/src/server/fuse_fs: triggers this abort: libc suspend() called from non-user context (0x118edfa) - aborting
This is true for NTFS, ext2 ..etc. Solved it by adding lots of "with_libc" constructs around calls to fuse()->op in fuse_fs_main.cc, file.h, directory.h; not sure if that's the correct to attack this.

re. directory.h specifically: in _open_path(), calling FUSE opendir() returns ENOENT when dealing with ntfs-3g, which results in throwing an exception. I remed out the "throw" there, and can then read NTFS directories successfully. Didn't touch the contrib code for lack of understanding it, though it might well be the better choice for a tweak.

directory.h again: re inodes: need to prevent a "zero" e->inode (same as with fatfs_fs)
(tweaked read(), after the strncpy(), added e->inode = 1)

util.h (2x):
valid_filename(): currently rejects any path with a ":" colon character
(not just for creating new files, but also for reading existing ones). Those are used freely for directory names and file names in BFS (why wouldn't they, everything is allowed except slash'es and nul bytes there :-) so I remed that out in both util.h headers so that I can access my files. I understand the policy for Sculpt might need to differ though?

Also in BFS, noticed an ioctl(B_GET_GEOMETRY) was crashing in libc, but couldn't really investigate so just remed it out, and the partition gets mounted fine.

Thanks!

cnuke, 2019-05-27:
Thanks for the report. For better or worse, the FUSE code has not been actively used for years now, which is why it was moved to world (and I would have proposed to drop it at some point). It is nice to see that it still (somewhat) works and that there is actual interest in putting it to good use. On that account, spending some time to modernize the code seems reasonable after all.

If nobody steps forward to do that, I might be able to squeeze it in somewhere (I cannot give a time frame though).

ttcoder, 2019-05-27:
Well thanks for looking into it in the future (no hurry indeed, it will be weeks or months yet before I get my head out of the water and can "git pull" again, I'm still on 18.11 for now). I understand you'll need to setup and run tests ..etc to not apply my changes "in the blind" so that will take time away from other things.

ttcoder, 2019-10-08:
[diff_for_FUSE_ticket_3383.txt](https://github.com/genodelabs/genode/files/3703600/diff_for_FUSE_ticket_3383.txt)

ttcoder, 2019-10-08:
So I just attached a "git diff" above, going like this:

	git diff  repos/libports/src/lib/fuse/fuse.cc  repos/libports/src/server/fatfs_fs/directory.h  repos/libports/src/server/fuse_fs

I figure providing an actual diff will help build motivation :-)  I could also mention, don't worry about testing this very much, since I appear to be the only one (?) using FUSE like this.

There are lots of "Genode::log()" lines to be (of course) ignored, the important bits are outlined in my previous message.
The fat_fs fix can probably be ignored, due to #3512 (and I'm not planning to use FAT32 in production, plus there is an alternative way to get FAT32 support anyway).

The rest, however, is required if I am to run TTS apps: to recall, I rely on FUSE for file systems which are not available any other way, e.g. BFS. So Genode's FUSE support is high on my list.
Thanks!

ttcoder, 2019-10-10:
To be sure I understand,
does that mean there is one implementation using a hybrid libc+server model, or to two different implementations ?

In Genode 18.11 (anterior to "world" move) I see what appears to be two *distinct* ways to use FUSE, LibC and "FS" --

To confirm,
- I am not using  src/lib/libc_fuse
- I am relying on  src/server/fuse_fs

Both seem to have been moved at the same time to "world".

Are both of them deprecated ? Or maybe only the former should be retired, whereas the latter uses modern APIs ?

I.e. this one could be retired:
	https://github.com/genodelabs/genode-world/tree/master/src/lib/libc_fuse
	https://github.com/genodelabs/genode-world/tree/master/src/test/libc fuse (?)

And this one kept ? :
	https://github.com/genodelabs/genode-world/tree/master/src/server/fuse_fs
	(as a very minor side note: "ntfs-3g" could be renamed to "fuse-ntfs-3g", in accordance with the naming convention used for "fuse-ext2")

From what I understand of the Genode architecture so far, the latter uses the more modern client/server API.
That is to say, server/fuse_fs serves Vfs::Fs_file_system requests, which are made by the client program.

If it also has non-up-to-date parts I'll look into it, just to know what kind of changes to expect in that code base in the future :-)

**

In case it's useful, let me describe my setup:

 	<start name="CommandCenter_HoG" caps="350"> 
		..
		<route>
			<service name="File_system" label="part2">
				<child name="fs2_bfs"/>
			</service>

 	<start name="fs2_bfs"> 
		<binary name="bfs_fuse_fs"/>
		<provides> <service name="File_system"/> </provides>
		<config>
			<vfs>
				<dir name="dev">
					<block name="blkdev"/>

I assume that means that CC "talks" to BFS using a libc -> Fs_file_system (client) -----> Fs_file_system (server) -> FUSE -> BFS call chain.

(instead of the deprecated libc -> FUSE (client) ----> FUSE (server) .... call chain)

nfeske, 2019-10-11:
Yes, you got the picture quite right.

Let me give you a little bit more background. 

Originally, we had two hooks for supporting file-systems in Genode.

1. The libc plugin interface allowed one to link a file-system implementation directly to an application and access it via the libc.
2. The file-system session interface allowed for the implementation of a file-system server component. The file-system code lives in a (server) program that is separate from the application. This is useful in the common case where one file system is shared among multiple applications.

The libc-plugin interface was nice at the beginning but over time, we identified annoying limitations. Most importantly, in the presence of multiple libc plugins (let's say libc_terminal and libc_fatfs), there was no coherent file-system view from the application that captured both plugins in a consistent way. Worse, there was friction and inconsistencies between plugins. With Noux, we fixed these problems with a proper VFS implementation. The Noux VFS allows the structural composition of a virtual file system structure according to an XML description. This became one of the most valuable features of Noux. One particular cool thing of the VFS is the ability to integrate file-systems a plugins (which have the form of a shared library). Whenever the VFS encounters an unknown file-system type in its XML configuration, it tries to load a correspondingly named shared library for handling this particular file system type. With this mechanism, the file-system support of Noux could be enhanced without changing Noux.

Once we identified the power of Noux' VFS, we had the urge to equip regular libc-using programs with the same powers. One thing led to another. We separated the VFS from Noux, put the code into a distinct library, and created a libc-plugin that uses the VFS library. At this point, we became aware that the VFS would enable us to make the libc plugin interface redundant. Everything that we used the libc plugin interface for, we could also achieve with the VFS and VFS plugins.

At this point, we reached a situation where Genode had 3 hooks for file-system drivers. (1) The libc plugin interface, (2) the file-system session interface, and (3) the VFS plugin interface. This is flexible but also a maintenance hell:

* We found that most file-system server implementations contained a lot of boilerplate code copied from one another. Worse, the different implementations were not always consistent, e.g., regarding the application of a policy configuration.
* A change of the file-system session interface requires the adaptation of many file-system server components.
* For supporting a new file system, one had to implement multiple (up to 3), overlapping APIs.

To overcome this maintenance struggle while keeping the flexibility, we went for the following approach:

* All file-system drivers should have the form of VFS plugins only.

* We consolidate the various fs servers into a single "vfs" server that uses the VFS library internally and exports the virtual file system as a file-system session. By combining the VFS server with any of the file-system drivers (as plugin), the traditional file-system-specific server components can be "emulated". The VFS server + VFS plugin is a real drop-in replacement.
 
This way, supporting a new file system within Genode (regardless of whether this file system code ought to be used locally within an application, or as a file-system-server, or mounted within a Noux runtime) comes down to implementing a single VFS plugin (shared library).

Nowadays, whenever I encounter a file-system server other than the VFS server, I have the urge to get rid of it. ;-) The file-system driver code should really go into a VFS plugin to get the maximum flexibility and the smallest possible maintenance costs. That said, there is no technical reason against fs servers such as your `bfs_fuse_fs`. It would nevertheless be cool to explore the possibility if implementing the FUSE-based file-system drivers as VFS plugins. But this is not a must.

The libc plugin interface, however, - as you observed - is really deprecated. It stands in the way and will go away. So all libc plugins will definitely go with it. But according to the scenario you sketched above, this shouldn't be any problem for you.

ttcoder, 2019-10-14:
Gotcha -- the piece of the puzzle I was missing was the distinction between number 2 and number 3:
- FS session
- VFS plug-in

Implementing the former restricts you to being accessed on "remote" only (via IPC). Whereas implementing as a VFS plugin allows your code to be used either in a separate server, or be plugged in like a Lego block into the application itself; very flexible and elegant. If I had more time available it would make me want to *also* re-implement Haiku's "packagefs" as a Genode VFS plugin.. Seems it would be easier here than the original! Maybe in a year or two..

So I've started a complete overhaul of http://chiselapp.com/user/ttcoder/repository/genode-book/wiki?name=Book:Storage

Supposedly my roadmap would look something like this:
- wait for the fuse_vfs_plugin implementation
- port my BFS code to it (i.e. bfs_fuse_fs becomes bfs_fuse_vfs plugin)
- fuse_fs server can be retired.

Actually now would be a good time for me to take a look, whilst the code is fresh in my mind.

This puts my patch into question though: my patch consists heavily of with_libc() statements. But those statements might be structured in a completely different way in fuse_vfs, by opposed to fuse_fs, so maybe it will have to wait and be adapted, it makes little sense to apply it right now? Anyway now that i've filed a solid attachement here, I know there is a "backup" of my changes, so no matter what they will not be lost.

nfeske, 2019-10-28:
I'm happy that I could lift the clouds a bit. It's great that you keep detailed notes about your findings!

Let us move further discussions of FUSE to the genode-world issue tracker.

closed 2019-10-28 by nfeske