genodelabs/genode #3385: tool/run: power_on with external command

tomga, 2019-05-27:
I'm preparing to run automated tests on my raspberry pi boards. I have setup an Arduino like board that (with addtional module with relays) allows mi to perform resets of rpi. I can control this from linux with a python script using firmata library. This description is mostly to show that it is a custom non-generic solution.

To intergrate with run tool I implemented a simple power_on/command extension that calls external command that perfoms a device reset. and can be used with something like:
```
RUN_OPT += --include power_on/command
RUN_OPT += --power-on-command-cmd /script/do_reset.sh
RUN_OPT += --power-on-command-param device1
```
Maybe it may be useful to someone else.

chelmuth, 2019-05-28:
Let me reopen this for consideration after the release.

tomga, 2019-12-02:
@chelmuth, currently it is about after release, so maybe you could consider it? There is no need to have it open, I think, unless you are interested in it but with some changes.

chelmuth, 2019-12-04:
@tomga merged to staging

commit b0b3d50a (2019-12-04): tool/run: run external command plugin for power_on subsystem

Fixes #3385
commit b57a4c98 (2019-12-19): tool/run: run external command plugin for power_on subsystem

Fixes #3385
closed 2019-12-19 by chelmuth