genodelabs/genode #3386: tool/run: load/tftp and symbolic links pointing outside of directory served by tftp

tomga, 2019-05-27:
tftp load plugin of run tool creates symlinks in target tftp directory. Default configuration of tftp server on ubuntu (tftpd-hpa package) performs chroot to directory with served files which is good but it doesn't work with symlinks that link outside of tftp root dir.

Currently I resolved it for me with changing ``ln -sf`` to ``cp``. Would it be acceptable as a solution or would you prefer an option that would allow to specify copying instead of linking? Or maybe I should just keep my change locally just for me?

rite, 2019-05-28:
My local `run/boot_dir` files are patched the same way for hw, nova and sel4 ;)

rite, 2019-08-22:
- b874da6 is @tomga's commit rebased on current master (_hw_)
- 72731be does the same for _foc_ and _sel4_

rite, 2019-08-22:
- 4f71cf5 for _nova_

sid-com, 2019-09-17:
Previous commit broke iPXE boot. This commit dereferences all symlinks while copying to served location. Also, it makes sure that the copy process is done at the end of everything.

chelmuth, 2019-09-17:
Considering the activity in this issue I have a question regarding the implementation: May it be possible to *add* the copying of files as an *optional* feature while *retaining* the original behavior?

sid-com, 2019-09-19:
@chelmuth I like your idea. For my setup, I don't really mind the symlink approach. However, I understand the need of avoiding serving simlinks pointing outside the served directory.

chelmuth, 2019-09-20:
@sid-com I don't question the need for the feature and think we agree in opinion. So, I'd appreciate a fixup that replaces the symlinking by copying only if  `--load-ipxe-copy-files` etc. is provided. I understand that this requires refactoring *load/tftp* in a way that it provides transparent boot dir update as with *load/ipxe*.

tomga, 2019-12-02:
Above commit introduces option `--load-tftp-copy-files`. @chelmuth I named it differently from your proposal as it changes only behaviour for tftp.

I tested patch for `hw` but for `foc` and `sel4` probably should be ok too.

Second patch from @rite and the one from @sidhussmann modify `ipxe` part that I don't use but similar option (probably `--load-ipxe-copy-files` in this case) can be implemented similarly.

rite, 2019-12-03:
@tomga the patch for `ipxe` implementing @chelmuth suggestion is still on my todo list. Thanks for the blueprint!

chelmuth, 2019-12-04:
@tomga thanks for the nudge. I like the idea and also your implementation but still hesitate to merge as the current solution does not support all kernel-board combinations, esp. the quite common nova-pc is missing.

tomga, 2019-12-04:
Patch covers all places I found in code that refer to tftp directory parameters so I thought it is complete. I don't know the relation between `tftp` and `ipxe` so possibly I'm missing something. I just thought that `ipxe` needs its own option for similar behavior. Am I correct with this?

chelmuth, 2019-12-04:
Yeah, you're right, I'm only referring to TFTP. I'm no expert with the tftp-load implementation in combination with PC kernels but your solution does not adapt these. The new option *--tftp-load-copy-files* on the other hand is generic and just silently does nothing on PC.

chelmuth, 2019-12-04:
It may be the code in https://github.com/genodelabs/genode/blob/8cf7aaad6525de5e533715f937546d8a490909d6/tool/run/load/tftp#L36-L63

The TFTP boot-directory layout is different between Pulsar and Uboot: The first uses a link names *--load-tftp-offset-dir* in *--load-tftp-base-dir* while the uses a subdirectory.

tomga, 2019-12-05:
@chelmuth, I completely missed the fact that tftp is used for other kernels through `generate_tftp_config` function and agree with your remarks. Do I correctly understand that ensuring that this function understands `--load-tftp-copy-files` option will make this patch complete in your opinion?

chelmuth, 2019-12-06:
Exactly, IMO the differing interpretation of *base-dir* and *offset-dir* is confusing and should be streamlined to the Uboot approach. @alex-ab would you agree?

alex-ab, 2019-12-06:
@chelmuth: as long as the same functionality is kept, yes, otherwise no. The requirement up to now was that we have to accommodate tftp servers, where some of them using absolute paths and other us relative paths regarding a base directory. This is relevant for the generated pulsar config files, which have either have to have absolute or relative paths in, depending on your tftp server.

chelmuth, 2019-12-06:
I was always wondering what this *relative-path* scheme actually means. Or better I'd like to ask: Does it contradict to the Uboot approach of implementing *offset-dir* as a actual directory (to be `mkdir`d), which hosts links to (or copies of) boot configs and modules. Could you or @cnuke please enlighten me? Could we remove the generated step-stone *config-00-00-00-00-00-00* and just use *config-52-54-00-12-34-56*?

cnuke, 2019-12-06:
@chelmuth sorry, I can't recall why it's done this way/

alex-ab, 2019-12-06:
@chelmuth: this all gets off-topic. Our release notes are your friend https://genode.org/documentation/release-notes/13.05#TFTP_boot_x86

chelmuth, 2019-12-06:
@alex-ab I take this a *go* to remove the intermediate *config-00-00-00-00-00-00* and to just generate the real config into *var/run/TEST/config-00-00-00-00-00-00*.

- Without *--load-tftp-copy-files*, the links will be generated in *base-dir/offset-dir*:
  ```
  config-00-00-00-00-00-00 -> <path-to>/var/run/TEST/config-00-00-00-00-00-00
  boot -> <path-to>/var/run/TEST/boot
  ```
- With *--load-tftp-copy-files*, *config-00-00-00-00-00-00* and *boot* are copied recursively to *base-dir/offset-dir*.

Do I miss something?

alex-ab, 2019-12-06:
> boot -> 

Who is supposed to set this, e.g. on each run/<script> invocation some script has to that !?

alex-ab, 2019-12-06:
It does not matter, as said, keep the same functionality counts at the end of the day.

chelmuth, 2019-12-06:
> > boot ->
> Who is supposed to set this, e.g. on each run/<script> invocation some script has to do that !?

This is the duty of `generate_tftp_config` in *tool/run/load/tftp*.

On second thought this is also a asymmetry to all the Uboot-specific code in the boot_dir/ scripts.

tomga, 2019-12-06:
I did not think this issue will cause so much discussion but given that in some cases I didn't know at the beginning there is a need to copy directories I'd like to propose to use `rsync` instead of `cp` to synchronize content of `tftp` directory with build directory.

This would remove the need to recursively delete folders from scripts and I'm always afraid of placing `rm -rf` with configurable paths in scripts.

Additionally code for copying/synchonizing probably would be cleaner.

The bad think is that `rsync` currently is not a build dependency but given that it is a standard tool and only required after enabling specific option making it a dependency should not be a problem. 

What do you think? I'd prepare patch for this.

chelmuth, 2019-12-06:
To be honest, I do not like rsync to be added to our run-tool dependencies that casually. In fact, I'd be pretty confident with just using Tcl `glob`, `file copy`, and `file delete` for this purpose.

tomga, 2019-12-10:
It's good I asked before implementing it. I understand your point (and that's why I listed it as a potential problem) but personally I wouldn't see a problem with optional requirement for a quite standard tool in code that already requires something as big as `tftp` server.

On the other hand I don't have personal interest in writing directory synchronization code in tcl (to be honest in any language). I expect complications if targets exist and when file types do not match those in source. It seems for me to not be a simple task at all.

For me the first patch I provided is enough. I wanted to provide clean option to support my use case but it will be too much code that I don't need and would even have problems with any testing so I will not implement it. For me the issue can be closed. @chelmuth I'll leave decision to you in case you want it to be left open for fixing someday.

nfeske, 2021-02-26:
I merged https://github.com/genodelabs/genode/commit/a49d379ae736937d72085186bc317f1bfbae59c9 along with the fixup https://github.com/genodelabs/genode/commit/57c59e3234807e6f3ab22555b5fd79f2b2fa93ba to staging.

The latter fixes the switching between both options.

tomga, 2021-03-01:
> The latter fixes the switching between both options.

@nfeske thanks for spotting this problem and for merging this code. It was a minor issue but something I had to repeatedly apply every now and then.

commit 57c59e32 (2021-02-26): squash "tool/run: option to copy tftp served files"

Fixes #3386
commit 23e3079f (2021-03-12): tool/run: option to copy tftp served files

Fixes #3386
closed 2021-03-12 by nfeske