genodelabs/genode #3389: libc: silence isatty dummy

cnuke, 2019-05-29:
The same behaviour, i.e. return true for all fds, existed in the old libc as well but without the warning. We will have to provide a proper implementation eventually but for now it prefer to keep the noise down.

chelmuth, 2019-05-29:
With the libc update I also experienced lots of
```
posix_fadvise
sysctl for [6][10])
```
warnings. Could those also be silenced like above?

cnuke, 2019-05-29:
I have not looked at `posix_fadvise` but the sysctl is warning is the `HW_FLOATINPT` query  (see 87b58fd).

chelmuth, 2019-05-29:
@cnuke thanks for reminding me of the `sysctl(HW_FLOATINGPT)` patch. I renamed this issue appropriately and stashed both commits. I also learned `posix_fadvise()` was already silenced. but looking at noux_tool_chain_auto there are still many warnings about `signal not implemented`. May we mute this too or is it some WiP?

cnuke, 2019-05-29:
Since we return `SIG_ERR` in the dummy implementation and all test-cases currently seem to work fine I don't mind making it silent too. When we enable signal handling in the libc we'll have to change it anyway (which hopefully will happen in the next release cycle).

cnuke, 2019-06-03:
@chelmuth please merge 2b28997 (fixes ieee754 test) and 642d1e5 (silence signal) to staging.

cnuke, 2019-06-04:
@chelmuth please merge f9d5d33 (vim calls sigaction repeatedly).

commit e02c43c9 (2019-05-29): libc: slient isatty dummy

For the moment it is not possible to determine if the fd belongs to
a tty. For whatever reasons, the check is done multiple times, e.g.
'tclsh', which will spam the LOG.

Fixes #3389.
commit 178080cf (2019-05-29): libc: silence noncritical dummy implementations

* isatty

  For the moment it is not possible to determine if the fd belongs to a
  tty. For whatever reasons, the check is done multiple times, e.g.
  'tclsh', which will spam the LOG.

* sysctl(HW_FLOATINGPT)

  FPU is enabled on all our platforms, so return true.

Fixes #3389.
commit c572cad3 (2019-06-11): libc: silence noncritical dummy implementations

* isatty

  For the moment it is not possible to determine if the fd belongs to a
  tty. For whatever reasons, the check is done multiple times, e.g.
  'tclsh', which will spam the LOG.

* sysctl(HW_FLOATINGPT)

  FPU is enabled on all our platforms, so return true.

Fixes #3389.
commit 6b48a086 (2019-06-13): libc: silence noncritical dummy implementations

* isatty

  For the moment it is not possible to determine if the fd belongs to a
  tty. For whatever reasons, the check is done multiple times, e.g.
  'tclsh', which will spam the LOG.

* sysctl(HW_FLOATINGPT)

  FPU is enabled on all our platforms, so return true.

Fixes #3389.
closed 2019-06-13 by chelmuth