genodelabs/genode #3391: noux: spawned child gets stuck apparently

cnuke, 2019-05-31:
Every now and than the import subsystem in the *genodians.org* appliance gets stuck (it currently runs on [cnuke/genodians-2019-05-29](https://github.com/cnuke/genode/commits/genodians_2019-05-29)). I have set the `verbose` option in the noux configuration which shows the arguments of the spawned children. The following LOG excerpt illustrates the issue:

```
[…]
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] generate html/ehmry/2019-03-22-depot_9P ...
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] child /bin/bash exited with exit value 0
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "/"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "bash"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(1): "-c"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(2): "./tool/gosh/gosh --style html --html-img-format png --utf8 --style style/nice_date --style style/posting.gosh --top-path "../" \
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate]         --link ehmry/2019-03-22-depot_9P --author "Emery Hemingway" --flair '  ' content/ehmry/2019-03-22-depot_9P.txt > html/ehmry/2019-03-22-depot_9P"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "b"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "/usr/bin/tclsh"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(1): "./tool/gosh/gosh"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(2): "--style"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(3): "html"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(4): "--html-img-format"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(5): "png"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(6): "--utf8"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(7): "--style"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(8): "style/nice_date"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(9): "--style"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(10): "style/posting.gosh"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(11): "--top-path"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(12): "../"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(13): "--link"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(14): "ehmry/2019-03-22-depot_9P"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(15): "--author"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(16): "Emery Hemingway"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(17): "--flair"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(18): "  "
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(19): "content/ehmry/2019-03-22-depot_9P.txt"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] child /usr/bin/tclsh exited with exit value 0
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] child /bin/bash exited with exit value 0
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "/"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "bash"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(1): "-c"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(2): "echo generate html/ehmry/2019-03-07-tup ..."
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] generate html/ehmry/2019-03-07-tup ...
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] child /bin/bash exited with exit value 0
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "/"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "bash"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(1): "-c"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(2): "./tool/gosh/gosh --style html --html-img-format png --utf8 --style style/nice_date --style style/posting.gosh --top-path "../" \
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate]         --link ehmry/2019-03-07-tup --author "Emery Hemingway" --flair '  ' content/ehmry/2019-03-07-tup.txt > html/ehmry/2019-03-07-tup"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "b"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(0): "/usr/bin/tclsh"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(1): "./tool/gosh/gosh"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(2): "--style"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(3): "html"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(4): "--html-img-format"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(5): "png"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(6): "--utf8"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(7): "--style"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(8): "style/nice_date"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(9): "--style"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(10): "style/posting.gosh"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(11): "--top-path"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(12): "../"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(13): "--link"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(14): "ehmry/2019-03-07-tup"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(15): "--author"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(16): "Emery Hemingway"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(17): "--flair"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(18): "  "
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] arg(19): "content/ehmry/2019-03-07-tup.txt"
[init -> date_log] 2019-5-30 5:27:15 [genodians -> import -> generate] child /usr/bin/tclsh exited with exit value 0
[init -> date_log] 2019-5-30 6:11:15 [genodians -> lighttpd] 2019-05-30 06:10:47: (/data/develop/depot/local/src/lighttpd/2019-05-26/src/app/lighttpd/src/connections.c.1025) connection closed - read failed:  Input/output error 5
[…]
```

The exit of the `tclsh` child is the last message we get from the `generate` noux instance. In all other cases this exit is followed by the exit of the `bash` child (see the generate of the previous post).

cnuke, 2019-06-17:
Apparently it gets stuck in `Child_policy::exit` in line 197 which is the `_file_descriptor_registry.flush()` call.

cnuke, 2019-06-19:
I added some debug messages to `_reset_fd` where _before_ is printed before `_fds[fd].io_channel = Shared_pointer<Io_channel>();` and *after* is printed afterwards.

The import Subsystem got stuck again yesterday:
 
```
[…]
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 3 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 3 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] child /usr/bin/tclsh exited with exit value 0
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:192
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 0 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 0 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 1 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 1 after
[init -> date_log] 2019-6-18 17:0:41 [genodians -> nic_router] [uplink] dynamic IP config: none
[…]
```

When the child exit succeeds, which it did several times before, it looks as follows:

```
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 3 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] child /usr/bin/tclsh exited with exit value 0
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:192
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 0 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 0 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 1 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 1 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 2 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 2 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:201
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:204
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:208
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:216
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] child /bin/bash exited with exit value 0
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:192
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 0 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 0 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 1 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 1 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 2 before
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: _reset_fd: fd: 2 after
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:201
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:204
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:208
[init -> date_log] 2019-6-18 16:49:6 [genodians -> import -> generate] Error: exit:216
```

(Don't let the date fool you, it happend all in the same second.)

nfeske, 2019-11-06:
We're deprecating noux in favor of using the libc directly. So noux won't get any further attention.

closed 2019-11-06 by nfeske