genodelabs/genode #3392: FOC: build fails if contrib is a symbolic link

trimpim, 2019-05-31:


chelmuth, 2019-06-03:
What do you mean by *contrib* being a symbolic link? *\<genode-dir\>/contrib* is a symbolic link in my case and the build works flawlessly.

trimpim, 2019-06-04:
yes, on my machine (Ubuntu 18.04) when `<genode-dir>/contrib` is a symbolic link (to a different disk) the build fails. I also have tried to mount the second disk directly to `<genode-dir>/contrib` this also failed the build.
So maybe the problem isn't with the symbolic link but with the contrib sources being on an other disk.

chelmuth, 2019-06-04:
I still have no clue what the error is. Do you see warnings/errors from `cp`? I tried your scenario with a second disk (*/tmp* mount) and got the following errors if the source directory was not given as absolute path to `cp`.

```
user@veo:~/plain/tmp> cp -rs genode /tmp/genode/
cp: /tmp/genode/genode/file1: can make relative symbolic links only in current directory
cp: /tmp/genode/genode/file2: can make relative symbolic links only in current directory
cp: /tmp/genode/genode/dir1/file1: can make relative symbolic links only in current directory
cp: /tmp/genode/genode/dir1/file2: can make relative symbolic links only in current directory
cp: /tmp/genode/genode/dir2/file1: can make relative symbolic links only in current directory
cp: /tmp/genode/genode/dir2/file2: can make relative symbolic links only in current directory
user@veo:~/plain/tmp> ls -alR /tmp/genode/
/tmp/genode/:
total 24
drwxrwxr-x  4 user user  4096 Jun  4 08:46 ./
drwxrwxrwt 13 root    root    12288 Jun  4 08:46 ../
drwxrwxr-x  2 user user  4096 Jun  4 08:46 dir1/
drwxrwxr-x  2 user user  4096 Jun  4 08:46 dir2/

/tmp/genode/dir1:
total 8
drwxrwxr-x 2 user user 4096 Jun  4 08:46 ./
drwxrwxr-x 4 user user 4096 Jun  4 08:46 ../

/tmp/genode/dir2:
total 8
drwxrwxr-x 2 user user 4096 Jun  4 08:46 ./
drwxrwxr-x 4 user user 4096 Jun  4 08:46 ../
user@veo:~/plain/tmp> cp -rs $(readlink -e genode) /tmp/genode/
user@veo:~/plain/tmp> ls -alR /tmp/genode/
/tmp/genode/:
total 28
drwxrwxr-x  5 user user  4096 Jun  4 08:48 ./
drwxrwxrwt 13 root    root    12288 Jun  4 08:47 ../
drwxrwxr-x  2 user user  4096 Jun  4 08:46 dir1/
drwxrwxr-x  2 user user  4096 Jun  4 08:46 dir2/
drwxrwxr-x  4 user user  4096 Jun  4 08:48 genode/

/tmp/genode/dir1:
total 8
drwxrwxr-x 2 user user 4096 Jun  4 08:46 ./
drwxrwxr-x 5 user user 4096 Jun  4 08:48 ../

/tmp/genode/dir2:
total 8
drwxrwxr-x 2 user user 4096 Jun  4 08:46 ./
drwxrwxr-x 5 user user 4096 Jun  4 08:48 ../

/tmp/genode/genode:
total 16
drwxrwxr-x 4 user user 4096 Jun  4 08:48 ./
drwxrwxr-x 5 user user 4096 Jun  4 08:48 ../
drwxrwxr-x 2 user user 4096 Jun  4 08:48 dir1/
drwxrwxr-x 2 user user 4096 Jun  4 08:48 dir2/
lrwxrwxrwx 1 user user   31 Jun  4 08:48 file1 -> /plain/user/tmp/genode/file1
lrwxrwxrwx 1 user user   31 Jun  4 08:48 file2 -> /plain/user/tmp/genode/file2

/tmp/genode/genode/dir1:
total 8
drwxrwxr-x 2 user user 4096 Jun  4 08:48 ./
drwxrwxr-x 4 user user 4096 Jun  4 08:48 ../
lrwxrwxrwx 1 user user   36 Jun  4 08:48 file1 -> /plain/user/tmp/genode/dir1/file1
lrwxrwxrwx 1 user user   36 Jun  4 08:48 file2 -> /plain/user/tmp/genode/dir1/file2

/tmp/genode/genode/dir2:
total 8
drwxrwxr-x 2 user user 4096 Jun  4 08:48 ./
drwxrwxr-x 4 user user 4096 Jun  4 08:48 ../
lrwxrwxrwx 1 user user   36 Jun  4 08:48 file1 -> /plain/user/tmp/genode/dir2/file1
lrwxrwxrwx 1 user user   36 Jun  4 08:48 file2 -> /plain/user/tmp/genode/dir2/file2
```

Still, I'd like to prevent to copy all files into the build directory.

trimpim, 2019-06-04:
I get the following error (when building run/log):
    COMPILE  stack_area_addr.o
    COMPILE  thread_start.o
      [l4re-core/l4util]    [l4util] ==> Installing libl4util.so to local build-tree
      [l4re-core/uclibc-minimal]  mkdir -p /home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src
      [l4re-core/uclibc-minimal]  cp -sfr /contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/../contrib/uclibc/libc /contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc  /contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/../contrib/uclibc/libm /contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/../contrib/uclibc/libcrypt /home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/misc/dirent/dirstream.h' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/misc/dirent/dirstream.h' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/signal/raise.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/signal/raise.c' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/signal/sigjmp.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/signal/sigjmp.c' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/stdlib/_atexit.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/stdlib/_atexit.c' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/stdlib/abort.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/stdlib/abort.c' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/stdlib/malloc-standard/malloc.h' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/stdlib/malloc-standard/malloc.h' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/sysdeps/linux/common/getdents.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/sysdeps/linux/common/getdents.c' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/sysdeps/linux/common/getsid.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/sysdeps/linux/common/getsid.c' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/sysdeps/linux/common/longjmp.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/sysdeps/linux/common/longjmp.c' are the same file
      [l4re-core/uclibc-minimal]  cp: '/contrib/foc-32f08f400da60630ddfd4f93cba642dce3a6f882/src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/unistd/sleep.c' and '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/libc/unistd/sleep.c' are the same file
      [l4re-core/uclibc-minimal]  make[5]: *** No rule to make target '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/src/.links-done', needed by '/home/pirmin/GapFruit/genode/build/imx6q_apalis/var/libcache/syscall-foc/imx6q_apalis-build/pkg/l4re-core/uclibc-minimal/libc/.general.d'.  Stop.
      [l4re-core/uclibc-minimal]  make[5]: *** Waiting for unfinished jobs....
      [l4re-core/uclibc-minimal]  ../../../mk/subdir.mk:41: recipe for target 'libc' failed
      [l4re-core/uclibc-minimal]  make[4]: *** [libc] Error 2
/home/pirmin/GapFruit/genode/repos/base-foc/lib/mk/l4_pkg.inc:58: recipe for target 'l4re-core/uclibc-minimal.imx6q_apalis.tag' failed
make[3]: *** [l4re-core/uclibc-minimal.imx6q_apalis.tag] Error 2
[... cut ...] oc.lib' failed
make[2]: *** [kernel-foc.lib] Error 2
make[2]: *** Waiting for unfinished jobs....
    MERGE    base-foc.lib.a
Makefile:286: recipe for target 'gen_deps_and_build_targets' failed
make[1]: *** [gen_deps_and_build_targets] Error 2
make[1]: Leaving directory '/home/pirmin/GapFruit/genode/build/imx6q_apalis'
Error: Genode build failed
Makefile:323: recipe for target 'run/log' failed
make: *** [run/log] Error 252
make: Leaving directory '/home/pirmin/GapFruit/genode/build/imx6q_apalis'

trimpim, 2019-06-04:
There are files with the same name in the directories `src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/../contrib/uclibc/libc/` and `src/kernel/foc/l4/pkg/l4re-core/uclibc/lib/uclibc/ARCH-all/libc/`. I assume, that the idea is that the later overwrite the former ones.
For some reason this doesn't work if the contrib sources are on an other partition. The target files still point to the first directory after `cp` has finished with the warning.

chelmuth, 2019-06-04:
Could you tell us which filesystem your mounted volumes use (ext4 or anything else)? I tested again with plain ext4 for *build* and ecryptfs for *contrib*, which works for me.

trimpim, 2019-06-04:
My partitions are:
* for `/ ` ext4 on an encrypted volume (this contains `<genode-dir>/build/`)
  - `/dev/mapper/xubuntu--vg-root on / type ext4 (rw,relatime,errors=remount-ro,data=ordered)`
* for `/contrib` ext4 on a plain disk
  - `/dev/sdb1 on /contrib type ext4 (rw,relatime,data=ordered)`

trimpim, 2019-06-04:
@chelmuth your comment inspired me to change the arrangement of my directory slightly. By moving my `build` directory to the same disk as the` contirb` directory, I'm now able to build Fiasco.OC.

closed 2019-06-04 by trimpim