genodelabs/genode #3393: ttf_font: out of bound memory access causes page fault on Sculpt

alex-ab, 2019-06-01:
In [ttf_font](https://github.com/genodelabs/genode/blob/master/repos/gems/src/lib/ttf_font/ttf_font.cc#L159) x0 may become negative. It is later used to calculate a [pointer](https://github.com/genodelabs/genode/blob/master/repos/gems/src/lib/ttf_font/ttf_font.cc#L177) which goes out of bound. The contrib code writes to this pointer and may cause a page fault (seen on Sculpt - leitzentrale -> fonts_fs).

alex-ab, 2019-06-01:
As far as I understand x0 is already considered in dx and width for the glyph, so x0 is not necessary ? @nfeske: please check b88f8e4200a87f1d97fe257dd27a8e92c6d5e7d4

nfeske, 2019-06-03:
Thanks for reporting! I have experienced this issue sporadically in the past but was not able to stably reproduce it. Do you have a way to trigger the problem deterministicaly?

> As far as I understand x0 is already considered in dx and width for the glyph, so x0 is not necessary for the pointer calculation ?

The `x0` value is used for the line-length calculation (dx, width) but also for the horizontal positioning (the use of the value as pointer offset). Apparently, the shape of some characters goes beyond the left border of bounding box. It would be good to know which character is that.

alex-ab, 2019-06-03:
Just use run/text_painter and instrument x0, you will see x0 is negative already for the simple test case. Since the '_values' pointer is the start of allocated chunk of memory ``` Opacity * const _values = (Opacity *)alloc.alloc(_num_bytes()); ```, every x0 smaller 0 will go out of bound and corrupt memory before that pointer.

ssumpf, 2019-06-04:
FYI: The same thing happens here.

nfeske, 2019-06-04:
@ssumpf Could you please give commit d96e26b8499c89134813f586156141200055309f a try?

ssumpf, 2019-06-04:
@nfeske: The commit works fine on my notebook.

commit 088c3500 (2019-06-04): ttf_font: prevent out-of-bounds access

Thanks Alexander Böttcher for investigating.

Fixes #3393
commit 7e174e73 (2019-06-13): ttf_font: prevent out-of-bounds access

Thanks Alexander Böttcher for investigating.

Fixes #3393
closed 2019-06-13 by chelmuth