genodelabs/genode #3395: ada: exception handling on ARM broken

chelmuth, 2019-06-03:
During the update to GCC 8.3.0, we discovered in issue #3371 that the exception handling in `test-spark_exception` fails on ARM (while it works on x86) like follows.

```
[init -> test -> test-spark_exception] Ada exception test
[init -> test -> test-spark_exception] Error: Exception raised: explicit raise in except.adb:6
```

@jklmnn discovered that `__gnat_personality_v0` is called with strange arguments.

chelmuth, 2019-06-03:
When testing with commit https://github.com/Componolit/ada-runtime/commit/4127fb7505da318423cde00cd829e618509f48f9 the `exception` argument points to `__gnat_personality_v0@plt`. Therefore, I tried BIND_NOW linking with

```diff
diff --git a/repos/libports/recipes/pkg/test-spark_exception/runtime b/repos/libports/recipes/pkg/test-spark_exception/runtime
index fe7cda91c1..bc603cae9b 100644
--- a/repos/libports/recipes/pkg/test-spark_exception/runtime
+++ b/repos/libports/recipes/pkg/test-spark_exception/runtime
@@ -29,6 +29,7 @@
                <default caps="100"/>
                <start name="test-spark_exception">
                        <resource name="RAM" quantum="10M"/>
+                       <config ld_verbose="yes" ld_bind_now="yes"/>
                </start>
        </config>
 </runtime>
```

and had to add missing symbol implementations to *spark.lib.so*

```diff
diff --git a/platform/genode.cc b/platform/genode.cc
index 1262c0f..9e543e9 100644
--- a/platform/genode.cc
+++ b/platform/genode.cc
@@ -53,6 +53,9 @@ extern "C" {
         }
     }
 
+char get_char(void) { return (char)0; }
+char get_int(void) { return (int)0; }
+
     void put_char(const char c)
     {
         print_stdout.out_char(c);
```

Unfortunately, the eager binding did not change the result and `exception` still refers to the PLT address.

ssumpf, 2019-06-03:
As a note, looking into GNAT's personality implementation (https://github.com/morbos/gnat2015/blob/master/ravenscar-full-rm46/common/raise-gcc.c), there exists special code for ARM (__ARM_EABI_UNWINDER__). Even the signature of the personality function seems to be different:
`PERSONALITY_FUNCTION (_Unwind_State state,
		      struct _Unwind_Exception* ue_header,
struct _Unwind_Context* uw_context);`, which suggests there is no `exception` argument.

jklmnn, 2019-06-03:
Thanks for that info. It looks like this is the actual problem, I'll look into that.

jklmnn, 2019-06-07:
I took a look into the ARM personality code and tried to build the same semantics as for the x86 code. But I couldn't find any way to pass the exception to Genodes exception handling mechanism. Returning `_URC_CONTINUE_UNWIND` results only in indefinite calls to the personality function (which indicates that there is no other handler that is responsible for handling this exception). Currently I return `_URC_FAILURE` which indicates that the personality function was not able to handle the exception which in this case seems to be the best option.

chelmuth, 2019-06-07:
Does this render the test successful or useless on ARM?

jklmnn, 2019-06-07:
This renders the test useless on ARM. A useful test for ARM would be that the component actually returns with a non-successful status code (and maybe that the exception is caught by Genodes last chance handler, which isn't the case if the wrong value is returned by the personality function). So we could test that an exception indeed ends the component unsuccessfully and that it doesn't cause a page fault.

m-stein, 2019-06-07:
@jklmnn Is it possible to tell (log output) where the exception originates from when it terminates the component?

jklmnn, 2019-06-07:
Yes the exception message will still be printed as this is done in out runtime implementation. It just cannot be caught from surrounding C++ code.

m-stein, 2019-06-07:
Ah, thanks. IMO this is fine for now. At least I never had a use case for throwing exceptions across language barriers and if desired one could also translate them manually. Or am I missing a thing?

jklmnn, 2019-06-07:
I don't know what you mean with translating them manually? Those are already C++ exceptions just that they're thrown from within Ada code (raising an exception calls C++ code on Genode that then throws a C++ exception).

chelmuth, 2019-06-07:
I suggest the following. Merge your cleanup of the personality stuff and disable the test on ARM. If you agree I'd appreciate a patch for this.

chelmuth, 2019-06-11:
I disabled *test-spark_exception* on ARM with 2f83c21a492a2253d4a0a18f4ab8ba005d73ccf3

jklmnn, 2019-06-11:
I updated the runtime to the behavior stated above in https://github.com/jklmnn/genode/commit/11da8dfd5f1fc5bcdffb884574c2676b5335830b.

chelmuth, 2019-06-11:
Thanks, I merged the commit to staging.

chelmuth, 2019-07-09:
Fix merged.

closed 2019-07-09 by chelmuth