genodelabs/genode #3398: x86_32 qemu pistachio: test-fs_rom_update

m-stein, 2019-06-03:
The ``fs_rom`` in this test complained about  "/dynamic_rom ROM file is missing" last night:

```
[init -> depot_autopilot] --- Run "test-fs_rom_update" (max 20 sec) ---[0m[0m
[init -> depot_autopilot] [0m[0m
[init -> depot_autopilot] 5.446 [init -> fs_rom] /dynamic_rom ROM file is missing[0m[0m[0m
[init -> depot_autopilot] 5.542 [init -> dynamic_rom] dynamic_rom: change (iteration 1)[0m[0m[0m
[init -> depot_autopilot] 5.546 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m
[init -> depot_autopilot] 5.569 [init -> fs_rom] /dynamic_rom ROM file is missing[0m[0m[0m
[init -> depot_autopilot] 5.591 [init -> fs_rom] /dynamic_rom ROM file is missing[0m[0m[0m
[init -> depot_autopilot] 5.618 [init -> rom_logger] ROM 'dynamic_rom':[0m[0m[0m
[init -> depot_autopilot] 7.568 [init -> dynamic_rom] dynamic_rom: change (iteration 2)[0m[0m[0m
[init -> depot_autopilot] 7.575 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m
[init -> depot_autopilot] 9.576 [init -> dynamic_rom] dynamic_rom: change (iteration 3)[0m[0m[0m
[init -> depot_autopilot] 9.583 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m
[init -> depot_autopilot] 11.590 [init -> dynamic_rom] dynamic_rom: change (iteration 4)[0m[0m[0m
[init -> depot_autopilot] 11.603 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m
[init -> depot_autopilot] 13.605 [init -> dynamic_rom] dynamic_rom: change (iteration 1)[0m[0m[0m
[init -> depot_autopilot] 13.610 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m
[init -> depot_autopilot] 15.625 [init -> dynamic_rom] dynamic_rom: change (iteration 2)[0m[0m[0m
[init -> depot_autopilot] 15.631 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m
[init -> depot_autopilot] 17.644 [init -> dynamic_rom] dynamic_rom: change (iteration 3)[0m[0m[0m
[init -> depot_autopilot] 17.651 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m
[init -> depot_autopilot] 19.661 [init -> dynamic_rom] dynamic_rom: change (iteration 4)[0m[0m[0m
[init -> depot_autopilot] 19.674 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds[0m[0m[0m

I was not able to reproduce this neither with my qemu (same version as mogli) nor with hardware. The test also failed only last night and not the nights before at the weekend.

m-stein, 2019-06-03:
The "ROM file missing" might not be bad, it sometimes occurs at the beginning of the test:
```
[init -> depot_autopilot] --- Run "test-fs_rom_update" (max 20 sec) ---
[init -> depot_autopilot] 
[init -> depot_autopilot] 2.596 [init -> fs_rom] /dynamic_rom ROM file is missing
[init -> depot_autopilot] 2.624 [init -> fs_rom] /dynamic_rom ROM file is missing
[init -> depot_autopilot] 2.634 [init -> fs_rom] /dynamic_rom ROM file is missing
[init -> depot_autopilot] 2.641 [init -> fs_rom] /dynamic_rom ROM file is missing
[init -> depot_autopilot] 2.645 [init -> rom_logger] ROM 'dynamic_rom':
[init -> depot_autopilot] 2.712 [init -> dynamic_rom] dynamic_rom: change (iteration 1)
[init -> depot_autopilot] 2.717 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[init -> depot_autopilot] 2.755 [init -> rom_logger] ROM 'dynamic_rom':
[init -> depot_autopilot] 2.758 [init -> rom_logger]   <config iteration="1" />
[init -> depot_autopilot] 2.761 [init -> rom_logger]   					
[init -> depot_autopilot] 4.718[init -> depot_autopilot] 
[init -> depot_autopilot]  test-fs_rom_update              ok         8.799  log "&lt;config iteration="4" />" [init -> dynamic_rom] dynamic_rom: change (iteration 2)
[init -> depot_autopilot] 4.736 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[init -> depot_autopilot] 4.779 [init -> rom_logger] ROM 'dynamic_rom':
[init -> depot_autopilot] 4.784 [init -> rom_logger]   <config iteration="2" />
[init -> depot_autopilot] 4.788 [init -> rom_logger]   					
[init -> depot_autopilot] 6.724 [init -> dynamic_rom] dynamic_rom: change (iteration 3)
[init -> depot_autopilot] 6.739 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[init -> depot_autopilot] 6.783 [init -> rom_logger] ROM 'dynamic_rom':
[init -> depot_autopilot] 6.788 [init -> rom_logger]   <config iteration="3" />
[init -> depot_autopilot] 6.793 [init -> rom_logger]   					
[init -> depot_autopilot] 8.729 [init -> dynamic_rom] dynamic_rom: change (iteration 4)
[init -> depot_autopilot] 8.745 [init -> dynamic_rom] dynamic_rom: sleep 2000 milliseconds
[init -> depot_autopilot] 8.795 [init -> rom_logger] ROM 'dynamic_rom':
[init -> depot_autopilot] 8.799 [init -> rom_logger]   <config iteration="4" />
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-fs_rom_update              ok         8.799  log "&lt;config iteration="4" />"
```

m-stein, 2019-06-03:
Currently, I have no idea how to proceed here. Even after 100 iterations, the bug didn't occur on my machine. I also tried to trigger timing issues by delaying several code paths ("/dynamic_rom ROM file is missing" showed up in the failing example after "change (iteration 1)" which I never observed locally).

cproc, 2019-06-03:
I could reproduce the problem a few times with base-linux. The last thing I noticed today was that 'ram_fs' did not notify 'fs_rom' about the change, but the problem does not need to be in 'ram_fs'. I'm going to continue tomorrow.

cproc, 2019-06-06:
I can reproduce the symptom reliably with base-linux now with the debug commit 0e47dd8.

It appears to occur when the 'dynamic_rom' test file is created just between fs_rom's failed watch request for the file itself and the successful watch request for the parent directory of the file. Commit 
bffd22f tries to fix this problem by trying to watch a path element again if the watch request for this element failed before and the watch request for its parent directory was successful.

commit 7e6ed169 (2019-06-07): fs_rom: retry watching a path element after watching its parent dir

If a path element cannot be watched because it does not exist yet, the next
available parent directory is watched to detect the creation of the missing
path element. If a missing path element got created just between the failed
watch request for the element and the successful watch request for its parent
directory, the creation of the path element could remain unnoticed.

With this commit, 'fs_rom' tries to watch a path element again if the watch
request for its parent directory was successful.

Fixes #3398
closed 2019-06-13 by chelmuth