genodelabs/genode #3401: ssh: add exec channel

sid-com, 2019-06-04:


m-stein, 2019-06-04:
Cool, that you sticked to this also after the hack'n'hike!
I just scrolled briefly through it - could it be that the file headers need an update? E.g. all headers in the ssh_terminal have the same description "Component providing a Terminal session via SSH" and all of them contain @cnuke as author. Or am I missing something?

sid-com, 2019-06-04:
Thanks @m-stein !
Well, the large ssh_terminal/main.cc file is split up, because it made debugging easier. Most of it is code that @cnuke wrote... What is best practice in this case?

cnuke, 2019-06-05:
Thanks for adding this feature, I appreciate that and will review it.

Normally one adds themself to the description header when “substantial” changes were made but we do not have guidelines for that AFAIK. That being said, I'm fine with how it is. However, providing only a single commit that includes the refactoring work as well as the newly added feature makes reviewing the changes more involved - from that perspective it would have been better to have sperate commits for that. When the changes enter the `master` branch they can be squashed anyway.

sid-com, 2019-06-09:
I see your point. We came up with the same conclusion in our internal review. Let me know if there is anything I can do to make the review less cumbersome.

cnuke, 2019-06-12:
Well, what's done is done - but please keep it in mind for future contributions.

cnuke, 2019-06-12:
@sid-com I reviewed your commit. So far it looks good, both `ssh_terminal` run scripts still work. I rebased your commit to staging and applied a fix we did there to the component (see 7770b0d). 
Please see 4ed3cb7 for changes I made during the review - I mostly reverted cosmetic changes to the code as well as applied our coding style (alignment *space* vs. indentation *TAB*). That being said, it would be nice if you could extend the *README* with regard to the changed behaviour:

- There is now a `exit` report.
- Reports from concurrent logins will override each other and potentially lead to lost
  reports (this happens because there is now only one reporter for all logins - so far
  using multiple logins is not a common use0case so we can address this later but it
  should be documented).
- Detaching a terminal client will lead to a disconnect for the corresponding SSH sessions.
  Although have not used this "misfeature" of replacing a terminal client while keeping the
  session open, it should be documented nonethe less.

sid-com, 2019-06-18:
@cnuke I added your suggestions to the README in commit 4c0e07d

sid-com, 2019-06-19:
@chelmuth and @cnuke while testing we found that vim did not work properly. Fixed at eafd9e41510becce3b8f690d428a9cc3e5111560

chelmuth, 2019-07-09:
I merged the series as 8019518dfea80e8dfbc06fcb3e62b6ca7d982793 just now.

commit 8019518d (2019-07-09): ssh: add exec channel and exit on interactive

This commit implements the ssh exec channel request. It also handles
some shortcommings on the interactive channel like exit and concurrent
session establishments.

Pipes into the channel do not work yet. E.g.:
echo foobar | ssh noux@localhost -p 5555 "cat > /rw/test.txt"

The issue described with FIXME in Ssh::Server::incoming_connection()
could not be reproduced and might have been fixed with the improved
file descriptor handling.

Fixes #3401
commit 68e1e98c (2019-07-12): ssh: add exec channel and exit on interactive

This commit implements the ssh exec channel request. It also handles
some shortcommings on the interactive channel like exit and concurrent
session establishments.

Pipes into the channel do not work yet. E.g.:
echo foobar | ssh noux@localhost -p 5555 "cat > /rw/test.txt"

The issue described with FIXME in Ssh::Server::incoming_connection()
could not be reproduced and might have been fixed with the improved
file descriptor handling.

Fixes #3401
commit eaefcc2c (2019-08-13): ssh: add exec channel and exit on interactive

This commit implements the ssh exec channel request. It also handles
some shortcommings on the interactive channel like exit and concurrent
session establishments.

Pipes into the channel do not work yet. E.g.:
echo foobar | ssh noux@localhost -p 5555 "cat > /rw/test.txt"

The issue described with FIXME in Ssh::Server::incoming_connection()
could not be reproduced and might have been fixed with the improved
file descriptor handling.

Fixes #3401
closed 2019-08-21 by chelmuth