genodelabs/genode #3404: test-dynamic_config_loader foc pbxa9: no log

m-stein, 2019-06-04:
Doing
```
TEST_REPEAT=until_failed TEST_PKGS="test-dynamic_config_loader" \
   TEST_SRCS= KERNEL=foc BOARD=pbxa9 make run/depot_autopilot
```
with current staging fails after a while because the log output of the test runtime doesn't show up sometimes:
```
[init -> depot_autopilot] --- Run "test-dynamic_config_loader" (max 20 sec) ---
[init -> depot_autopilot] 
[init -> depot_autopilot] 
[init -> depot_autopilot] ------- captured log ---------
[init -> depot_autopilot] 
[init -> depot_autopilot] --- matches of log event 0 ---
[init -> depot_autopilot] ------------------------------
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-dynamic_config_loader      failed    19.998  timeout 20 sec
```
So far, I compared the configs of the dynamic/runtime inits for the good and the bad case and they are the same. Also, the init verbose output is the same.

chelmuth, 2019-06-04:
Is it completely stuck also with `TEST_REPEAT=until_forever`?

m-stein, 2019-06-04:
No, the depot_autopilot successfully continues with the other tests (or the same test).

chelmuth, 2019-06-04:
I discovered a potentially related issue in current nightlies with depot_autopilot on qemu.x86_32.pistachio. At the test-libc_vfs step it takes **6 seconds** to the first test-related log message, which forces the test step into a timeout.

rite, 2021-01-13:
I was able to reproduce this issue on base-linux/x86_64 with:
```
TEST_REPEAT=until_failed TEST_PKGS="test-dynamic_config_loader" KERNEL=linux BOARD=linux make run/depot_autopilot
```
...while at the same time maxing out the CPU (for example by starting a few instances of `yes > /dev/null`). It may run for a few iterations but blocks eventually.

Of course I can't expect the test to meet the timing constraints but the test doesn't seem to just take longer but blocks forever while waiting for log messages, even after terminating the `yes`-instances. I verified this by increasing the test timeout from 20 s to 200 s:
```
[init -> depot_autopilot] --- Run "test-dynamic_config_loader" (max 200 sec) ---
Number of tests to run: 1
[init -> depot_autopilot] 
[init -> depot_autopilot] 1.525 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 0 from config
[init -> depot_autopilot] 1.699 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 1 from config
[init -> depot_autopilot] 1.951 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 2 from config
[init -> depot_autopilot] 2.203 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 3 from config
[init -> depot_autopilot] 2.455 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 4 from config
[init -> depot_autopilot] 2.708 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 5 from config
[init -> depot_autopilot] 2.960 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 6 from config
[init -> depot_autopilot] 3.212 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 7 from config
[init -> depot_autopilot] 3.466 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 8 from config
[init -> depot_autopilot] 3.719 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 9 from config
[init -> depot_autopilot] 3.972 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 10 from config
[init -> depot_autopilot] 4.224 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 11 from config
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-dynamic_config_loader      ok         4.224  log
[init -> depot_autopilot] 
[init -> depot_autopilot] --- Run "test-dynamic_config_loader" (max 200 sec) ---
Number of tests to run: 1
[init -> depot_autopilot] 
[init -> depot_autopilot] 1.192 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value -1 from config
[init -> depot_autopilot] 1.346 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 0 from config
[init -> depot_autopilot] 1.598 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 1 from config
[init -> depot_autopilot] 1.851 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 2 from config
[init -> depot_autopilot] 2.104 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 3 from config
[init -> depot_autopilot] 2.355 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 4 from config
[init -> depot_autopilot] 2.608 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 5 from config
[init -> depot_autopilot] 2.860 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 6 from config
[init -> depot_autopilot] 3.113 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 7 from config
[init -> depot_autopilot] 3.366 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 8 from config
[init -> depot_autopilot] 3.618 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 9 from config
[init -> depot_autopilot] 3.872 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 10 from config
[init -> depot_autopilot] 4.124 [init -> loader -> test-dynamic_config_loader ->  -> test-label] obtained counter value 11 from config
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-dynamic_config_loader      ok         4.124  log
[init -> depot_autopilot] 
[init -> depot_autopilot] --- Run "test-dynamic_config_loader" (max 200 sec) ---
[init -> depot_autoNumber of tests to run: 1
pilot] 
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-dynamic_config_loader      failed   199.988  timeout 200 sec
[init -> depot_autopilot] 
[init -> depot_autopilot] --- Finished after 200.054 sec ---
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-dynamic_config_loader      failed   199.988  timeout 200 sec
[init -> depot_autopilot] 
[init -> depot_autopilot] succeeded: 0 failed: 1 skipped: 0
[init -> depot_autopilot] 
[init] child "depot_autopilot" exited with exit value -1
```

rite, 2021-01-26:
> I discovered a potentially related issue in current nightlies with depot_autopilot on qemu.x86_32.pistachio. At the test-libc_vfs step it takes **6 seconds** to the first test-related log message, which forces the test step into a timeout.

@chelmuth fyi, I also executed "test-libc_vfs" the same way as I described above trying to trigger the timeout on  base-linux/x86_64. It didn't happen though, the test run successfully for > 500 min.

m-stein, 2021-02-26:
The dynamic_config_loader nightly test didn't fail on pbxa9+foc for a while now. However, I can still trigger this issue on my own machine with the initially mentioned command line after only a handful of iterations.

m-stein, 2021-02-26:
@nfeske @chelmuth To be honest, I don't have plans yet to dive deeper into this topic in the near future. Should I close the issue?

nfeske, 2021-02-26:
Yes.

closed 2021-02-26 by nfeske