genodelabs/genode #3405: hw: support for ARM64

skalk, 2019-06-04:


skalk, 2019-06-04:
Current state comes to the point where the kernel wants to proceed the first time with the first core-thread.

skalk, 2019-06-11:
The fundamental things are running on Qemu/Rpi3, therefore I mark this issue as done.

nfeske, 2019-06-12:
Fantastic!

skalk, 2019-06-12:
Now, it also works on Raspberry Pi 3 hardware.

m-stein, 2019-06-13:
:-)

nfeske, 2019-06-13:
I have to express my deep respect for @skalk's work here again. The code is simply a pleasure to read and a poster child of Genode's API building blocks at work. It is so carefully crafted that it has a certain text-book quality to it. I also appreciate the separate commit for the interface change preceding the actual 64-bit commit. It is nice to see interface changes decoupled from large additions.

Thanks Stefan for your great care!

m-stein, 2019-06-13:
Thanks also, the code is indeed enjoyable to read! I wonder whether it would be a desirable to have an back-end for the ``Register_set`` template that targets ARM co-processor registers? These registers are widely used and currently, the most convenient way to access them are macros, a new back end would eliminate the need for macros.

skalk, 2019-06-13:
Thank you for your nice compliment!

commit dbaee9e7 (2019-06-13): hw: support for ARM64 Raspberry Pi 3

Restriction: enables only cpu core 0 and the timer interrupt by now.

Fix #3405
commit 90d07741 (2019-07-09): hw: support for ARM64 Raspberry Pi 3

Restriction: enables only cpu core 0 and the timer interrupt by now.

Fix #3405
closed 2019-07-09 by chelmuth