genodelabs/genode #3409: vbox: page fault on shutdown

nfeske, 2019-06-05:
In Sculpt on staging, I observe the following page fault when shutting down a VM:

```
[core] page fault, pd='init -> runtime -> vm -> vbox' thread='EMT-0' cpu=0 ip=0x204e6d3 address=0x0 stack pointer=0xa02fef18 qualifiers=0x4 irUwp reason=1[0m
```

The IP points to the VFS, specifically at the `Fifo::remove` method (https://github.com/genodelabs/genode/blob/staging/repos/base/include/util/fifo.h#L100), which seems to be called from `Fs_file_system::close` (https://github.com/genodelabs/genode/blob/staging/repos/os/src/lib/vfs/fs_file_system.h#L856).

@ehmry Do you have an idea how we could end up with the nullptr-dereference here?

ehmry, 2019-06-06:
The `Fs_file_system` plugin looks fine to me, but I suspect there is a race condition within the libc when files are synced on close, I'm not in a position to reproduce this error right now, but I have patch at 24845a650cc4c5510c6a18feb4c497caa8d38013 to the libc that should be applied regardless. It seems that having a split implementation between a *.h and *.cc file caused a incomplete locking refactor.

nfeske, 2019-06-06:
Thanks Emery for the prompt investigation! Looks good so far. I applied your patch, shut down the VM repeatedly three times, and the problem did not occur. :-) I merged your commit to staging.

nfeske, 2019-07-09:
The patch definitely improves the situation. It ended up at the master branch already. I removed the fixed label nevertheless because the page-fault issue still occurs sometimes.

nfeske, 2020-10-09:
This issue is solved with the libc update of Genode 20.08.

closed 2020-10-09 by nfeske