genodelabs/genode #3411: depot_autopilot.run: maxiumum uptimes

m-stein, 2019-06-07:
Some platforms cannot manage to execute all tests in a single boot. For instance sel4 on x86_32. Arbitrary tests always fail because the platform got stuck and has to be re-booted. But the individual tests aren't the problem in this case.

m-stein, 2019-06-07:
@chelmuth I was unsure about which platforms to limit and to which time because I couldn't find the reboots we talked about in the most recent nightly. Thus, I added an example limit for sel4 according to its typical total autopilot duration on x86_32 (hardware). Please adjust according to your needs:
 
59b4b691c6 depot_autopilot.run: maximum uptimes

m-stein, 2019-06-07:
@chelmuth I noticed that there is still a problem with the test-timeout calculation in the run-script. For instance, for test-tcp_bulk_lxip qemu/x86_64.sel4 it's a very close thing:
```
[init -> depot_autopilot] 26.609 [init -> recv] [31mError: freeing non-empty slab block[0m[0m[0m
[2019-06-07 04:48:46] 
[init -> depot_autopilot] 26.612 [init -> recv] [34mWarning: 14 dangling allocations at allocator destruction time[0m[0m[0m
[2019-06-07 04:48:46] 
[init -> depot_autopilot] 26.677 [init] child "recv" exited with exit value 0[0m[0m[0m
[2019-06-07 04:48:46] 
[init -> depot_autopilot] [0m[0m
[2019-06-07 04:48:46] 
[init -> depot_autopilot]  test-tcp_bulk_lxip              ok        26.677  log "child "recv" exited with exit va ..."[0m[0m
[2019-06-07 04:48:47] 
[init -> depot_autopilot] [0m[0m
[2019-06-07 04:48:47] 
[init -> depot_autError: Test execution timed out
```
Thus, I relaxed the timeouts further:
cefdedd56e depot_autopilot.run: relax test timeout

chelmuth, 2019-06-07:
The test were imx7d and imx6q with sel4.

chelmuth, 2019-06-11:
It seems the time-based approach collides with the maximum runtime of *test-init*, which is 330 sec for slow platforms. Could we reconsider the *reboot-after-n-tests* approach?

chelmuth, 2019-06-11:
Also, I wonder if we still need the *NOT FOR MASTER* commits because rebasing them may become a hassle on next master.

m-stein, 2019-06-11:
Sorry for the inconvenience. I have created a new version of the depot-autopilot commit. Regarding the "NOT FOR MASTER" commits: I'd like to reach 0 fails in the depot autopilot before removing them. I have reverted them before the depot-autopilot commit and added them again afterwards:

7ecd4c24c3 NOT FOR MASTER "depot_autopilot.run: raise depot autopilot quota"
f38eca9224 NOT FOR MASTER "depot_autopilot: list log matches"
d8d05d4c92 NOT FOR MASTER "depot_autopilot: hex dump of failed test log"
5e5352ca1d depot_autopilot.run: relax test timeout
6109b834de depot_autopilot.run: max nr of tests per boot

chelmuth, 2019-06-11:
Thanks, merged to staging :crossed_fingers:

commit bfff6b1d (2019-06-07): depot_autopilot.run: maximum uptimes

Some platforms cannot manage to execute all tests in a single boot. Thus, we
re-boot them periodically to avoid that arbitrary tests fail due to the long
uptime and not to the tests themselves.

If the maximum uptime is set to 0, no uptime limit is applied.

Fixes #3411
closed 2019-06-13 by chelmuth