genodelabs/genode #3412: cpu_bench: race condition in handling results in run tool

tomga, 2019-06-09:
In ``cpu_bench.run`` there is:
```
proc run_test {name serial_id} {
	run_genode_until "start $name.*\n"    20  $serial_id    <-- <3>
	set t1  [clock milliseconds]
	run_genode_until "finished $name.*\n" 200 $serial_id
	set t2  [clock milliseconds]
	return [expr {$t2 - $t1}]
}

run_genode_until "Cpu testsuite started.*\n" 60                 <-- <1>
set serial_id       [output_spawn_id]
set bogomips        [run_test "bogomips" $serial_id]            <-- <2>
```
In app code there is:
```
	Genode::log("Cpu testsuite started");
	Genode::log("start bogomips");
```
and when after going throug ``<1>`` and ``<2>`` script code gets to ``<3>``  second log (``start bogomips``) is somehow already processed and test times out waiting for it. Adding sleep between those ``Genode::log`` messages makes test passing but I it's only a workaround. I don't know how to properly fix this issue or even if it is fixable. I suspect that error is somewhere in ``run`` tool in ``wait_for_output``.

skalk, 2019-06-11:
@tomga could you test-wise change line `<1>` into `run_genode_until "Cpu testsuite started." 60`? Under what circumstances are you trigger this behaviour (host expect version, target platform == rpi3)?

tomga, 2019-06-11:
The change you asked to test helps.
expect version is 5.45.4 - generally ubuntu 18.04 LTS
platform is rpi3 - my branch rebased on latest genode release - not published yet
I also wanted to check on rpi1 (I purchased one to test if I don't break anything with my changes for rpi3) but it doesn't build there:
```
...
error copying "bin/hw_timer_drv": no such file or directory
    while executing
"file copy -force bin/[kernel_specific_binary $binary] [run_dir]/genode/$binary"
    (procedure "copy_genode_binaries_to_run_dir" line 4)
    invoked from within
...
```
It seems that it is not because of my local changes. Build fails for me also on clean 19.05. I can try to find out later what is the problem but maybe you will just know.

tomga, 2019-06-11:
Ok. The build issue is not specific for rpi but I just had luck with rpi3 because earlier I built something that build timer. Folowing trivial patch helps:
```--- a/repos/os/run/cpu_bench.run
+++ b/repos/os/run/cpu_bench.run
@@ -5,7 +5,7 @@ if { [get_cmd_switch --autopilot] } {
        }
 }
 
-build "core init test/cpu_bench"
+build "core init timer test/cpu_bench"
 
 create_boot_directory
```

For both rpi1 and rpi3 test consistently times out. For both each of workarounds work (sleep I mentioned in initial report and removing .*\n from ``<1>`` but I don't really understand that. I could explain myself why each of workarounds help but not both.

tomga, 2019-06-11:
When running on linux it is also reproducible for me but not every time. I think in about 50% of attempts.

skalk, 2019-06-12:
Ok, the timer was not build, I'll fix that.
But what about Linux? The test fails even with the small change I advised earlier, which fixed it for rpi3?

tomga, 2019-06-12:
I wasn't precise with information about linux - your proposed fix curates the problem.

I couldn't understand why this change helps until I started thinking about that in your proposed fix you left dot at end of pattern passed to ``run_genode_until``. Normally in regular expressions dot means any character but not a newline and I thought it is that way in expect too. Fact that your proposal works means that dot matches newline here and now it is obvious for me why this change helps.

But then all other places with ``.*\n`` passed to ``run_genode_untli`` in run scripts (more than 90) are vulnerable to this race condition. Am I right?

skalk, 2019-06-12:
> I couldn't understand why this change helps until I started thinking about that in your proposed fix you > left dot at end of pattern passed to run_genode_until. Normally in regular expressions dot [...]

Actually, I left the dot accidentally, and I assume it will work without the dot. Obviously, the way how many characters will be consumed by expect is affected by the `.*\n` expression here.

> But then all other places with .*\n passed to run_genode_untli in run scripts (more than 90) are vulnerable to this race condition. Am I right?

Not necessarily. Most run-scripts do not use expect repeatedly. Therefore, I do not think that we need to change all run-scripts. Moreover, until now we did not hit this problem with various tests although we test quite different hardware using different host Linux systems for expect.

tomga, 2019-06-12:
> Actually, I left the dot accidentally,
I suspected that but it helped me to understand what is really happening and why both changes work.

> and I assume it will work without the dot.
Yes, it works. I tested it yesterday.

> Obviously, the way how many characters will be consumed by expect is affected by the `.*\n` expression here.
Actually this consumes all available input and then waits for newline - at least such behaviour explains symptoms.

> > But then all other places with .*\n passed to run_genode_untli in run scripts (more than 90) are vulnerable to this race condition. Am I right?
> Not necessarily. Most run-scripts do not use expect repeatedly. Therefore, I do not think that we need to change all run-scripts.
Ok.
> Moreover, until now we did not hit this problem with various tests although we test quite different hardware using different host Linux systems for expect.
Well, it's just a timing issue. If you run scripts from slower machine probably you'll probably be hit by this more often.
Leaving such constructs is misleading and this pattern will probably be copied to new tests. That's why I think it would be better to fix it, but I have no problem with your decision either. For me the puzzle is solved and I will remember the problem when I hit it again :-).

tomga, 2019-06-12:
@skalk, in case you haven't already pushed a fix my is at 98ae9640becdc99623f22b7c3b95075c1a944f51.

skalk, 2019-06-13:
@tomga thanks for the commit!

skalk, 2019-06-13:
@tomga I'll re-open it, and mark it as fixed. Thereby, our staging branch holders @nfeske and @chelmuth can see that there is a solution that still needs to be applied to staging.

commit 585494a0 (2019-07-09): cpu_bench: fixes to make it work reliably

Added missing build dependency.

Eliminated race condition when processing test output.

Fixes #3412
commit 7579034a (2019-07-12): cpu_bench: fixes to make it work reliably

Added missing build dependency.

Eliminated race condition when processing test output.

Fixes #3412
commit 4189157d (2019-08-13): cpu_bench: fixes to make it work reliably

Added missing build dependency.

Eliminated race condition when processing test output.

Fixes #3412
closed 2019-08-21 by chelmuth