genodelabs/genode #3414: test-read_only_rom: false positive test result on base-linux

cproc, 2019-06-12:
On base-linux the test log shows errors which are not caused by writing to a read-only ROM dataspace:

```
[init -> depot_autopilot] --- Run "test-read_only_rom" (max 20 sec) ---
[init -> depot_autopilot] 
[init -> depot_autopilot] 0.212 [init -> vfs] session opened for 'cached_fs_rom -> ' at '/'
[init -> depot_autopilot] 0.222 [init -> cached_fs_rom] Error: Region_map_mmap::attach: attaching w/o local addr not supported
[init -> depot_autopilot] 0.222 [init -> cached_fs_rom] Error: Uncaught exception of type 'Genode::Region_map::Region_conflict'
[init -> depot_autopilot] 0.223 [init -> cached_fs_rom] Warning: abort called - thread: ep
[init -> depot_autopilot] 
[init -> depot_autopilot]  test-read_only_rom              ok        20.000  timeout 20 sec
```

m-stein, 2019-06-13:
As a side note: I'm not happy with the success condition of this test yet. It is considered successful when running 20 seconds without printing a given message. It marks an exception amongst all depot autopilot test as it's the only one with such a weak condition. Maybe we could add a runtime component that acts as pager for the test and prints access type violations. Then the depot autopilot could evaluate the test in a more sensible way.

nfeske, 2021-02-26:
This test should be disabled on base-linux.

commit c4f75f49 (2024-04-12): depot_autopilot: skip test-read_only_rom on linux

Fixes #3414
closed 2024-04-19 by chelmuth