genodelabs/genode #3415: support for rpi2 and rpi3 (32bit)

tomga, 2019-06-13:
[rpi_all_19.05](/tomga/genode/commits/rpi_all_19.05) is my branch where I pushed preliminarily polished changes to make 32 bit support for rpi2 and rpi3 (and some other changes). As I'd like my patches to be accepted upstream I'd need some answers and advises what to do before I go further with it. 

Early boot

8c4268b098360f954b97f46ee54bd22da7d82597 - this is required to skip executing ``cps`` commands as they are not allowed in hyp mode and rpi3 is booted in this mode by firmware code (at least for 32bits kernel). I tested it on rpi1 and it doesn't break it. Can it be accepted in such form or would you suggest some other way to achieve the same goal?

Timer

f29cfec9f527283ac0757d8e02683ee995af0b02 and 33f382e24b022392d6144176d3ce49111f1dd57f (only [timer.cc](/tomga/genode/blob/rpi_all_19.05/repos/base-hw/src/core/spec/rpi3bplus/timer.cc) and [timer_driver.h](/tomga/genode/blob/rpi_all_19.05/repos/base-hw/src/core/spec/rpi3bplus/timer_driver.h) ) - implementation is based on informations from ARM Cortex-A53 MPCore Processor TRM and is not specific for rpi. Would you like to have it in ``genode`` repository as cortex_a53 timer or should I move it to ``genode-world`` as we earlier decided with my code supporting 32bit rpi?

PIC

33f382e24b022392d6144176d3ce49111f1dd57f (only [pic.h](/tomga/genode/blob/rpi_all_19.05/repos/base-hw/src/core/spec/rpi3bplus/pic.h) and [pic.cc](/tomga/genode/blob/rpi_all_19.05/repos/base-hw/src/core/spec/rpi3bplus/pic.cc) ) - this probably can be shared between 32bit and 64bit. @skalk I haven't had time to look at your implementation yet but I've seen your comment that currently one core is enabled. In this implementation IPI is implemented and smp tests are working. So do you think I should move this to ``genode-world`` or leave it in ``genode`` repository?

Rest of board definitions

33f382e24b022392d6144176d3ce49111f1dd57f (without timer and PIC) - I plan to move this to ``genode-world`` with rename from ``rpi3bplus`` to ``rpi3bplus_32`` (or maybe it would be better to have ``rpi2`` or ``rpi3_32``?)

Framebuffer

354b67945d3775761e27c938b3304223d7fffc94 - test is currently not building after rebase to latest release but I think that those changes will not break existing rpi1 support. They are only changes to existing driver therefore I think they should stay in ``genode`` repository. When I make it work again I'll let you know.

USB

4a68a09ff7fbcc7ff3cd4fee85939fc8a15b5d30 - port of dwc2 usb host driver. I think that ``rpi3bplus`` should be renamed to something more generic. Just ``rpi`` or maybe ``dwc2``?

Bus address for USB

e4250547c8c8fc3a03182d9a2864776cbdf132f7 - required to make usb work on rpi3 and probably it is ok on other models but definitely something like this is not appropriate for other arm boards. I don't really know how to do it properly. Probably adding call to get bus address for dataspace would be in my range but it would be against #2243. Adding this to platform drv after implementing #3299 would probably take me not less than a month (given limited time, lack of experience in this area and task complexity). Any suggestions how can we proceed with this?

Ethernet

1efa2417d6a7928015780fb1c79425c101016dd2 - port of lan78xx driver. @ssumpf I remember you saying during Hack'n'Hike that I definitely shouldn't need something that I added during my porting work. Could you please check in (usb_net.list)(/tomga/genode/commit/1efa2417d6a7928015780fb1c79425c101016dd2#diff-2fd872a906132230dcc240f0c1abe175) and remember what it is?. Was it about `` linux-x.x.x/drivers/net/phy/mdio_bus.c``?

Sorry for long list of things but given that I don't know your decisions how to proceed with it further I didn't know how to split it into multiple issues (creating one for each part now would make it to be too fragmented, I think).

Also I'd like you to maybe reconsider accepting ``rpi3_32`` support in ``genode`` repository. Majority of above changes will not be moved to ``genode-world`` repository anyway and given that device you have can be used to test this mode maybe it would be better? Definitely splitting support for one device working in two different modes is strange to me and definitely it would be easier for me without this split.

nfeske, 2019-06-14:
Thanks @tomga for the very interesting summary of your work. It is very impressive! I'll leave most of your code-related questions to others who are more qualified than me (@skalk @ssumpf @m-stein).

Regarding the distinction between 32 and 64 bit, I think it would sensible to take one choice for a given board. E.g., always using 32 bit on Rpi1 and 64 bit on Rpi3. This limits the variety of configurations to test and support.

I like your suggestion to call the new usb driver `dwc2_usb_drv` instead of tying it to a particular board.

Doing the bus-address calculation locally in the USB driver is appropriate as a shortcut for the platform-driver topic. It would be a pity to put the platform driver on the critical path for including your work. If moving the conversion to a dedicated function with two implementations (one for the dwc2 that performs the conversion and one for all other platforms that just returns the input value) this would look clean to me. Once the platform-driver infrastructure is in place, it will be trivial to adjust the code in the USB driver.

skalk, 2019-06-14:
@tomga first and foremost: thank you very much for your great contributions. It is impressive to see how much effort you spent to support the Raspberry Pi ecosystems.

Please understand that currently we are quite busy to settle the changes necessary for ARM v8 and incorporating the new platforms rpi3 64-bit and i.MX8, which will produce unfortunately some friction in between our different patch series. I try to answer as much as I can...

> Early boot
> 
> 8c4268b - this is required to skip executing cps commands as they are not allowed in hyp mode and rpi3 is booted in this mode by firmware code (at least for 32bits kernel). I tested it on rpi1 and it doesn't break it. Can it be accepted in such form or would you suggest some other way to achieve the same goal?

I would actually prefer to keep the assembler entry path as minimal and generic as possible. Alternatively, I could imagine having a sub-assembler routine using a global symbol, which prepares the processor for the right privilege level. In that case, you could do nothing and the current common case switches to system mode (0x1f).

> 
> Timer
> 
> f29cfec and 33f382e (only timer.cc and timer_driver.h ) - implementation is based on informations from ARM Cortex-A53 MPCore Processor TRM and is not specific for rpi. Would you like to have it in genode repository as cortex_a53 timer or should I move it to genode-world as we earlier decided with my code supporting 32bit rpi?

For the time-being, just move it to the genode-world path. In general, we already make use of the Generic Timer defined since ARM v7. It is used for i.MX7 Dual SABRE, now by Raspberry Pi 3 (64-bit), and in upcoming commits by i.MX8. But until now, there are slightly differences, I'd like to factor out when integrating i.MX8. That's why they reside in the board specific directories until now.
Moreover, I've used the system registers for programming and reading the physical counter instead of using memory-mapped I/O. When unifying the implementations is done. We might migrate the rpi3 32-bit version to use the unified version?

> PIC
> 
> 33f382e (only pic.h and pic.cc ) - this probably can be shared between 32bit and 64bit. @skalk I haven't had time to look at your implementation yet but I've seen your comment that currently one core is enabled. In this implementation IPI is implemented and smp tests are working. So do you think I should move this to genode-world or leave it in genode repository?
> 

I'll have a look at it, when time permits it, and you can put it to world. Alternatively, if you like you might test-out the rpi3 64-bit version, and try to incorporate your PIC implementation using core0 only. If it works with one core, we can take it upstream right now.

> 
> Rest of board definitions
> 
> 33f382e (without timer and PIC) - I plan to move this to genode-world with rename from rpi3bplus to rpi3bplus_32 (or maybe it would be better to have rpi2 or rpi3_32?)

I expect them to be equal to the 64bit version. Could you have a look in `repos/base-hw/src/include/hw/spec/arm_64/rpi3_board.h`. I think it would be best to add your additional definitions there.

> 
> USB
> 
> 4a68a09 - port of dwc2 usb host driver. I think that rpi3bplus should be renamed to something more generic. Just rpi or maybe dwc2?

I think as long as the resource names (mmio range, irq number) are hardcoded inside the driver, it should be named rpi3. If you change it to some configuratiuon value dwc2 would be perfectly fine.

> 
> Bus address for USB
> 
> e425054 - required to make usb work on rpi3 and probably it is ok on other models but definitely something like this is not appropriate for other arm boards. I don't really know how to do it properly. Probably adding call to get bus address for dataspace would be in my range but it would be against #2243. Adding this to platform drv after implementing #3299 would probably take me not less than a month (given limited time, lack of experience in this area and task complexity). Any suggestions how can we proceed with this?

For the time being, I would advise to factor this out in a separate function which XOR the different bus address in the rpi case, but not elsewhere. Later (within the next months), when the platform_drv is in place, we can fix this.

> 
> Sorry for long list of things but given that I don't know your decisions how to proceed with it further I didn't know how to split it into multiple issues (creating one for each part now would make it to be too fragmented, I think).

You do not have to apologize. On the contrary, we are pleased by your contributions. Regarding the further process, I think it would be good to open an issue per change of existing components in genode (given above quotes), and one-by-one getting them upstream. The part that add rpi2 and rpi3_32 into world, can be done in one go. The stuff in genode-world is getting compiled by genode labs (if it is targeted by packages), but not tested. Here, the contributor and all users are responsible for.
> Also I'd like you to maybe reconsider accepting rpi3_32 support in genode repository. Majority of above changes will not be moved to genode-world repository anyway and given that device you have can [...]

As, already explained in the past, we will not take 32-bit support for a 64-bit ARM platform into the genode repository. As @nfeske already wrote, to us the rpi3 will be maintained as 64-bit variant. We can not effort to support all variations possible. With support, I mean maintaining the code-base and all automatic test run night by night. We decided to not support legacy 32-bit x86 mode in our hw kernel, and we do not plan to support legacy 32-bit mode for 64-bit ARM platforms.
You are free to use the 64-bit version, and if you like you can incorporate your 32-bit variant into genode-world, which has less restrictions, but is not granted to be tested by us.
I hope you can understand this.

All the best!

tomga, 2019-06-14:
@nfeske, @skalk, thanks for answers. 

@skalk, the part about assembly is not clear. Either I don't understand what you mean or you misunderstood what this patch does. Anyway I'd need clarification.

> I would actually prefer to keep the assembler entry path as minimal and generic as possible. 
Understandable. The only thing this patch does is: check if currntly in hyp mode and skip ``cps`` commands because they cause hang on rpi3 in 32bit mode (probably some exception that is not handled).

> Alternatively, I could imagine having a sub-assembler routine using a global symbol, which prepares the processor for the right privilege level. In that case, you could do nothing and the current common case switches to system mode (0x1f).
If I correctly understand this subassembler routine would be called somewhere in the beginning (before the part I touched) and would have to switch from hyp mode to supervisor mode. But then I don't get what you mean by "do nothing".

Could you please explain in more detail this?

skalk, 2019-06-17:
What I meant was, exactly at the point where now the `cps` instruction stands to call a sub-routine like:
```
bl __crt0_switch_to_system_mode
```
In the common case in this routine nothing more than the `cps` instruction is done, whereby in your case you do not have to do anything, because `cps` is not valid in hyp mode.

But maybe we can postpone this assembler entry issue a bit, because I would like to proof first whether we can get rid of these cps instructions here at all. In the past it got used to set the stack-pointer Thereby, you wouldn't need to change anything. Nevertheless, I do not have the time right now to check this, so as an interim solution you could make a copy of the crt0.s file with your change for the time being, and getting rid of it as soon as the `cps` instructions vanished in the original file.

tomga, 2019-06-17:
Thanks for explanation. Everything is clear now.

I considered something like this but I thought that to call subroutine I need stack which is set up later.

tomga, 2021-12-26:
Almost three years passed since last activity in this issue but this did not mean that I completely disregarded idea for support for Raspberry Pi boards on Genode. When time permitted I tried to update my working branch to current Genode state and make some progress with functionality. It is a time when I'd like to make an attempt to merge my work.

### Current state

[rpi_master_20211217](url/tomga/genode/tree/rpi_master_20211217) is my current working branch and it is based on current master. 

Changes on branch include extending support for rpi and rpi3 and adding support for rpi2 and rpi3_32. 

Besides some basic tests I managed to have following scenarios working on all configurations:

 * run/demo - uses `drivers_interactive` package that integrates framebuffer driver and usb_host driver using new platform driver.

 * run/ping - uses `drivers_nic` package that uses usb based network cards available on rpi devices.

### Commits

Commits are splitted in two *dimensions*. Common parts and parts specific for different rpi models are committed separately to allow selective merge (I know that parts that are specific for rpi2 and rpi3_32 will not be merged - at least not until repository specific for rpi will be created). Changes that touch different functional parts are also committed separately to make it easy to review.

 * early boot - 6f6784cd05afcfe70bad3632e54dc723eed9a5db did not change at all since previous discussion. It is required for rpi2 and rpi3_32 only so given that currently there is no support for those models it can wait until some decision is made.

 * hw (interrupts and timer) - d5d60ce59845b916424b31b4b35da839e417a99a mostly extends interrupt controller for rpi3 to *cascade* to bcm2835 for all but most basic interrupts.

 * new platform driver - changes are splitted into parts for easier review (note: some changes introduced in first commit are later replaced with code that retrieves data from configuration; it was implemented before new platform driver even existed and I decided to leave it like it is for review; it will be squashed before final commit anyway). Most notable changes are: mbox addresses are taken from configuration (without support for reconfiguration) and `Rpi_mbox` rpc service is introduced to allow framebuffer driver to use this platform driver.

 * legacy platform driver - changes needed for framebuffer before it is converted to new platform driver. There are also changes to allow it to build on arm64. I left those changes but probably not needed for merge as that driver will be removed soon.

 * framebuffer driver - migration to new platform driver and changes to build on arm64 (probably changes platform driver changes along with framebuffer changes can be discussed in #4303 if you prefer). There is also a workaround for #4331 that allows framebuffer to properly work on rpi3.

 * usb_host driver - changes needed for build on arm64 and those that allow to install `phys_to_bus` dma address conversion function. Currently it depends on configuration directly passed to the driver (`rpi_version` attribute). I've seen some code that I think should allow to receive this information from the platform driver but couldn't find any example for it so I left it like it is for now. Currently it works and I think this change may be implemented some time later.

 * usb_net driver - added lan78xx support required for rpi3.

 * drivers_interactive and drivers_nic packages - separately for each board so they can be committed separately.

Rest of commits contain changes to different run scripts that allow me to run scenarios directly on hardware. If there is a will to merge them I can curate them from rpi2 and rpi3_32 but maybe it would be better to change them to use some configuration from files instead of introducing conditions for different boards.

NOTE: I've added a quite ugly `REQUIRES` clauses to programs that I needed to be built on `arm` and `arm_64` and only on those platforms. I did not found a cleaner local solution for it.

skalk, 2022-01-03:
@tomga I've prepared an alternative solution to prevent the 'CPS' instruction. Could you please try out commit d6ddd2e while starting rpi2 or rpi3 in 32-bit mode?

tomga, 2022-01-03:
@skalk, I've tested your alternative solution to 'CPS' instruction and it allows to boot without problems on rpi2 and rpi3 in 32-bit mode.

I created new branch [rpi_master_20211217a](/tomga/genode/commits/rpi_master_20211217a) that is based on your `merge_to_staging`. I have tested this branch and `run/demo`, `run/ping` and `run/framebuffer` work on all models.

Next set of changesets affect platform driver. When I looked at those commits today I came to conclusion that I could have split them better for review and I even wanted to do that today but I decided that rebase will be more valuable today. I can reorganize those commits (probably tomorrow) into smaller parts and easier for review.

I believe that after we agree on changes that affect the platform driver (mostly Rpi specific) other parts will go quite easy.

skalk, 2022-01-04:
@tomga thank you very much for testing the commit!

skalk, 2022-01-10:
@nfeske @chelmuth would you mind to review and merge commits a305e53 and e6a1a99 as first steps to solve tomga's issue? They are ready for staging in my eyes.

nfeske, 2022-01-10:
I merged both commits to staging.

skalk, 2022-02-15:
@tomga: in my eyes all relevant commits are now applied or moved to the new genode-rpi repository. The only thing left are board support / depot recipes for rpi2, and respectively rpi3 (32-bit) if you still like to have it. Both can be handled as separate issues within the genode-rpi repository itself. Therefore I would like to close this issue here, if you agree.

skalk, 2022-03-28:
All rpi related issues should be discussed in the genode-rpi repository now. Therefore, I'll close this issue.

closed 2022-03-28 by skalk