genodelabs/genode #3417: autopilot: support architecture-specific build directories

chelmuth, 2019-06-14:
Switch from deprecated board-specific directories to arch-specific. Related to #3316.

chelmuth, 2019-06-14:
Please review f0d859a126e4f494a512a3c7a1f1ded1e88baffa.

chelmuth, 2019-06-14:
Log of an example run.
```
> ./tool/autopilot --force -d /tmp/copilot/qemu -i infofile.log -p arm_v7a-pbxa9 -k hw -t arm_v8a-rpi3-hw -r log -j4
genode dir    : /plain/krishna/src/genode/genode_staging.git
targets       : arm_v7a-pbxa9-hw arm_v8a-rpi3-hw
architectures : arm_v7a arm_v8a
run scripts   : log
test dir      : /tmp/copilot/qemu
make -j       : 4

--- architecture arm_v7a ---
pbxa9 hw:                      log                    -> OK (1:46)
pbxa9 hw:                      cleanall               -> DONE
-> ERROR
      unexpected: ./bin
      unexpected: ./bin/init.xsd

--- architecture arm_v8a ---
rpi3 hw:                       log                    -> OK (1:56)
rpi3 hw:                       cleanall               -> DONE
-> ERROR
      unexpected: ./bin
      unexpected: ./bin/init.xsd
--- done (errors occurred) ---
```

nfeske, 2019-06-17:
That's very cool! I like that both `-t` and the combination of `-k` and `-p` are possible. That's very flexible yet still convenient.

skalk, 2019-06-17:
@chelmuth thanks for completing the work of issue #3316 in such an elegant way!

commit 58243b90 (2019-06-17): autopilot: support arch-specific build directories

Replace the notion of board-specific platforms by clear definition
architecture and board.

Also added new command line switch for "targets"

  -t <target>         test target as triple of architecture-board-kernel,
                      e.g., arm_v7a-pbxa9-hw

Related to #3316
Fixes #3417
closed 2019-07-09 by chelmuth