genodelabs/genode #3418: tool chain: TLS support

cproc, 2019-06-15:
The Genode 'signal' test occasionally fails with an 'uncaught exception' error for an exception which is actually caught when looking at the Genode source code. It turned out that the exception handling code in libsupc++ (eh_globals.cc) uses an 'eh_globals' object which is supposed to be thread-local, but has currently only a single instance because of missing TLS support.

There are two alternative code paths in libsupc++ for thread-local access of the 'eh_globals' object. One path uses the '__thread' keyword and the other path uses the 'gthreads' interface, which is similar to the 'pthreads' interface and currently not implemented with multi-threading support for Genode. The '__thread' keyword (and the 'thread_local' keyword since C++11) is implemented either with hardware support (for example by using the 'fs' register on x86) or "emulated" by libgcc using the 'gthreads' interface if hardware TLS support is unavailable or disabled (which is currently the case with the Genode tool chain, because using the 'fs' register is not supported by some kernels).

I took a closer look at the TLS emulation code (emutls.c) and it basically consists of an implementation of the '__emutls_get_address()' function, which is called whenever a thread-local variable is accessed and returns the address of the variable instance for the current thread. The function receives a pointer to an 'emutls object', which was generated by the compiler for each thread-local variable and describes the properties (size, alignment and initial content) of the variable to be able to create the thread-local variable instances.

I implemented this function in the 'cxx' library and enabled TLS support in libsupc++ and this appears to fix the problem with the 'signal' test and should also make it possible to use the '__thread' and 'thread_local' keywords in general. The implementation is mostly separate from the 'Genode::Thread' class (except for freeing the TLS data when a thread is destroyed), so it also works for the main thread.

nfeske, 2019-06-18:
Thank you for this excellent work and the very informative issue description.

During an offline discussion, @skalk raised the concern that the added locking _may_ require special handling for the base-hw kernel. If the kernel uses exceptions internally but the exception handling acquires a `Genode::Lock`, we may need to override the default implementation for base-hw.

chelmuth, 2019-09-24:
Testing impacts on base-hw is still pending because genode-toolchain-19.05 does not use our emutls implementation for thread-safe exceptions. So, we have to schedule building a current tool chain for testing.

chelmuth, 2019-12-20:
TLS works on master.

commit ab017607 (2019-09-02): tool chain: TLS support

Fixes #3418
closed 2019-12-20 by chelmuth