genodelabs/genode #3421: Portable shebangs

ehmry, 2019-06-17:
The the use of the `#!/usr/bin/env CMD` shebang rather than `#!/usr/bin/CMD` is portable across Unixes without a "standard" `/usr/bin`.

ehmry, 2019-06-17:
Patch at f027698e341949c8f89520eac838b8ea38c73152.

nfeske, 2019-06-17:
Out of curiosity, which non-Linux Unix system motivated you to create the patch?

Does this change introduce another call of execve on each execution of such a tool?

Have you tested the change with the noux-tool-chain test? The test calls those tools (e.g., check_abi) from a noux environment. I ask because - as far as I know - the noux instances used in Sculpt don't have /usr/bin/env available.

cnuke, 2019-06-17:
As a side note, `env -S` was first introduced in coreutils v8.30 - which is AFAICT not available on Ubuntu 16.04 and Debian 9.

ehmry, 2019-06-17:
Patch dd5ef8b4ebe996eb9aafd9ad61b6430ff3209a8a updates coreutils to v8.31, patch 35f9f721463ac3f8e800c9060415cd149a3deb75 adds `env` to coreutils-minimal.

Does it add another execve? Probably.

Tested with `#!/usr/bin/env -S make -f` in Noux.

nfeske, 2019-06-17:
According to @cnuke's comment, this change rules out the use of Ubuntu 16.04 LTS for Genode development. I feel uneasy about that.

ehmry, 2019-06-17:
Documentation says use a current version of Ubuntu LTS. Ubuntu 16.04 is two years behind that.
  https://github.com/genodelabs/genode/blob/3c4c460f8256297659f42df5a2281dad93101fe9/doc/tool_chain.txt#L17

Ref #2047

cnuke, 2019-06-17:
I mentioned Ubuntu 16.04 LTS because that is still the version running on our nightly CI machine. That being said, apparently even 18.04 LTS just provides v8.28 (I haven't checked any ppa yet - please correct me if I'm wrong).

I understand that using the tools as is is inconvenient for you, now that you are back on NixOS. Having `-S` available in recent coreutils versions certainly addresses my remake from #2407 but for now it seems that we are still stuck with the older versions.

A workable workaround certainly could be something along the lines of `alias M='/foo/bar/make -f'` and invoking the tools like `M tool/depot/create …` (that is if all tools make internally use of *$(MAKE)* and having such an alias does not break anything else). At least when using FreeBSD to build Genode calling `gmake -f …` worked fine.

chelmuth, 2019-06-18:
> According to @cnuke's comment, this change rules out the use of Ubuntu 16.04 LTS for Genode development. I feel uneasy about that.

I'm also still using 16.04 LTS...

chelmuth, 2019-06-18:
> Patch [dd5ef8b](https://github.com/genodelabs/genode/commit/dd5ef8b4ebe996eb9aafd9ad61b6430ff3209a8a) updates coreutils to v8.31, patch [35f9f72](https://github.com/genodelabs/genode/commit/35f9f721463ac3f8e800c9060415cd149a3deb75) adds `env` to coreutils-minimal.

@ehmry did you already check if the coreutils update builds for arm_v8a? It would be great to address this open issue in passing.

closed 2019-07-12 by ehmry