genodelabs/genode #3422: util/print_lines: don't skip end of long lines

m-stein, 2019-06-18:
The print_lines utility currently just skips the end of over-sized lines. It should rather split-up an over-sized input line to multiple consecutive output lines.

m-stein, 2019-06-18:
Fixed:
952d7b0de7 util/print_lines: don't skip end of long lines

m-stein, 2019-06-18:
I've adapted the commit according to the suggestions of @nfeske and tested it for correct output with different line lengths:
b2d87facaf7 util/print_lines: don't skip end of long lines

m-stein, 2019-06-19:
I've tested the version suggested by @nfeske and added documentation. The documentation may as well be removed:
a90165d46c util/print_lines: don't skip end of long lines

m-stein, 2019-06-19:
@nfeske I have to correct myself: Your version does print an extra empty line for each line that is <= MAX_LINE_LEN. This is because the '\n' now gets copied to the buffer and is not overwritten with a '\0' as before. I was not able to quickly find a solution that is not as complicated as my second suggestion.

m-stein, 2019-06-19:
@nfeske maybe this one?
63103eaaa1 util/print_lines: don't skip end of long lines

nfeske, 2019-06-19:
The latest version looks good. Thanks for the welcome improvement. I merged the commit to staging.

commit 573333d1 (2019-06-19): util/print_lines: don't skip end of long lines

Print the end of oversized lines (> 200 chars) to new lines instead of
skipping it

Fixes #3422
closed 2019-07-09 by chelmuth