genodelabs/genode #3423: depot_autopilot: potential memory leak

rite, 2019-06-18:
@rite The problem with test-fault_detection seems to be a memory leak in the depot_autopilot. [...] Anyway, it should receive a dedicated issue. Thanks again for documenting!

_Originally posted by @m-stein in https://github.com/genodelabs/genode/issues/3238#issuecomment-503084017_

chelmuth, 2019-12-20:
I cannot confirm these behavior with current master. Can we close this issue?

rite, 2020-01-06:
I struggled a bit trying to reproduce the conditions for this issue. But by skipping some tests like `test-signal`, 'test-xml_generator', I was able to let `run/depot_autopilot `execute 72 iterations which took around 15 min. The script eventually failed but as far as I can see from other reasons than a memory leak. So let's close this issue.

closed 2020-01-06 by rite