genodelabs/genode #3425: ./tool/create_uboot usb_armory crashes with fatal error : linux/compiler-gcc8.h no such file or directory

sheepNo, 2019-06-19:
I'm running Linux 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux and I'm trying to follow this https://github.com/genodelabs/genode/blob/master/repos/os/run/tz_vmm.run to install Genode on the USB Armory.

I'm running into the following problem : 
```bash
./tool/create_uboot usb_armory/genode_usb_armory'
Votre branche est à jour avec 'origin/genode_usb_armory'.
Building U-Boot config...
make[1] : on entre dans le répertoire « /home/user/genode/contrib/uboot »
  GEN     /home/user/genode/contrib/uboot/build/usb_armory/Makefile
Configuring for usbarmory - Board: usbarmory, Options: IMX_CONFIG=board/inversepath/usbarmory/imximage.cfg
make[1] : on quitte le répertoire « /home/user/genode/contrib/uboot »
Building U-Boot images...
make[1] : on entre dans le répertoire « /home/user/genode/contrib/uboot/build/usb_armory »
  GEN     include/autoconf.mk.dep
  GEN     include/autoconf.mk
In file included from /home/user/genode/contrib/uboot/include/linux/compiler.h:40,
                 from /home/user/genode/contrib/uboot/include/linux/byteorder/little_endian.h:12,
                 from /home/user/genode/contrib/uboot/arch/arm/include/asm/byteorder.h:29,
                 from /home/user/genode/contrib/uboot/include/compiler.h:112,
                 from /home/user/genode/contrib/uboot/include/image.h:19,
                 from /home/user/genode/contrib/uboot/include/common.h:90:
/home/user/genode/contrib/uboot/include/linux/compiler-gcc.h:93:1: fatal error: linux/compiler-gcc8.h: Aucun fichier ou dossier de ce type
 #include gcc_header(__GNUC__)
 ^~~~
compilation terminated.
In file included from /home/user/genode/contrib/uboot/include/linux/compiler.h:40,
                 from /home/user/genode/contrib/uboot/include/linux/byteorder/little_endian.h:12,
                 from /home/user/genode/contrib/uboot/arch/arm/include/asm/byteorder.h:29,
                 from /home/user/genode/contrib/uboot/include/compiler.h:112,
                 from /home/user/genode/contrib/uboot/include/image.h:19,
                 from /home/user/genode/contrib/uboot/include/common.h:90:
/home/user/genode/contrib/uboot/include/linux/compiler-gcc.h:93:1: fatal error: linux/compiler-gcc8.h: Aucun fichier ou dossier de ce type
 #include gcc_header(__GNUC__)
 ^~~~
compilation terminated.
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  GEN     /home/user/genode/contrib/uboot/build/usb_armory/Makefile
  UPD     include/generated/timestamp_autogenerated.h
  Using /home/user/genode/contrib/uboot as source for u-boot
  CHK     include/generated/version_autogenerated.h
  CC      lib/asm-offsets.s
  CC      arch/arm/lib/asm-offsets.s
In file included from /home/user/genode/contrib/uboot/include/linux/compiler.h:40,
                 from /home/user/genode/contrib/uboot/include/linux/byteorder/little_endian.h:12,
                 from /home/user/genode/contrib/uboot/arch/arm/include/asm/byteorder.h:29,
                 from /home/user/genode/contrib/uboot/include/compiler.h:112,
                 from /home/user/genode/contrib/uboot/include/image.h:19,
                 from /home/user/genode/contrib/uboot/include/common.h:90,
                 from /home/user/genode/contrib/uboot/arch/arm/lib/asm-offsets.c:15:
/home/user/genode/contrib/uboot/include/linux/compiler-gcc.h:93:1: fatal error: linux/compiler-gcc8.h: Aucun fichier ou dossier de ce type
 #include gcc_header(__GNUC__)
 ^~~~
compilation terminated.
In file included from /home/user/genode/contrib/uboot/include/linux/compiler.h:40,
                 from /home/user/genode/contrib/uboot/include/linux/byteorder/little_endian.h:12,
                 from /home/user/genode/contrib/uboot/arch/arm/include/asm/byteorder.h:29,
                 from /home/user/genode/contrib/uboot/include/compiler.h:112,
                 from /home/user/genode/contrib/uboot/include/image.h:19,
                 from /home/user/genode/contrib/uboot/include/common.h:90,
                 from /home/user/genode/contrib/uboot/lib/asm-offsets.c:15:
/home/user/genode/contrib/uboot/include/linux/compiler-gcc.h:93:1: fatal error: linux/compiler-gcc8.h: Aucun fichier ou dossier de ce type
 #include gcc_header(__GNUC__)
 ^~~~
compilation terminated.
/home/user/genode/contrib/uboot/./Kbuild:83 : la recette pour la cible « arch/arm/lib/asm-offsets.s » a échouée
make[4]: *** [arch/arm/lib/asm-offsets.s] Erreur 1
make[4]: *** Attente des tâches non terminées....
/home/user/genode/contrib/uboot/./Kbuild:34 : la recette pour la cible « lib/asm-offsets.s » a échouée
make[4]: *** [lib/asm-offsets.s] Erreur 1
/home/user/genode/contrib/uboot/Makefile:1066 : la recette pour la cible « prepare0 » a échouée
make[3]: *** [prepare0] Erreur 2
Makefile:138 : la recette pour la cible « sub-make » a échouée
make[2]: *** [sub-make] Erreur 2
Makefile:26 : la recette pour la cible « all » a échouée
make[1]: *** [all] Erreur 2
make[1] : on quitte le répertoire « /home/user/genode/contrib/uboot/build/usb_armory »
tool/create_uboot:108 : la recette pour la cible « /home/user/genode/contrib/uboot/build/usb_armory/u-boot.imx » a échouée
make: *** [/home/user/genode/contrib/uboot/build/usb_armory/u-boot.imx] Erreur 2
```

The compilation fails with `linux/compiler-gcc8.h: no such file or directory.

I tried installing the genode toolchain and did `sudo update-alternatives --install /usr/bin/gcc gcc /usr/local/genode/tool/19.05/bin/genode-arm-gcc 100` to no avail.

 Is there anything I'm missing ?

ssumpf, 2019-06-19:
The uboot version is outdated. It only offers `compiler-gcc4.h` and ` compiler-gcc3.h`. As a quick fix, using the Genode tool chain 16.05 (https://sourceforge.net/projects/genode/files/genode-toolchain/16.05), putting it into `/usr/local/genode/tools/16.05` and updating the `current` link to point at `16.05.` should do the trick for uboot. Please keep building Genode with the 19.05 tool chain though.

sheepNo, 2019-06-24:
I've forgotten what I tried, but I'm pretty sure your solution + undoing `sudo update-alternatives --install /usr/bin/gcc gcc /usr/local/genode/tool/19.05/bin/genode-arm-gcc 100` fixed the issue. I'm not exactly sure what it was interfering with but the problem seems to be fixed now.

closed 2019-06-24 by sheepNo

nfeske, 2019-06-24:
@sheepNo Thank you for revisiting and closing the issue.