genodelabs/genode #3426: hw: i.MX8M Quad Evaluation Kit support

skalk, 2019-06-20:


skalk, 2019-06-20:
Ok, interrupts are not working correctly yet. But I pushed it so that @ssumpf and @alex-ab can take over.

cproc, 2019-07-15:
With commit 8b653a3, the lwip test succeeds on i.MX8.

@ssumpf: not entirely sure about the 'Ctlr' structure change, but this version matches the one described as "When access is Non-secure, in a system that supports two Security states" in the GIC manual.

ssumpf, 2019-07-15:
@cproc: Good work! I to have misinterpreted the `ITEMS` field of an `Register_array`. Yeah, the secure non-secure thing is annoying. To my knowledge I thought we are secure in bootstrap on imx8, but who knows.

cproc, 2019-07-15:
@ssumpf: I changed the structure only because the comment says "(non-secure access)". It didn't seem to make a difference though.

cproc, 2019-07-16:
So, it is secure access on i.MX8. I reverted the struct change and changed the comment accordingly.

commit ae826c32 (2019-07-11): hw: support for i.MX8M Quad EVK

Fix #3426
closed 2019-08-21 by chelmuth