genodelabs/genode #3428: fec_nic_drv: MDIO timeout

trimpim, 2019-06-24:
I work on a module with an i.MX6 Quad. To change the configuration of the system I have a ssh_terminal that is connected via a nic_router to the fec_nic_drv. The system is modeled in a similar fashion as SCULPT. `fec_nic_drv`, `nic_router` and `ssh_terminal` are started by `depot_deploy` in a sub-init.

I often get log messages "nic_drv: netdev_err: MDIO read timeout", where after no more network communication to the board is possible. This happens sometimes already during initialization of the driver.

I have a small patch to the `dde_linux` contrib sources, that seems to fix the issue for me. While discussing the patch with my colleges the question came up, if this patch would not just gloss over an other deeper problem.

cproc, 2019-06-27:
We haven't decided yet what to do about this issue (patch or debug), but you might also give the updated driver from issue #3432 a try and see if it improves anything in this regard.

trimpim, 2019-06-28:
@cproc I'll test the new driver and give feedback on it. But I  have to finish some other things first. You'll hear from me about Tuesday next week.

trimpim, 2019-07-02:
@cproc  I have tested the new driver from #3432. Without my patch, the new diver has the same problem with the "nic_drv: netdev_err: MDIO read timeout" in my scenario.

cproc, 2019-07-04:
@trimpim: thanks for testing

nfeske, 2022-11-17:
This issue has become obsolete since a new version of the FEC driver entered the genode-imx repository.

closed 2022-11-17 by nfeske