genodelabs/genode #3430: Provide platform-identification information from SMBIOS/DMI

m-stein, 2019-06-25:
Read out platform-identification information from sources like SMBIOS/DMI and provide them as XML-structured report.

m-stein, 2019-06-25:
This is a port of dmidecode that - at least on my qemu - prints sensible SMBIOS/DMI information to the log:
3979700e29 dmidecode: first working prototype

The port currently uses a local, modified posix environment to map the SMBIOS/DMI region as IOMEM. As a first option to get rid of this, I could add a component that maps the IOMEM to a ROM and then include the ROM as VFS dev/mem. Later, I could add an IOMEM plugin for VFS to get rid of the extra component.

Also, the conversion to an XML report is missing for now.

chelmuth, 2019-06-26:
The customized POSIX environment is another example (beside posix lib and Qt #3049) where it would be handy to have these functions as utility header, e.g., *repos/libports/include/libc/argv_envp.h*.

m-stein, 2019-06-26:
Unfortunately, translating the output of dmidecode to XML can't be done through a simple back-end replacement as dmidecode calls printf in each place individually. Therefore I remember all structure headers, and parse the date afterwards myself. This is done so far for type 0 structures or BIOS information. It works on all our Fuji models.

chelmuth, 2019-06-27:
Could you make a rough estimate how much code is still used from dmidecode with this approach? It sounds like we only reuse the heuristics to locate the table now.

m-stein, 2019-06-27:
@chelmuth This depends. For the mere goal of giving a report of type 0-2 structures, we use mainly the table searching with about 1KLOC original code. But, so far, we have replaced only <200 LOC of the ~6KLOC original code with our own code. The rest of dmidecode provides a lot more information than the report. I liked the approach to gradually enhance the feature set of our own code while still having the full log output of dmidecode as fall back. What do you think?

m-stein, 2019-06-27:
The new version reports additional table information as well as type 1 (sys info) and 2 (base board info) structure data. It is tested for Fujitsu Lifebook S9XX, S937, S938 and Lenovo X201, T430s.

chelmuth, 2019-06-28:
Your approach makes sense from a practical perspective but I'd like to avoid situations where code fragments in files with Genode headers are almost exact copies from *GPLv2 or later* source-code files (e.g., `report_dmi_bios_character_1()`.

m-stein, 2019-06-28:
I've tested the current version with UEFI. On all above mentioned systems, it doesn't find a table with UEFI but dmidecode on Ubuntu 18.04 does.

@chelmuth I agree with you. I plan to replace these code paths with our own code in the final version but I wanted to provide a documentation of the process also.

In the context of the adaptions yet to apply for UEFI, I think it's sensible to create a completely native Genode component. But first I'll have a look at how the tables are found on Linux+UEFI.

m-stein, 2019-06-28:
My plan so far is to add searching for dmi/smbios tables to grub and creating an additional multiboot-info tag:
legacy: grub-core/loader/i386/multiboot_mbi.c:grub_multiboot_make_mbi
uefi: grub-core/loader/multiboot_mbi2.c:grub_multiboot2_make_mbi

Then, Core/Bootstrap reads that info and creates a ROM with another copy of the table:
base-sel4/src/core/platform.cc:Platform::_init_rom_modules
base-nova/src/core/platform.cc:Platform::Platform
base-hw/src/bootstrap/spec/x86_64/platform.cc:

It seems that on nova, also the kernel has to be adapted to provide the info in it's HIP.

chelmuth, 2019-07-01:
What I know from the dmidecode code is, that SMBIOS information is found in the *systable* when UEFI is used. Does GRUB already provide the systable to booted kernels?

m-stein, 2019-07-01:
The systable you mention seems to originate from the EFI system table in the EFI struct in linux which is marked extern and apparently is provided by an earlier stage in the boot process. I din't figure out so far how to receive the EFI struct. I'd like to try the multiboot-info approach first. I'm currently trying to find a way to let GRUB find out the SMBIOS/DMI-table address.

cnuke, 2019-07-01:
IIRC GRUB2 should already store the address of the EFI system table in the MB header. Maybe it makes sense to try to access the system table and look up the SMBIOS/DMI table from there as we maybe want to use the `efi_call` API to interact with the firmware in the future anyway.

m-stein, 2019-07-01:
@cnuke Thanks for the hint! Unfortunately, although in our setup the EFI system table can be read from the multiboot2 info, the table contains an empty configuration table (which is used, for instance, in OpenBSD to get the SMBIOS table).

m-stein, 2019-07-02:
I'm now able to read SMBIOS EPS pointers from the EFI system table in hw/bootstrap, validate their content and copy EPS and referenced SMBIOS table structures to a contiguous RAM for Core. Next I let Core create and provide a ROM module for that RAM region.

m-stein, 2019-07-02:
@alex-ab @chelmuth @cnuke Thanks for all your help regarding this issue!

chelmuth, 2019-07-03:
Thank you, @m-stein, for your perseverance in taming BIOS and (U)EFI equally ;-)

m-stein, 2019-07-04:
There's now a native component app/dmi_decoder which, so far, reports SMBIOS/DMI entrypoint data, "BIOS" structure header/body data, and of the other structures only header data. It works with UEFI and legacy on Qemu and the Fujitsu laptops.

m-stein, 2019-07-04:
It always receives the DMI info as ROM dataspace from Core.

chelmuth, 2019-07-04:
I really like the progress you make. Do you already have plans how to solve the *smbios_info* ROM creation on NOVA (and other x86 kernels)?

m-stein, 2019-07-04:
I plan to do this tomorrow. Today I'd like to reach full functionality (bios, system, base board) with the new dmi_decoder.

m-stein, 2019-07-04:
But no, currently I have no plan how exactly to do this. But, like in base-hw I would let me guide by the way, ACPI is propagated.

m-stein, 2019-07-04:
DMI decoder now also reports "Base Board" and "System". I plan to rename it SMBIOS decoder as DMI is part of SMBIOS and the decoder also supports non-DMI SMBIOS specs.

m-stein, 2019-07-07:
I tried to move the parsing of EFI system table and SMBIOS headers to the ACPI driver. This works just fine for systems that mark mem regions of header and SMBIOS table as reserved in the multiboot info so they get removed from the RAM allocators in core (is the case for all UEFI systems I tested). Unfortunately, this was not the case for the legacy BIOS systems I tested. Therefore, so far, I don't see a solution other than parsing SMBIOS in Core/Bootstrap. But I designed the code in ``drivers/smbios.h`` and ``drivers/efi_system_table.h`` in a way that makes it easy to move the parsing to the userland if we should find another solution.

Anyway, dmi_decoder now works on NOVA and HW with legacy BIOS and UEFI.

chelmuth, 2019-07-08:
Did I get it wrong that SMBIOS tables are located 0xe0000-0xfffff in legacy mode? This region should always be reserved as IOMEM.

chelmuth, 2019-07-08:
Okay, I tested some systems in our network and it's not always below 1M but all tables are located in e820 *reserved* regions resp. not located in *system RAM*. This hints the BIOS ensures the same properties for SMBIOS data structures as it does for ACPI tables.

m-stein, 2019-07-09:
NOVA and HW now povide the EFI system-table pointer from the multiboot2 info in the platform info ROM as ``<efi-system-table address="0x...">``. This tag is missing if no EFI system table was found. The parsing of EFI system table and SMBIOS headers is done only by the ACPI driver (renaming might be sensible). If no EFI system table is found, the driver looks out for the valid SMBIOS table with the highest version via a memory scan. The ACPI driver creates a report 'smbios_table' of the plain SMBIOS table data. The SMBIOS decoder uses the ROM to create an XML report of some of the SMBIOS info. This works with NOVA and HW on Fujitsu Lifebook S9XX, S937, S938 and Lenovo X201, T430s with BIOS and UEFI.

Genode:
8e3a5cd smbios_decoder: XML report from SMBIOS table ROM
0fb69d0 drivers/acpi: provide plain SMBIOS table as report
f561d33 base-nova: EFI sys-table pointer in platform info
0ab99a9 base-hw: EFI sys-table pointer in platform info

NOVA:
https://github.com/m-stein/NOVA/commit/afb1c261c96a8ef796648076b1ab4c4325f1a742

chelmuth, 2019-07-10:
@m-stein I'm pretty happy with the current solution as it keeps changes to critical components minimal and moves optional stuff into *acpi_drv* and *smbios_decoder*. Ready to merge after my remarks were addressed.

m-stein, 2019-07-10:
@chelmuth Thanks. I've updated the commits according to your suggestions.

m-stein, 2019-07-11:
@chelmuth @alex-ab Again without 32 bit EFI system table pointers:

NOVA:
https://github.com/m-stein/NOVA/commit/a93ce0661f12e156c92892e66c641689860b08e2

Genode:
4578651 smbios_decoder: XML report from SMBIOS table ROM
bc2830c drivers/acpi: provide plain SMBIOS table as report
04333b5 base-nova: EFI sys-table pointer in platform info
d06e994 base-hw: EFI sys-table pointer in platform info

alex-ab, 2019-07-11:
@chelmuth: d2de76bad78fcd6e3fc106014077198a63cf1c4f fixup for staging with the new nova commit

commit 5c4d854a (2019-07-11): smbios_decoder: XML report from SMBIOS table ROM

Fixes #3430
commit 758f4bef (2019-07-12): smbios_decoder: XML report from SMBIOS table ROM

Fixes #3430
commit fe426e6f (2019-08-13): smbios_decoder: XML report from SMBIOS table ROM

Fixes #3430
closed 2019-08-21 by chelmuth