genodelabs/genode #3431: tool/depot/publish: use multiple threads in xz

cnuke, 2019-06-27:
Publishing all archives takes a considerable amount of time in our nightly CI because `xz` is used single-threadily[sic] and we cannot take advantage of the beefy build machine. However, it features support for multi-threaded (de-)compression for some time now, so let's switch it on by using `--threads`. AFAICT it is a no-op in older versions (*5.1.0alpha* as used in Ubuntu 16.04 LTS) but the options is still recognized.

cnuke, 2019-06-27:
@chelmuth please add 7628918 to staging. I'll add a more recent `xz` version to the build environment on the build machine.

It should be noted that using threaded `xz` will lead to larger archives (the vbox5 src archive is about ~1% (400KiB) larger).

nfeske, 2019-06-28:
I wonder, have you tried invoking the publish tool with `-j` and let make do the parallelization?

cnuke, 2019-06-28:
No, I worked with the (presumably) out-dated knowledge that the `publish` does not support jobs.

nfeske, 2019-06-28:
Maybe you are right. Admittedly, I haven't checked if -j works. I'm not aware of a fundamental reason why if shouldn't.

cnuke, 2019-06-28:
I stand corrected, using jobs works after all. I changed the CI tools and now feed the list of to-be-published archives via `xargs` to `publish -j32` - let's see how that works out.

cnuke, 2019-07-01:
Unfortunately publishing *pkg* archives failed because of missing dependencies, which prevented the archives from being published at all as the *keep-going* flag was missing.

commit fa0e0ac2 (2019-06-27): tool/depot/publish: use multiple threads in xz

Introduce 'XZ_THREADS' to override the default number of threads which
will use all available cores. Using multiple threads will increase
the amount of memory needed during the compression and setting the
variable might limit the memory usage.

Fixes #3431.
closed 2019-07-09 by chelmuth