genodelabs/genode #3439: arora browser does not load websites since libc update

cproc, 2019-07-04:
There's a new warning in the log, but it might not be related to the problem.

```
[init -> arora] Warning: getifaddrs: getifaddrs not implemented.
```

ehmry, 2019-07-05:
I added `getifaddrs` to dummies because it was a missing during the port transition, so this really could be the problem.

ehmry, 2019-07-05:
I have `getifaddrs` implemented at d7f24d8bca3790a819c2b5974614e2a48d26bd63, though I have not tested it with Arora.

cproc, 2019-07-05:
With Arora, 'getaddrinfo()' still failed, apparently because the 'IFF_UP' flag was not set in the 'ifa_flags' struct member.

ehmry, 2019-07-08:
Fixup at 34b042158333e8108a91f6214b37ce41a8de35f7, from what I can see IFF_UP is the only flag that can be reasonably set.

commit 36111a2e (2019-09-02): Libc: implement getifaddrs

Implement getifaddrs and freeifaddrs within the libc using socket
control files at the VFS. Add an "address" and "netmask" file to the
lwIP plugin.

Only a single IPv4 address is initially supported, and the broadcast
address returned will never be valid.

Fixes #3439
closed 2019-09-20 by chelmuth