genodelabs/genode #3441: part_blk: may enter endless loop on insufficient packet-stream space

chelmuth, 2019-07-09:
On packet-stream space shortage components with outstanding requests are enqueued into a wait queue and woken up if any I/O happend. The wake up works like follows.

repos/os/src/server/part_block/component.h:232
```c++
static void wake_up()
{
  for (; Session_component *c = wait_queue().first();)
  {
    wait_queue().remove(c);
    c->_req_queue_full = false;
    c->_handle_packet(c->_p_to_handle);
    c->_packet_avail();
  }
}
```

Then, `_handle_packet()` tries `perform_io = [&] ()`, which enqueues the component again on insufficient packet-stream space.

repos/os/src/server/part_block/component.h:104
```c++
} catch (Block::Session::Tx::Source::Packet_alloc_failed) {
if (!_req_queue_full) {
  _req_queue_full = true;
  Session_component::wait_queue().insert(this);
}
```

This creates a perfect endless loop in `wake_up()`.

*This behavior is tightly coupled to the current component design which tightly intervenes the server component implementation with the packet-stream logic on server and connection side. The separation of these aspects is addressed by the new Requests and Jobs APIs.*

ssumpf, 2020-02-17:
Should be fixed with issue #3652, code has been removed.

chelmuth, 2020-02-17:
Thanks @ssumpf for the clarification!

closed 2020-02-28 by nfeske