genodelabs/genode #3444: fatal error when running genode/tool/tool_chain

rainstared, 2019-07-11:
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/libgcc/libgcc2.c:27:
/usr/include/stdio.h:27:10: fatal error: bits/libc-header-start.h: No such file or directory
 #include <bits/libc-header-start.h>

This error is solved by running sudo apt-get install gcc-multilib g++-multilib
as linked here. https://github.com/sirfz/tesserocr/issues/130

However, when restarting, libraries are installed and then the installer seems to fail.
The relevant command line excerpt is below, followed by the relevant config.log exerpt.

Terminal lines:
`libtool: finish: PATH="/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/lib/gcc/armv7l-unknown-linux-gnueabihf/8.3.0/plugin
----------------------------------------------------------------------
Libraries have been installed in:
   /media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/lib/gcc/armv7l-unknown-linux-gnueabihf/8.3.0/plugin

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the `LD_RUN_PATH' environment variable
     during linking
   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to `/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
/bin/bash: line 3: cd: armv7l-unknown-linux-gnueabihf/libstdc++-v3: No such file or directory
Makefile:11314: recipe for target 'install-strip-target-libstdc++-v3' failed
make[2]: *** [install-strip-target-libstdc++-v3] Error 1
Makefile:2440: recipe for target 'install-strip' failed
make[1]: *** [install-strip] Error 2
tool_chain:465: recipe for target '/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/bin/g++' failed
make: *** [/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/bin/g++] Error 2
`

config.log lines: 
`
configure:4310: $? = 1
configure:4299: gcc -qversion >&5
gcc: error: unrecognized command line option '-qversion'; did you mean '--version'?
gcc: fatal error: no input files
compilation terminated.
configure:4310: $? = 1
configure:4330: checking whether the C compiler works
configure:4352: gcc -DTE_LINUX   conftest.c  >&5
/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/lib/gcc/armv7l-unknown-linux-gnueabihf/8.3.0/../../../../armv7l-unknown-linux-gnueabihf/bin/ld: cannot find crtbe$
/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/lib/gcc/armv7l-unknown-linux-gnueabihf/8.3.0/../../../../armv7l-unknown-linux-gnueabihf/bin/ld: cannot find -lgcc
/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/install/lib/gcc/armv7l-unknown-linux-gnueabihf/8.3.0/../../../../armv7l-unknown-linux-gnueabihf/bin/ld: cannot find -lgcc$
collect2: error: ld returned 1 exit status
configure:4356: $? = 1
configure:4394: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME ""
| #define PACKAGE_TARNAME ""
| #define PACKAGE_VERSION ""
| #define PACKAGE_STRING ""
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| /* end confdefs.h.  */
|
| int
| main ()
| {
|
|   ;
|   return 0;
| }
configure:4399: error: in `/media/raspex/swap/genode/build/tool_chain-19.05/aarch64/binutils':
configure:4401: error: C compiler cannot create executables
See `config.log' for more details
`

rainstared, 2019-07-11:
While I have your attention and after you modify your makefile to handle the intricacies of life:

Can I please ask why the design of the interface of sculpt's system composition tool feels so wrong?
it's as if someone never took a course in usability. It's as if punishment is pleasure.
Can I please ask why the design of the depot tool is so stateless?
it's as if someone never took a course in database design. it laughs at git.

As an ethnic german, I feel like somewhere along the way my family lost an essential spirit required to manage this challenge.

My principle example uses the brand new 19.07 demo. Say you want to add the qt5-textedit tool to your
sculpt instance by using the depot tool and you have not read the instructions.
You screw it up, managing to assign the system GUI server instead of the last option, which apparently gives the application the ability to access where your mouse is and so the application is in the background where you can't use it, taunting you for your failure.
You close it, but then what do you do? You have to download it all over again to run it again.
The files are already on the drive! Why are we downloading everything ALL OVER AGAIN?

Perhaps there is a simple set of instructions to follow. I sure hope so, that's what's keeping me going.
Or perhaps, if I had a masters degree level of experience with the underlying fundamentals of genode framework, I would know what to do. To my eyes, there is no obvious answer that would involve 3 clicks or 10 keystrokes to resurrect the files already on the partition, as evidenced when I go to the inspect tool and navigate on the command line. 

I get the feeling from the 30 hours or so i have spent monkeying with genode versions over the past 5 years, with reading the manuals, that, from an uneducated perspective,  if I wanted to do anything exponentially more sophisticated like bootstrap genode ON genode and upgrade it and then boot it, i would need to unwrap several esoteric tomes and light a large number of candles to the god of Leitzentrale in order to get to the underlying root system. And heaven forbid I ever attempt to bootstrap the genode toolchain on genode, even the developers themselves would need to perform ritual sacrifices in order to even type the first keystroke.

ssumpf, 2019-07-11:
@rainstared: Even though I think your tool chain findings are interesting. I can only ask you to keep your writing at a polite level, as we all do. It is a common practice and a simple thing humans should be able to do. I know things can get frustrating, views on things are different, and we are open for a meaningful discussion. So please reword the paragraph above into something that can be argued about in a meaningful way. If you would had build some tool and I would use it and tell it is utter ... you know ... Would you talk to me?

rainstared, 2019-07-11:
my apologies ssumpf.

rainstared, 2019-07-11:
I'm going to switch to staging using ./tool_chain cleanall and then (in the genode directory) git pull origin --rebase staging and then retry ./tool_chain aarch64 and see if changes made in the most recent commits affect or improve this situation.

ssumpf, 2019-07-11:
Cool, you are the first one trying that on rpi4. So best wishes and if you get it to work any fixes are welcome.

rainstared, 2019-07-11:
Toolchain libraries down to GCC compile and install correctly. I modified my scrollback to capture the entire first run.
https://pastebin.com/nD2dMnH7 The first run of the GCC compileation is here.
Many instances of "failed to merge target specific data of file" occur.
following [https://stackoverflow.com/questions/9753749/arm-compilation-error-vfp-registered-used-by-executable-not-object-file](https://stackoverflow.com/questions/9753749/arm-compilation-error-vfp-registered-used-by-executable-not-object-file), I run:
```
env set CFLAGS=" -march=armv7-a -mfloat-abi=hard -mfpu=neon -marm -mthumb-interwork"  ```
I don't really know if those are the right compiler flags, I just figured since RPI4 can run as arm7 they are likely compatible.
and then run **/tool_chain AARCH64 -e**  
Of course, it crashes again, but with a different error:

```
g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  gcov.o \
	hash-table.o ggc-none.o  -o gcov
gcov.o: In function `print_usage(int)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:813: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:814: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:815: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:816: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:817: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
gcov.o:/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:819: more undefined references to `fnotice(_IO_FILE*, char const*, ...)' follow
gcov.o: In function `print_usage(int)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:837: undefined reference to `bug_report_url'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:837: undefined reference to `bug_report_url'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:837: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
gcov.o: In function `gcov_allocate':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov-io.c:258: undefined reference to `xrealloc'
gcov.o: In function `canonicalize_name(char const*)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2287: undefined reference to `xmalloc'
gcov.o: In function `output_branch_count(_IO_FILE*, int, arc_info const*)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2692: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2702: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2705: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2707: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2687: undefined reference to `fnotice(_IO_FILE*, char const*, ...)'
gcov.o:/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2715: more undefined references to `fnotice(_IO_FILE*, char const*, ...)' follow
gcov.o: In function `mangle_name(char const*, char*)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2436: undefined reference to `lbasename'
gcov.o: In function `gcov_sync(unsigned int, unsigned int)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov-io.c:675: undefined reference to `fancy_abort(char const*, int, char const*)'
gcov.o: In function `print_source_line(_IO_FILE*, std::vector<char const*, std::allocator<char const*> > const&, unsigned int)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2836: undefined reference to `fancy_abort(char const*, int, char const*)'
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/gcov.c:2837: undefined reference to `fancy_abort(char const*, int, char const*)'
gcov.o: In function `gcov_open(char const*, int) [clone .constprop.393]':
[... cut ...]
[https://github.com/google/honggfuzz/issues/55](https://github.com/google/honggfuzz/issues/55) some other people seem to have similar issues.
at this point i ran ***sudo apt-get install binutils-dev  libiberty-dev***
and tried again, to same result.

[http://sourceware.org/ml/crossgcc/2003-10/msg00086.html](http://sourceware.org/ml/crossgcc/2003-10/msg00086.html)

[https://www.spinics.net/lists/netdev/msg541076.html](https://www.spinics.net/lists/netdev/msg541076.html)

it would seem with regards to binutils and arm this is a longstanding struggle

ssumpf, 2019-07-11:
This only seems to be gcov stuff, can you disable it somehow? It is not required.

rainstared, 2019-07-11:
how do I disable gconv?

rainstared, 2019-07-11:
I added --without-gcov to CFLAGS but i still get the errors. I'm pretty sure this doesnt only seem to be gcov stuff.

rainstared, 2019-07-12:
I think one problem is that i'm trying to compile arm64 toolkit on armhf system.
I'll change to arm and redo the entire process and see what happens.

rainstared, 2019-07-12:
I Upgraded to 19.04 as best I could, with GCC 8.3 along with all the supporting cruft.
They added some arm compilation features.

rainstared, 2019-07-12:
```
++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/. -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../include -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libcpp/include -I/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/include -I/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/include -I/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/include  -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libdecnumber -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libbacktrace   -o cc1-checksum.o -MT cc1-checksum.o -MMD -MP -MF ./.deps/cc1-checksum.TPo cc1-checksum.c
g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1 c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o arm-c.o glibc-c.o \
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a   -L/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/lib -L/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/lib -L/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
/usr/bin/ld: c/c-parser.o: in function `void va_heap::reserve<c_generic_association>(vec<c_generic_association, va_heap, vl_embed>*&, unsigned int, bool)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: /media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: /media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: c/c-parser.o: in function `void va_heap::reserve<tgmath_parm_kind>(vec<tgmath_parm_kind, va_heap, vl_embed>*&, unsigned int, bool)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: c/c-parser.o: in function `xcallocator<tree_node*>::data_alloc(unsigned int)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/hash-table.h:263: undefined reference to `xcalloc'
/usr/bin/ld: c/c-parser.o: in function `void va_heap::reserve<unsigned int>(vec<unsigned int, va_heap, vl_embed>*&, unsigned int, bool)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: c/c-parser.o: in function `void va_heap::reserve<c_token>(vec<c_token, va_heap, vl_embed>*&, unsigned int, bool)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: /media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: /media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: /media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
/usr/bin/ld: c/c-parser.o:/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: more undefined references to `xrealloc' follow
/usr/bin/ld: c-family/c-format.o: in function `avoid_dollar_number':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/c-family/c-format.c:1335: undefined reference to `_sch_istable'
[... cut ...]
make[2]: Leaving directory '/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/gcc'
make[1]: *** [Makefile:909: all] Error 2
make[1]: Leaving directory '/media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/gcc'
make: *** [tool_chain:462: /media/raspex/swap/genode/build/tool_chain-19.05/bootstrap/gcc/gcc/xg++] Error 2
```
On the other hand, it took a lot less time to reach this point, even with clean.
After overclocking the pi4 and running the compile one file at a time, it only takes 90 min.

rainstared, 2019-07-12:
Ok, so I successfully removed the gcov dependency by altering tool_chain.
The compile got a little further but still failed in the linking stage.
Attached is the build log from the point of first failure. it goes on for hundreds of lines thou.
```
g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/. -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../include -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libcpp/include -I/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/include -I/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/include -I/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/include  -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libdecnumber -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/../libbacktrace   -o cc1-checksum.o -MT cc1-checksum.o -MMD -MP -MF ./.deps/cc1-checksum.TPo cc1-checksum.c
g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1 c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o arm-c.o glibc-c.o \
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a   -L/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/lib -L/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/lib -L/media/raspex/swap/genode/build/tool_chain-19.05/lib-install/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
/usr/bin/ld: c/c-parser.o: in function `void va_heap::reserve<c_generic_association>(vec<c_generic_association, va_heap, vl_embed>*&, unsigned int, bool)':
/media/raspex/swap/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/vec.h:288: undefined reference to `xrealloc'
```

chelmuth, 2019-07-12:
May I ask why you try to build Genode on the Rpi4 in contrast to do cross compilation on PC? Are you planning to run Genode/Linux on the Pi or do you want to address Genode on a microkernel? I'm asking as I get the feeling you're the first person walking these undiscovered territory.

nfeske, 2019-07-12:
> collect2: fatal error: ld terminated with signal 9 [Killed]

Have you checked `dmesg`? This could be an OOM during the linking.

rainstared, 2019-07-12:
> May I ask why you try to build Genode on the Rpi4 in contrast to do cross compilation on PC? Are you planning to run Genode/Linux on the Pi or do you want to address Genode on a microkernel? I'm asking as I get the feeling you're the first person walking these undiscovered territory.

RPI4 with a couple of heatsinks is safely, warrantee-covered overclocked to 1.75ghz. The performance as a host is less than desired but acceptable.

In contrast to cross-compile, direct compile exercises a different set of muscles and would allow discovering bugs that are relevant and meaningful in the future, when you are working from arm hosts to build code for arm targets. We have affordable ARMv8 laptops now like the pinebook 64 and the pinebook pro coming out, and since the RPI4's hardware is mostly supported it makes a reasonable incremental step towards that future. The ultimate dogfood goal for any operating system is for that operating system to be selfhosting on the target platform. If I can provide a proof of work that genode compiles and runs on armv8, on a RPI4, it's another feather in your bonnet.

I planned to attempt to compile the run/demo with the Fiasco.OC microkernel on the PI, as linked in a tutorial I found on the web. I would like to include the arora browser in that demo, and any supporting libraries programs and services sufficient to allow it to successfully navigate to genode.org, but just the demo is the start.

> > collect2: fatal error: ld terminated with signal 9 [Killed]
> 
> Have you checked `dmesg`? This could be an OOM during the linking.
*face palm* This RPI4 only has 1gb. I set up a 7GB swap file on a usb3 flash but I forgot to put it in fstab and rebooted during the updating process a couple of times.. It could be useful in the future for tool_chain to do an check on available ram and swap space combined and warn the user if it won't compile. I will make one more attempt to compile the tool chain after making sure there is enough swap space.

rainstared, 2019-07-12:
I cannot get gcc to compile without gcov! i put --disable-gcov and --without-gcov in both the gcc_bootstrap and the gcc_options sections of tool_chain and it still is trying to compile it!

rainstared, 2019-07-12:
I managed to get --disable-gconv to work by putting it in the common options section of tool_chain.

Ok, so i attached ANOTHER! flash drive and allocated 14GB of swap. the total VM for this system is now 22GB. 
[https://gitlab.haskell.org/ghc/ghc/issues/5240](https://gitlab.haskell.org/ghc/ghc/issues/5240)
[https://github.com/bitcoin/bitcoin/issues/7471](https://github.com/bitcoin/bitcoin/issues/7471)

Apparently there ARE things people can do that reduce how much memory is used, and I will also try them if this compile pass fails.

rainstared, 2019-07-12:
ok, so I did everything I know how, and I used all the options I have presented here.

With
 `export CFLAGS=" -march=armv7-a -mfloat-abi=hard -mfpu=neon -marm -mthumb-interwork --param gcc-min-xpand=1 --param gcc-min-heapsize=32728 -Xlinker --reduce-memory-usage -Xlinker -- hash-size=31 -fuse-ld=gold" 
export CXXFLAGS=" -march=armv7-a -mfloat-abi=hard -mfpu=neon -marm -mthumb-interwork --param gcc-min-xpand=1 --param gcc-min-heapsize=32728 -Xlinker --reduce-memory-usage -Xlinker -- hash-size=31 -fuse-ld=gold"  
`
and COMMON_BOOTSTRAP_CONFIG in tool_chain modified to:

`
215  ' --disable-multilib \
216 +'--disable-gcov \ 
217 +'--without-gcov
 `
It gets further and faster than ever before but still fails, because of one crucial flaw.
armhf is a 32 bit architecture with a floating point MMU and LD32 can only access 2gb of VM.
**GCC 8.3, which needs ~4-6GB of VM, cannot be bootstrapped with ld32 or any 32 bit binutils build.**

Now, the hardware CAN do it. ARM CORTEX-A72 as used in RPI4 BCM2711 is 64 bit.
But the distros build for this device are being built for multiple raspberry pi systems and so are armhf.

I will have to wait for a 64 bit arm linux build before I can effectively attempt this process.

rainstared, 2019-07-12:
64 bit ubuntu, raspian, and raspex are not yet available but ubuntu MATE 18.04 apparently has a 64 bit build. I will repeat this entire process and see what happens.

ssumpf, 2019-07-13:
What a ride! While we have been focused using Genode on our own notebooks lately, we did not lose self hosting out of sight. There are still some prerequisites for this to work well necessary, like getting rid of noux and have a  meaningful  `fork` implementation in libc.

rainstared, 2019-07-17:
After obtaining a RPI4 ubuntu server 18.04 image, I was able to update and configure it to meet what I hope will be the needs of this project.
The stated goal of this project is to compile the genode toolkit on AARCH64 RPI4 successfully, and report any legitimate errors.

 uname -a  Linux ubuntu 4.19.57-v8+ #2 SMP PREEMPT Tue Jul 9 20:31:37 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
 gcc --versiongcc (Ubuntu/Linaro 8.3.0-6ubuntu1) 8.3.0

rainstared, 2019-07-17:
buntu@ubuntu:/media$ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7320372 6010548    977508  87% /
devtmpfs          436272       0    436272   0% /dev
tmpfs             469520   43144    426376  10% /dev/shm
tmpfs              93904     932     92972   1% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs             469520       0    469520   0% /sys/fs/cgroup
/dev/loop0         49408   49408         0 100% /snap/lxd/11107
/dev/loop1         80256   80256         0 100% /snap/core/7275
/dev/mmcblk0p1    258095  113763    144332  45% /boot/firmware
tmpfs              93904      12     93892   1% /run/user/1000
/dev/sdb1       14805376   80884  13952704   1% /media/ubuntu/genode
ubuntu@ubuntu:/media$ free
              total        used        free      shared  buff/cache   available
Mem:         939040      641260       61888       56964      235892      225892
Swap:      15108092       40200    15067892

rainstared, 2019-07-17:
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 8.3.0-6ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=aarch64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --disable-libphobos --enable-multiarch --enable-fix-cortex-a53-843419 --disable-werror --enable-checking=release --build=aarch64-linux-gnu --host=aarch64-linux-gnu --target=aarch64-linux-gnu
Thread model: posix

Architecture:        aarch64
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A72
Stepping:            r0p3
CPU max MHz:         1750.0000
CPU min MHz:         600.0000
BogoMIPS:            108.00
Flags:               fp asimd evtstrm crc32 cpuid

ubuntu                      
    description: Computer
    product: Raspberry Pi 4 Model B Rev 1.1
    serial: 100000000ae5b78b
    width: 64 bits
    capabilities: smp cp15_barrier setend swp
  *-core
       description: Motherboard
       physical id: 0
     *-cpu:0
          description: CPU
          product: cpu
          physical id: 1
          bus info: cpu@0
          size: 1750MHz
          capacity: 1750MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-cpu:1
          description: CPU
          product: cpu
          physical id: 2
          bus info: cpu@1
          size: 1750MHz
          capacity: 1750MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-cpu:2
          description: CPU
          product: cpu
          physical id: 3
          bus info: cpu@2
          size: 1750MHz
          capacity: 1750MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-cpu:3
          description: CPU
          product: cpu
          physical id: 4
          bus info: cpu@3
          size: 1750MHz
          capacity: 1750MHz
          capabilities: fp asimd evtstrm crc32 cpuid cpufreq
     *-memory
          description: System memory
          physical id: 5
          size: 917MiB
     *-pci
          description: PCI bridge
          product: Broadcom Inc. and subsidiaries
          vendor: Broadcom Inc. and subsidiaries
          physical id: 0
          bus info: pci@0000:00:00.0
[... 133 lines cut ...]
                            logical name: /media/ubuntu/genode
                            version: 1.0
                            serial: 2ce0bd29-96d1-41c4-9528-89173acef790
                            size: 14GiB
                            capacity: 14GiB
                            capabilities: primary journaled extended_attributes large_files huge_files dir_nlink recover extents ext4 ext2 initialized
                            configuration: created=2019-07-17 15:30:35 filesystem=ext4 lastmountpoint=/media/ubuntu/genode modified=2019-07-17 15:40:42 mount.fstype=ext4 mount.options=rw,relatime mounted=2019-07-17 15:40:42 state=mounted
  *-usbhost
       product: DWC OTG Controller
       vendor: Linux 4.19.57-v8+ dwc_otg_hcd
       physical id: 1
       bus info: usb@3
       logical name: usb3
       version: 4.19
       capabilities: usb-2.00
       configuration: driver=hub slots=1 speed=480Mbit/s
  *-network
       description: Ethernet interface
       physical id: 2
       logical name: eth0
       serial: dc:a6:32:00:3b:b8
       size: 100Mbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=bcmgenet driverversion=v2.0 duplex=full ip=10.10.10.10 link=yes multicast=yes port=MII speed=100Mbit/s

rainstared, 2019-07-17:
./tool/tool_chain  -e MAKE_JOBS=1 aarch64
 Need to have 'libexpat' installed.
tool/tool_chain:178: *** Please install missing tools..  Stop.

any ideas?

```
lrwxrwxrwx 5 root root     17 Jun 26 11:21 /lib/aarch64-linux-gnu/libexpat.so.1 -> libexpat.so.1.6.8
-rw-r--r-- 2 root root 190536 Jun 26 11:21 /lib/aarch64-linux-gnu/libexpat.so.1.6.8
lrwxrwxrwx 1 root root     17 Jun 27  2017 /snap/core/7275/lib/aarch64-linux-gnu/libexpat.so.1 -> libexpat.so.1.6.0
-rw-r--r-- 1 root root 145488 Jun 27  2017 /snap/core/7275/lib/aarch64-linux-gnu/libexpat.so.1.6.0

SHELL=/bin/bash
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-LXDE:/etc/xdg
XDG_MENU_PREFIX=lxde-
_LXSESSION_PID=798
SSH_AUTH_SOCK=/tmp/ssh-2yFin9ls39jS/agent.798
XDG_DATA_HOME=/home/ubuntu/.local/share
XDG_CONFIG_HOME=/home/ubuntu/.config
DESKTOP_SESSION=LXDE
SSH_AGENT_PID=845
XDG_SEAT=seat0
PWD=/media/ubuntu/genode/genode
LOGNAME=ubuntu
XDG_SESSION_TYPE=x11
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
CXXFLAGS= -march=native -mtune=native -mfpu=neon -mfloat-abi=hard -mthumb-interwork --enable-fix-cortex-a53-843419 -disable-libquadmath --disable-libquadmath-support --param gcc-min-xpand=1 --param gcc-min-heapsize=32728 -Xlinker --reduce-memory-usage -Xlinker -- hash-size=31 -fuse-ld=gold
XAUTHORITY=/home/ubuntu/.Xauthority
LDFLAGS= -L/usr/lib -lexpat
HOME=/home/ubuntu
LANG=C.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=LXDE
VTE_VERSION=5602
INVOCATION_ID=b0189a781f1745caa4355d55a677538a
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=ubuntu
DISPLAY=:0.0
SHLVL=1
XDG_VTNR=7
XDG_SESSION_ID=c2
XDG_RUNTIME_DIR=/run/user/1000
JOURNAL_STREAM=8:15132
XDG_DATA_DIRS=/usr/local/share:/usr/share:/usr/share/gdm:/var/lib/menu-xdg:/usr/local/share/:/usr/share/:/usr/share/gdm/:/var/lib/menu-xdg/
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
CFLAGS= -march=native -mtune=native -mfpu=neon -mfloat-abi=hard -mthumb-interwork --enable-fix-cortex-a53-843419 -disable-libquadmath --disable-libquadmath-support --param gcc-min-xpand=1 --param gcc-min-heapsize=32728 -Xlinker --reduce-memory-usage -Xlinker -- hash-size=31 -fuse-ld=gold -I/usr/include
SAL_USE_VCLPLUGIN=gtk
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
OLDPWD=/media/ubuntu/genode
_=/usr/bin/env

```

rainstared, 2019-07-17:
Selecting previously unselected package libexpat1-dev:arm64.
(Reading database ... 208507 files and directories currently installed.)
Preparing to unpack .../libexpat1-dev_2.2.6-1ubuntu0.19.04_arm64.deb ...
Unpacking libexpat1-dev:arm64 (2.2.6-1ubuntu0.19.04) ...
Setting up libexpat1-dev:arm64 (2.2.6-1ubuntu0.19.04) ...
ubuntu@ubuntu:/media/ubuntu/genode/genode$ ./tool/tool_chain  -e MAKE_JOBS=1 aarch64
tool/tool_chain:215: *** recipe commences before first target.  Stop.
ubuntu@ubuntu:/media/ubuntu/genode/genode$ nano ./tool/tool_chain
ubuntu@ubuntu:/media/ubuntu/genode/genode$ ./tool/tool_chain  -e MAKE_JOBS=1 aarch64
preparing gcc...
make[1]: Entering directory '/media/ubuntu/genode/genode'
make[2]: Entering directory '/media/ubuntu/genode/genode'
gcc  download ftp://ftp.fu-berlin.de/gnu/gcc/gcc-8.3.0/gcc-8.3.0.tar.xz

rainstared, 2019-07-18:
./tool/tool_chain -e MAKE_JOBS=1 aarch64 --debug=v V=s  2> toolchainerrors.log2

```
 Must remake target 'cc1'.
g++ -no-pie    -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o cc1 c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-fold.o c/gimple-parser.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-ubsan.o c-family/known-headers.o c-family/c-attribs.o c-family/c-warn.o c-family/c-spellcheck.o aarch64-c.o glibc-c.o \
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a   -L/media/ubuntu/genode/genode/build/tool_chain-19.05/lib-install/lib -L/media/ubuntu/genode/genode/build/tool_chain-19.05/lib-install/lib -L/media/ubuntu/genode/genode/build/tool_chain-19.05/lib-install/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz

```
**Reaping losing child**
lots of xmalloc not found.

For some reason, GCC as presently configured is not finding the libc headers on my system.

rainstared, 2019-07-18:
libtool: install: warning: remember to run `libtool --finish /media/ubuntu/genode/genode/build/tool_chain-19.05/bootstrap/install/libexec/gcc/aarch64-unknown-linux-gnu/8.3.0'

Configuring in ./gcc
configure: WARNING: decimal float is not supported for this target, ignored
configure: WARNING: fixed-point is not supported for this target, ignored
Using `/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/aarch64/aarch64.c' for machine-specific logic.
Using `/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/aarch64/aarch64.md' as machine description file.
Using the following target machine macro files:
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/aarch64/biarchlp64.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/aarch64/aarch64.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/dbxelf.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/elfos.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/gnu-user.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/linux.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/glibc-stdint.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/aarch64/aarch64-elf.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/aarch64/aarch64-linux.h
	/media/ubuntu/genode/genode/contrib/gcc-893d85767720fc2b3c695c56d737d2307b1f2229/src/noux-pkg/gcc/gcc/config/initfini-array.h
Using host-linux.o for host machine hooks.
Links are now set up to build a native compiler for aarch64-unknown-linux-gnu.
gcc_driver_version: 8.3.0
mkdir -p -- .deps
mkdir -p -- build/.deps
mkdir -p -- ada/.deps
mkdir -p -- brig/.deps
mkdir -p -- c/.deps
mkdir -p -- cp/.deps
mkdir -p -- fortran/.deps
mkdir -p -- go/.deps
mkdir -p -- jit/.deps
mkdir -p -- lto/.deps
mkdir -p -- objc/.deps
mkdir -p -- objcp/.deps
mkdir -p -- c-family/.deps
mkdir -p -- common/.deps
mkdir -p -- build-aarch64-unknown-linux-gnu/libiberty
Configuring in build-aarch64-unknown-linux-gnu/libiberty
mkdir -p -- build-aarch64-unknown-linux-gnu/fixincludes
Configuring in build-aarch64-unknown-linux-gnu/fixincludes
mkdir -p -- build-aarch64-unknown-linux-gnu/libcpp
Configuring in build-aarch64-unknown-linux-gnu/libcpp
mkdir -p -- .deps

rainstared, 2019-07-18:
preparing ali2dep...
Cloning into '/media/ubuntu/genode/genode/build/tool_chain-19.05/ali2dep'...
remote: Enumerating objects: 107, done.
remote: Counting objects: 100% (107/107), done.
remote: Compressing objects: 100% (73/73), done.
remote: Total 107 (delta 36), reused 100 (delta 30), pack-reused 0
Receiving objects: 100% (107/107), 230.02 KiB | 762.00 KiB/s, done.
Resolving deltas: 100% (36/36), done.
building ali2dep...
Compile
   [Ada]          main.adb
   [C]            dummy.c
   [C]            version.c
   [Ada]          ali2dep.adb
   [Ada]          ali.adb
   [Ada]          namet.adb
   [Ada]          types.adb
   [Ada]          butil.adb
   [Ada]          debug.adb
   [Ada]          fname.adb
   [Ada]          opt.adb
   [Ada]          osint.adb
   [Ada]          output.adb
   [Ada]          casing.adb
   [Ada]          gnatvsn.adb
   [Ada]          rident.ads
   [Ada]          table.adb
   [Ada]          tree_io.adb
   [Ada]          widechar.adb
   [Ada]          alloc.ads
   [Ada]          hostparm.ads
   [Ada]          fmap.adb
   [Ada]          sdefault.adb
   [Ada]          targparm.adb
   [Ada]          csets.adb
Bind
   [gprbind]      main.bexch
   [Ada]          main.ali
Link
   [archive]      libbuild.a
   [index]        libbuild.a
   [link]         main.adb
installing ali2dep...
installing tool chain to '/usr/local/genode/tool/19.05'...
ubuntu@ubuntu:/media/ubuntu/genode/genode$ 

**Successfully compiled and installed genode toolchain on raspberry pi 4 1gb**

rainstared, 2019-07-18:
As a futher comment i had to swap out all the config.guess files, there are like 5 of them, for copies of the one on my system.

closed 2019-07-18 by rainstared