genodelabs/genode #3445: hw: derive timer, pic, and cpu  from board.h

skalk, 2019-07-11:
Analogue to other SoC/Board relevant resources the concrete devices should be referenced explicitely by the board header instead of transparently putting them into the include pathes. Thereby, we lower the burden for developers that integrate new boards, and do not need new directory structures for each shared device, like generic ARM timer, gicv2, or gicv3 etc.
As a side effect, the semantic equivalent implementation of the generic timer for imx7 and rpi3 can be unified.
I favor this cleanup efforts before adding the next ARMv8 platform (i.MX8), because it will bring in the next PIC implementation and share the generic timer too.

m-stein, 2019-07-11:
@skalk It's really cool to see that you keep simplifying the kernel code. Sums up again to -200LoC. Thanks!

skalk, 2019-07-11:
@m-stein thanks for the kind words.

commit 27881c16 (2019-07-11): hw: sanitize arm trustzone/virtualization services

Those services are not SoC specific and have to reside at a generic place.

Fix #3445
commit 2b5d303d (2019-07-12): hw: sanitize arm trustzone/virtualization services

Those services are not SoC specific and have to reside at a generic place.

Fix #3445
commit fa1aa33f (2019-08-13): hw: sanitize arm trustzone/virtualization services

Those services are not SoC specific and have to reside at a generic place.

Fix #3445
closed 2019-08-21 by chelmuth