genodelabs/genode #3450: General system time concept

cnuke, 2019-07-14:
Briefly speaking, up to now there has been no notion of an overall concept of system time in Genode. Components that need to have access to some kind of real time are either configured locally, e.g. libc-based components access a configured “device” (*/dev/rtc*) which just might be an inline file system containing an artifical timestamp or the VFS `rtc` plugin, and other components query some `Rtc` session directly. Most of the time this is the `rtc_drv` on x86 machines, which is somewhat costly as reading the RTC via I/O ports takes time (no pun intended) and is therefor done scarcely; the libc will query a RTC source only once and uses this initial value to interpolate the current time.  However, for executing long-running components, it might be necessary to adjust the clock to componsate for any occurring clock drift or to correct a mis-configured clock in general. In addition, it would be nice to be able to use a remote time source, e.g., a NTP-server, to synchronize the system time.

To address this, @chelmuth and I came up with the following sketch:

[image]

It boils down to the following pieces:

* A proxy *Rtc* component in front of the `rtc_drv`, bundled up together in one `drivers-rtc-pc` package. The proxy component uses the driver to get the initial RTC value. It uses a *Timer* session (via the timeout framework) to locally interpolate the time. In contrast to querying the driver querying the proxy is fast. The runtime of the drivers package is set to request two *ROM* modules used to update the RTC value. The first one `system_set_rtc` is used to update the proxy component while the second one `hw_set_rtc` is used by the driver (#3438) component to write the value into the battery backed RTC.
* A different component, potentially accessing a remote time source, e.g. an SNTP client (#3448), can be used to update the proxy's RTC value. Furthermore, depending on the system policy, it could also be used to instruct the driver to set the RTC value.
* The `Rtc` session is extended with a *set*-signal that gets send whenever the RTC value is changed. Thereby clients will be notified and do not have to poll.
* The time backend of the libc is changed to instantiate a `Watch_handler` for the RTC device, which when triggered, will lead to re-reading the RTC value from the device.

Implementing this concept should yield, under normal operation, second to sub-second granularity updates for the overall system time, which should be enough.

A nice aspect of the concept is that we merely have to provide pre-configured `drivers-rtc-*` packages and it is up to the user to hook it up to the proper *ROM* module providers and those may be as complex as they need to be.

m-stein, 2019-07-22:
I had a short session on doing the propagation of rtc-set signals through vfs rtc up to the libc time back-end. My branch [3450_General_system_time_concept](https://github.com/m-stein/genode/commits/3450_General_system_time_concept) is based on the branch of @cnuke. It incorporates the SNTP client, provides a small test that repeatedly prints the libc time and has a prototype for the watch function in the rtc file system. The main thing that's missing now is the handling of VFS watch handlers in the libc. There is an unmerged example approach to this in [cnuke/2018-11-28-cmus](https://github.com/cnuke/genode/commits/issue%233033-2018-11-28-cmus) but it can't be directly applied anymore to the current state of VFS and libc. Until 5th of August, I wont be able to continue with this.

cnuke, 2019-08-12:
* 4cb66c9 extends the `Rtc` VFS plugin to notify the application whenever the *set-signal* was received
* 93f8e72 introduces an interface to request a `Vfs_watch_handle` for a path in the VFS
* 36bc691 merges the `Rtc` backend into the time backend
* 876f351 will use the `Vfs_watch_handle` to re-read the `Rtc` value from the configured file

With these commits in place the desired functionality is in place.

chelmuth, 2019-08-14:
Thanks @cnuke and @m-stein, I almost finished my review of the patches, which was a smooth ride. I discovered that we still miss the drivers_rtc-pc package recipe that should be added and used by the test script before the release too.

m-stein, 2019-08-14:
@chelmuth @cnuke I would do the drivers_rtc-pc package recipe this Friday if that's fine with you.

cnuke, 2019-08-15:
Thanks for the pkg @m-stein.

(For future reference, since we “define” the the `Rtc::Timestamp` as *UTC*, i.e. leap seconds are included, we have to substract them in cases where these are not anticipated.)

m-stein, 2019-08-16:
This is the package but I didn't actually know how to test it: 
ee063c9087 gems: recipes/pkg/drivers_rtc-pc

Any hints how I may test it?

cnuke, 2019-08-19:
As a first step you could use an artificial test environment, e.g. `dynamic_rom`, that mimicks the *system-policy* bubble depicted in the image above (basically just like system-rtc test run script).

Eventually it would be nice to have the whole mechanism integrated (use of the SNTP client for setting the `system_rtc` and maybe synchronizing the HW RTC if the time differs too much).

chelmuth, 2019-08-19:
Also, the package should be used by the test script IMO.

ssumpf, 2019-08-20:
@chelmuth: b7829f4 adds packages to system_rtc.run

commit 02063542 (2019-08-14): Configurable set_rtc values in RTC test

Fixes #3450
closed 2019-08-21 by chelmuth