genodelabs/genode #3451: [audio_drv] needs two tweaks for Asus ..etc support

ttcoder, 2019-07-15:
Hello Genode team,

I'm trying to get audio going on my Asus F2A55M mobo based tower -- it works in Haiku, but (not yet) on Genode. After an afternoon of work I'm very close to it.

The last (latest?) problem is that I get the error from dde_bsd/src/lib/audio/mem.cc:259 . Presumably because this computer provides lots of audio jacks, requiring more heap memory to store the labels and such. So I naively increased the slab size to the next power of two (17 instead of 16) but then I get a "quota guard" exception instead of the Bsd-Slab exception. What's the proper way to increase the slab size? And then let's commit that change to the next iteration of Genode..

The first problem I "solved" is, admittely, more than a mere tweak: src/lib/audio/bus.cc:186 has a special case if() test for machines with HDA devices on two different PCI slots. It so happens that I need a similar test, so I hacked it, to use the analog HDA on my motherboard (the first of the two is HDMI, no go). I suspect there are quite a few more cases like this though -- so maybe a proper fix is not to pile up if/then/else cases, but instead to probe each PCI candidate, and bail out and and try the next one if a slot is not a valid candidate (that's what the haiku driver does FWIW).

This code will go in production in a few months AFAIC by the way, but no pressure ;-)

cnuke, 2019-07-16:
> The last (latest?) problem is that I get the error from dde_bsd/src/lib/audio/mem.cc:259 .

I would appreciate if you could produce a LOG where the actual size the driver wants to allocate is shown (and if time permits a backtrace from where the allocation happens, sprinkling `__builtin_return_address(0)` + *no-omit-fp* around should be enough). 

(Most of the memory management stuff is copied over from the dde_linux backend where we do not use a slab whenever the allocation size exceeds a certain threshold. If that's necessary now I would rather implemented a *large_alloc* interface here as well.)

> Presumably because this computer provides lots of audio jacks, requiring more heap memory to store the labels and such. So I naively increased the slab size to the next power of two (17 instead of [...]

You increased the slab size alright - the way the whole memory allocation mechanism is built is somewhat intertwined. The bottom line is, you have to increase the `BLOCK_SIZE` in the *Slab_backend_alloc* to *2MiB* as well as the *Range_allocator* will bail otherwise (this leads to the exception you see). That's because the *Slab_alloc* constructor wants to create a slab block, that's **16** times the slab size (*128KiB*) - the number of slab entries is calculate by dividing the slab block size by the slab entry size (see `_calculate_block_size()`).

(All in all the memory management in the various DDEs could be made simpler but that's another story.)

> The first problem I "solved" is, admittely, more than a mere tweak: src/lib/audio/bus.cc:186 has a special case if() test for machines with HDA devices on two different PCI slots. It so happens that [...]

You are right, a blacklist will not work long-term, even the one that is currently there should be removed at some point.

In general drivers should be structured in a way that they only try to handle the one(!) device they were given by the platform driver and should not enumerature the PCI bus on their own. That, however, requires somewhat more logic on the outside, e.g. a device manager that knows which driver is able to drive which device. For convience reasons (or rather to cover more than one configuration used by our nightly CI), the `platform_drv` config used by most run-scripts (see *repos/base/run/platform_drv.inc*) just contains a device class alias (see *repos/os/src/drivers/platform/spec/x86/README*) and will hand out all devices belonging to one class to a driver.

In your case, where the platform is fixed, it makes sense to create a specific `platform_drv` config that will hand-out only the right audio device, e.g.:

```
<policy label="audio_drv"> <pci bus="0" device="5" function="0"/> <policy>
```

ttcoder, 2019-07-17:
file:///boot/home/Desktop/Asus/DSC05885.JPG genode audio on Asus

ttcoder, 2019-07-17:
Ok so file uploads in Web+ are ignored, and in Otter they just result in inserting the file path *g*

So I uploaded the screenshot here instead: https://tts-genode.neocities.org/DSC05885_genode_audio_on_Asus.jpg

Not mentionned on the screenshot is the return address, which I added in a further run:

" > return address: 0x1034c6d"

I understood what that one was about because I've seen it before, exploring the Genode source. (how does one use that address BTW? I see passing ld_verbose=true to config will log the address segments of each library and add-on, but the function and methods inside are not shown)

As to your second suggestion, I'm going to need some help: is that a GCC option ?

***

Driver status: Thanks much, it works with the increase to 2MB. It now says "BSD driver enable playback".

Can't do a full scale test right now as the test scenario is set for AHCI, and it seems to not work on this AMD system (the HDD led remains solid lit-up). Probably due to AMD support issues as mentionned on the Genode HCL. I'll setup a test with a ROM mp3 instead.

ttcoder, 2019-07-17:
Attempting to play an mp3, the Asus system remains silent, with no particular audio_drv message to LOG (note to self: the very same usb stick tests okay on a thinkpad, and the Asus system tests ok with haiku)

So this will take a bit more work; I've had a bit of experience debugging the (haiku) HDA driver in the past, we'll see.

cnuke, 2019-07-19:
Thanks for reporting back.

> Not mentionned on the screenshot is the return address, which I added in a further run:
> " > return address: 0x1034c6d"

> I understood what that one was about because I've seen it before, exploring the Genode source. (how does one use that address BTW? I see passing ld_verbose=true to config will log the address segments of each library and add-on, but the function and methods inside are not shown)

Normally you would use the information from the dynamic linker to calculate the proper offset in a shared-object, i.e. library. Doing that manually would involve using *objdump*, e.g.,

`objdump -dlC <build-dir>/debug/audio_drv`

for every address in your backtrace. Using the contents of the *debug* directory is vital as the *bin* directory only contains stripped binaries.

There is also `Genode::backtrace()` (see *repos/os/include/<spec>/os/backtrace.h*), which will traverse the stack on its own.

>As to your second suggestion, I'm going to need some help: is that a GCC option?

Right, normally GCC will omit storing the stack frame pointer. Depending on how you debug, e.g. backtrace in GDB, it might be helpful to have access to this information. Using `-fno-omit-frame-pointer` will instruct the compiler to always store it. It can be set either just for one component by adding

`CC_OPT += -fno-omit-frame-pointer` to the *target.mk* or for all components by adding it to *<build-dir>/etc/tools.conf*.

> Driver status: Thanks much, it works with the increase to 2MB. It now says "BSD driver enable playback".

> Attempting to play an mp3, the Asus system remains silent, with no particular audio_drv message to LOG (note to self: the very same usb stick tests okay on a thinkpad, and the Asus system tests ok with haiku)

You might need to configure the widgets. The driver is a port of OpenBSD 5.9's HDA driver and supports using the same mixerctl knobs (see 47990f37d7f6267). (On that account I would check if the soundcard works properly with vanilla OpenBSD.)

Since you mentioned running on on AMD machine you could also check if the interrupts work for the soundcard (instrumenting `notify_play()` should do the trick).

ttcoder, 2019-07-22:
This is indeed the problem! The interrupt handler is not called on that AMD tower.

(whereas the same USB stick used on a thinkpad produces a flood of tracing)

Just in case, I tried force execution of the block starting with this heading:
/* disable MSI, use INTx instead */
(removing the if() condition, otherwise it is not called) but that makes no difference.

Not sure where to go from here ?

***

Re the backtrace, I've added my findings as a hints-and-tips example over here because it makes a good case study for people new to Genode like me: http://chiselapp.com/user/ttcoder/repository/genode-book/wiki?name=Book:Tips (see "backtrace" section)
Tell me if something else could be useful.

cnuke, 2019-07-24:
> This is indeed the problem! The interrupt handler is not called on that AMD tower.
> (whereas the same USB stick used on a thinkpad produces a flood of tracing)

> Just in case, I tried to force execution of the block starting with this heading:
> /* disable MSI, use INTx instead */
> (removing the if() condition, otherwise it is not called) but that makes no difference. (Edit: whereas on the old thinkpad, sound works regardless, either with or without MSI)

> Not sure where to go from here ?

Obviously altering the code might change how the driver will configure the device.  That being said, whether MSIs or GSI are used is up to the `platform_drv` and using MSIs is the default. You could try if setting the `irqmode="nomsi"` attribute in the platform driver's policy for the device (for more information please look into the README) makes a difference.

> (Edit) I heard there is an easy-to-install BSD that can be installed on a USB stick, but IIRC it's FreeBSD rather than OpenBSD. My best bet might be to examine the haiku HDA driver, seeing how differently it sets up interrupts compared to the BSD driver and the Genode dde_bsd environment. Will circle back to this in a few days possibly.

There are also some OpenBSD live systems (Fugulta comes to mind) but I never have used them myself.

> Re the backtrace, I've added my findings as a hints-and-tips example over here because it makes a good case study for people new to Genode like me: http://chiselapp.com/user/ttcoder/repository/genode-book/wiki?name=Book:Tips (see "backtrace" section)
> Tell me if something else could be useful.

That's nice, documenting those things is quite helpful. So far, I have no additions to make.

ttcoder, 2019-10-25:
Update: audio does *not* work in OpenBSD 5.5 indeed... So on a whim I tried a spare audio card and it worked (in BSD and Genode) -- we're using an Ensoniq card for now to start testing. Details follow:

Ensoniq: using a SoundBlaster PCI 128 I had lying around. Need to blacklist (in BIOS or in driver) the on-board HDA and the PCI 128 gets recognized without a problem. Works okay (some songs randomly start playing with a tshh-tsh-tshh sound lasting until the song's end), good enough for testing and stable so far.

Curious that this one has interrupts served correctly.. I know little on that topic, probably interrupts are setup differently for a PCI device and an on-board device..

HDA with Open BSD:
- looked into Fugu Ita, but it requires burning a CD to use for writing an USB stick
- then found out about http://liveusb-openbsd.sourceforge.net/ , which works well

cnuke, 2019-10-28:
To be honest I am mildly amused that a ES1370 card came to your rescue. After all, I only enabled the driver because it is one of the simpler ones and more to the point is provided by Qemu. I'm glad that you found a way to keep going further, though.

> Update: audio does not work in OpenBSD 5.5 indeed...

Just to clarify - did you actually use version *5.5* or is that a typo and it is ***6**.5*? (At some point I would like to update the driver to *6.6* as there are some changes to the driver stack I would like to pull in but I don't expect the update to fix your problem, especially if the card does not even work with a recent vanilla OpenBSD version.)

Unfortunately I did not get notified about your edits in the older comment and saw them only just now - please don't hesitate to create new comments instead of editing previous ones.

> kernel: we're using the NOVA kernel; I don't suppose there is another x86 kernel we could use that would impact IRQs usage ?

You could try one of the old 32 bit only kernels that do not use MSI or rather ACPI in general. In any case, it would be interesting to see the differences in the IRQ routing setup between Genode, Haiku and OpenBSD.

ttcoder, 2019-11-06:
Ensoniq cards are a "blast from the past" for me too! When we started (BeOS, 1999) we were selling systems with SB PCI 128 and SB Live! cards.

Confirming, when OpenBSD boots up it displays "5.5" version. The http://liveusb-openbsd.sourceforge.net/ possibly hasn't been updated in a long time. This Asus F2A55 motherboard was purchased 10+ years ago, so I doubt a new version of OpenBSD would support its interrupts/HDA model all of a sudden indeed.

This week I'm running around like a headless chicken, extinguishing fires in my code base /*g*/.. I expect this to calm down at some point and I'll attend to this again (and to my AHCI troubles on the same machine).

ttcoder, 2020-02-10:
Note to self : check out (maybe even back-porting) the great set of changes here ASAP : https://github.com/genodelabs/genode/commit/459afa7524eef7ae99a6530a69aad402c51fad60

Looks like my chiselapp comments have been useful, I see changes in printf() and friends :-) and SLAB_STOP_LOG2 seems to have changed too. I'm most curious about the AMD related interrupts tweaks though. It will probably take more than that to fix interrupts not being served, but who knows ?

nfeske, 2021-02-26:
@ttcoder May we close this (stale) issue?

ttcoder, 2021-02-26:
Indeed, Genode 20.11 might support my hardware, and if not I'll re-open (or open a new ticket).

closed 2021-02-26 by ttcoder