genodelabs/genode #3460: Delete control character isn't correct set!

cenapatop, 2019-07-20:
In /config/special.chargen the KEYDELETE must be replaced with asci="127"

<key name=“KEYDELETE" ascii="127"/>

chelmuth, 2019-07-20:
You're right, this change qualifies as work-around for the current terminal-bash combination. But DELETE as the other special keys (e.g., HOME) are destined to be handled explicitly by the Input clients as they do not produce real printable characters.

cenapatop, 2019-07-20:
Any idea how to fix it?

asci="127" doesn't works correct. It deletes on the right, and not on the left.

chelmuth, 2019-07-28:
@cenapatop sorry for the silence but I'm currently AFK most of the time. If you like KEY_DELETE to act like KEY_BACKSPACE you may just use `ascii="8"`.

chelmuth, 2019-09-02:
Can we close this issue?

cenapatop, 2019-09-03:
I think yes. But I'am not sure if this fix really works for every keyboard type!

chelmuth, 2019-09-03:
According to my recent experience with common PC keyboards this solution is quite sufficient (i.e. all keyboards had the <kbd>Del</kbd> key at the same physical position with is `KEY_DELETE`).

closed 2019-09-03 by chelmuth