genodelabs/genode #3462: Sculpt: captive portal VM

m-stein, 2019-07-22:
Provide a disposable VM that brings up a Firefox for a captive portal page of a selected Wifi. The VM image should be as light-weight as possible, have a boot time as short as possible, and should be resizable. It should be well documented how to reproduce the VM image.

m-stein, 2019-07-22:
Our current approach is based on latest Tinycore with Xorg server and i3 WM. So far, we managed to start Firefox in fullscreen without title, borders, or WM panel:
```
# download tinycore iso, vbox vm with net, 1G ram, 8G hdd, and iso image
# start vm, select "core (command line only)"
# follow tinycorelinux.net/install_manual.html; instead of the GUI app "Apps"
  use 'tce' command (search via 's' for extension name, select result, quit
  from description via 'q', install via 'i'); use ext3 as suggested, not ext4;
  instead of the suggested content write the following to the menu.lst:

! default 0
! timeout 0
! hiddenmenu
! title tinycore
! kernel /boot/vmlinuz quiet restore=sda1/tce
! initrd /boot/core.gz

# 'halt', remove tinycore iso from vm, start vm (no grub menu, shell appears)
# install xorg, i3, firefox via tce
# 'reboot' (i3 welcome message should appear)
# right-ctl + f1 to switch to shell
# stop xorg server: 'pkill Xorg'

# replace content of ~/.i3/config with this:
! font -misc-fixed-medium-r-normal--13-120-75-75-C-70-iso10646-1
! workspace_bar no
! new_window bb
! exec firefox

# backup configuration to mydata.tgz
! .filetool.sh -b

# 'reboot' (firefox should start without bottom bar, borders, or title bar)
```
The i3 config was written using ``docs/userguide`` in https://i3wm.org/downloads/i3-3.e-bf3.tar.bz2 because it's a pretty old version. The next step is to build and install Vbox guest additions and make resizing work.

m-stein, 2019-07-22:
Updated the above documentation to also consider /opt/.filetool.lst.

m-stein, 2019-07-25:
We re-worked the above tutorial. It now contains all steps for creating the VM with tinycore (the documentation on how to integrate the VM in Genode follows soon):
```
set up tinycore
===============

set up a vm
-----------

* go to host shell
! sudo apt install virtualbox # we used virtualbox version 5.1.38
! wget https://distro.ibiblio.org/tinycorelinux/10.x/x86/release/Core-10.1.iso
! wget https://download.virtualbox.org/virtualbox/6.0.4/VBoxGuestAdditions_6.0.4.iso
! virtualbox &

* go to virtualbox window
* create new vm with network
  * type: Linux
  * version: Other Linux (64-bit)
  * ram: 1G
  * hdd: create new, 1G, vdi, dynamic

* start vm
* select 'Core-10.1.iso' as initial disk of vm

prepare hard disk
-----------------

* go to vm shell (the following steps were created using
  http://tinycorelinux.net/install_manual.html)
! [Enter]                                                                      # boot to user shell
! tce-load -wi cfdisk grub-0.97-splash                                         # download extensions cfdisk and grub
! sudo su                                                                      # open root shell
! cfdisk /dev/sda                                                              # create partition on hard disk
  ! New
    ! Primary
    ! [Enter]
  ! Bootable
  ! Write
    ! yes
  ! Quit

! mkfs.ext3 /dev/sda1                                                          # format partition
! rebuildfstab                                                                 # re-create file system table
! mount /mnt/sda1                                                              # mount hard disk

install grub
------------

! mount /mnt/sr0                                                               # mount disk drive
! mkdir -p /mnt/sda1/boot/grub /mnt/sda1/tce
! cp -p /mnt/sr0/boot/* /mnt/sda1/boot/                                        # install kernel, initrd
! cp -p /usr/lib/grub/i386-pc/* /mnt/sda1/boot/grub/                           # install grub
! touch /mnt/sda1/tce/mydata.tgz                                               # file for tinycore backup/restore
! vi /mnt/sda1/boot/grub/menu.lst                                              # create grub menu config
  * insert mode: 'i'
  * insert this
    ! default 0
    ! timeout 0
    ! hiddenmenu
    ! title tinycore
    ! kernel /boot/vmlinuz quiet restore=sda1/tce
    ! initrd /boot/core.gz
  * write, quit: '[Esc]:wq'

! grub
  ! root (hd0,0)
  ! setup (hd0)
  ! quit

! halt
[... 158 lines cut ...]
! tar -xf /tmp/vboxga604.tar.gz
! cd /tmp
! mksquashfs vboxga604/ ~/vboxga604.tcz
! cd ~
! md5sum vboxga604.tcz > vboxga604.tcz.md5.txt

install guest additions extension
---------------------------------

* go to virtualbox window
* start first vm

* go to window of first vm
! [Right-Control]+[F1]                                                         # switch to shell
! pkill Xorg                                                                   # stop xorg, i3, firefox
! scp <HOST_USER>@<HOST_IP>:/tmp/vboxga604.tcz* \                              # send extension from host to vm
  /mnt/sda1/tce/optional/
! rm -f /mnt/sda1/tce/optional/openssh*                                        # uninstall openssh
! vi /mnt/sda1/tce/onboot.lst                                                  # install guest additions extension
  * insert mode: 'i'
  * replace ''openssh.tcz'' with 'vboxga604.tcz'
  * write quit: '[Esc]:wq'
! sudo reboot
* check whether vm is resizable
```

nfeske, 2019-10-10:
Wouldn't that content make a splendid Genodians article?

m-stein, 2019-10-16:
You're right and this was also my plan ^^ but I found that it's still difficult for me to regularly prioritize article writing over other work. I'll give it another try.

jschlatow, 2021-06-30:
I'd like to share a few additions to the instructions from @m-stein posted above...

First of all, I recognised that the latest guest additions of virtualbox 5.2, i.e. 5.2.44, smoothly cooperates with the host version 5.2.38. The benefit of using 5.2.44 is that it supports the linux lts kernel 5.4 and therefore allows switching to the more recent TinyCore (TC) 11.1.

In order to get all parts of the guest additions running, I need three kernel modules: vboxguest, vboxvideo, vboxsf.
vboxguest is already shipped with TC 11.1. The vboxvideo module is contained in graphics-5.4.3-tinycore.tcz. I haven't found vboxsf in any package though.

I therefore compiled guest additions 5.2.44 on TC 11.1 to build vboxsf and install the VBoxClient binary (required for shared clipboard support and automatic display resize). Note, that the modules vboxvideo, vboxguest and vboxsf will be installed in `/lib/modules/tinycore-5.4.3/misc`, yet, the existing vboxvideo and vboxguest module take precedence. Running `VBoxClient --display` and `VBoxClient --clipboard` with the vboxguest module from TC, however, failed. I thus removed the vboxguest module from core.gz, which solved this issue.

I also noticed that the display resizing does not quite work with the newly built vboxvideo module but with the module from graphics-5.4.3-tinycore.tcz.

jschlatow, 2021-06-30:
The downside of using TC 11.1 though is that it does not come with a working i3 package. There was some discussion on the TC forum about that. I managed to compile i3 4.7.2 for TC 11.1. For compiling i3, you need to tce-load yajl-dev, libev-dev, pango-dev, pcre-dev and libstartup-dev. i3 4.7.2 is the latest version that still compiles with yajl 1.x. Newer versions required yajl >= 2.0, which is not available in TC 11.1.

m-stein, 2022-09-01:
I finally published a corresponding Genodians article and therefore close this issue.

closed 2022-09-01 by m-stein

jschlatow, 2022-09-02:
Thanks for the documentation. I'd like to add that I built some tooling for remastering TinyCore images at https://github.com/jschlatow/tinycore-remaster.

m-stein, 2022-09-05:
@jschlatow Thanks for the hint! I added a paragraph to the article that references your additional resources :)