genodelabs/genode #3466: base-linux on ARM cross compillation

trimpim, 2019-08-07:
I'm trying to run a Genode run script (run/log for testing) using base-linux on arm_v7a.

As the Linux on the target has no compiler, and building a tool-chain for arm 32 bit is according issue #3444 not an easy task, I try to use cross compilation.

With some minor changes https://github.com/trimpim/genode/commit/b7c647be1bd7a9b8e33d983f9927d325c7fab25c, cross compilation for base-linux on arm_v7a worked.

After copying the resulting `var/log/genode` folder to to the target, I was able to successfully start core. This produced the following output:

```
Genode sculpt-19.07-2-gbd439408b
4095 MiB RAM and 8999 caps assigned to init
Warning: blocking canceled in entrypoint constructor
```

No further output was printed after this, but core was still reported as running with 4 threads by `htop`.

The warning is printed when core receives a signal during an IPC (SIGCHLD in this case, which indicates the exit of a child).

While debugging I noticed, that the `init` component, that core starts, exits with an unhandled page fault in `ld.lib.so`.

This page fault happens, when in ctr0.s the first value is written to the stack (for debug code  see https://github.com/trimpim/genode/commit/bd439408bb4c3631a0cac3ae6c8f523367bb7907).

The address to which ld.lib.so writes comes from the ELF file:

`/usr/local/genode/tool/current/bin/genode-arm-readelf -a build/linux_arm/var/libcache/ld-linux/ld-linux.lib.so | grep __initial_sp`

```
6857: 000f7610     0 NOTYPE  LOCAL  DEFAULT   15 __initial_sp
```

If my calculations are correct, this address lies inside the BSS section of ld.lib.so:

`/usr/local/genode/tool/current/bin/genode-arm-readelf -S build/linux_arm/var/libcache/ld-linux/ld-linux.lib.so`

```
...
[15] .bss  NOBITS  00093570 094564 066994 00  WA  0  0 16
...
```

For core that is built using the same crt0.s file, only the addresses are different.

trimpim, 2019-08-07:
@ssumpf Thanks for your direct reply to my second patch. I have applied the changes of your commit in my tree.

Unfortunately with this change, my cross compiled `base-linux` core crashes now with:

 `core: unhandled page fault (11) at 0x0100ddf0, code 0x81f`

Without the changes `core` started, but `init` crashed with a similar error.

By moving a "forced" crash to different locations, I isolated, that the instruction `str sp, [r4]` causes the page fault.

If I run `run/log` on `base-hw` (with this changes) the run script completes successful.

chelmuth, 2019-08-12:
If I get @ssumpf's patch right, the symbols are now used GOT relative and I ask myself why `0x0100ddf0` results in SIGSEGV (11). The address should be well inside the static binary text segment. Do you know more about the *code 0x81f* part of the log message - this is dmesg, right?

chelmuth, 2019-08-12:
The first Internet search hit was https://linux-arm-kernel.infradead.narkive.com/pHtV2CRi/understanding-an-unhandled-page-fault#post23, which strongly hints this is a write fault... text relocation in action?

trimpim, 2019-08-12:
@chelmuth and @ssumpf , yes my output is from `dmesg`.

ssumpf, 2019-08-12:
@trimpim: Can you post the output of `scanelf -qT ld.li.so` please.

trimpim, 2019-08-12:
@ssumpf  `scanelf -qT ld-linux.lib.so` prints the following output:

```
scanelf: scanelf_file_textrels(): ELF is missing relocation information
scanelf: scanelf_file_textrels(): ELF ld-linux.lib.so has TEXTREL markings but doesnt appear to have any real TEXTREL's !?
  ld-linux.lib.so
```

ssumpf, 2019-08-14:
1da0453 should fix the startup code issue, 21955d9 becomes obsolete.

ssumpf, 2019-08-15:
@chelmuth: 1da0453 was positively tested by @trimpim and may be merged to staging.

ssumpf, 2019-08-20:
@chelmuth: b933c56 fixes the startup code handling on arm_v7

ssumpf, 2019-08-20:
@chelmuth: c452b33 fixes ARM again.

rite, 2019-11-25:
26eeb4e allows to build a ARM toolchain supporting the hard-float ABI (armhf). Additionally it allows to "inject" a custom host toolchain.

rite, 2019-11-25:
repos/base/mk/spec/armhf_v7a.mk is missing in 26eeb4e. Here's the fixup:
- 072e31d9754b0c9a8cfa38399ab2c669cb2f50f4

chelmuth, 2019-11-25:
@rite I plan to look deeper into your work after the release. I expect some rebase effort to align with changes in #2829

rite, 2019-12-03:
Building `lx_fs` on a different development PC showed that another patch is required:
- b08ea64

chelmuth, 2019-12-04:
I'm afraid this merge will have to wait until #2829 made it through staging.

chelmuth, 2019-12-12:
I'm still not certain the time has come to merge your changes entirely. Therefore, I prepared https://github.com/chelmuth/genode/commits/issue3466 to merge all changes not directly introducing *armhf*. Will it be okay for you to maintain the remainder as a patch on your own for now?

rite, 2019-12-12:
@chelmuth Sure, that's totally fine with me.

chelmuth, 2019-12-20:
Okay, then let's close this issue for now.

cproc, 2021-05-12:
Should we reopen this issue for the 21.05 tool chain update? The road map mentions switching to hard-float on ARM and this issue looks like it covers this topic.

nfeske, 2021-05-12:
@cproc Good idea.

cproc, 2021-05-20:
Fixup commit b55bce9 is needed to execute run scripts on foc/armhf_v7a, but unfortunately the platform seems to reboot:

```
L4 Bootstrapper
  Build: #2 Thu May 20 15:25:07 CEST 2021, 10.3.0
  Scanning up to 256 MB RAM, starting at offset 32MB
  Memory size is 256MB (70000000 - 7fffffff)

L4 Bootstrapper
  Build: #2 Thu May 20 15:25:07 CEST 2021, 10.3.0
Error: platform rebooted unexpectedly
```

cproc, 2021-05-21:
base-linux/arm(hf)_v7a does not work for me on Ubuntu 20.04 armhf on Raspberry Pi 4 with commit fdb1a4d ("base-linux: avoid legacy syscalls").

With `log.run` only these messages appear:

```
spawn ./core
Genode sculpt-21.03-172-g5f8eaa7289
4095 MiB RAM and 8999 caps assigned to init
```

@rite, @trimpim: does the current Genode staging branch still work on your Linux target?

nfeske, 2021-05-21:
@cproc I think it comes down to re-building the seccomp policy (issue `make` in tool/seccomp`) and copying the resulting `seccomp_bpf_policy_arm_32.bin` file to `base-linux/src/lib/seccomp/spec/arm/seccomp_bpf_policy.bin`. I haven't touched the bin file in https://github.com/genodelabs/genode/commit/fdb1a4dd886a25eb6ca792220182fad489a4b8cd to avoid the wrong impression of me having tested Linux on arm_32.

cproc, 2021-05-23:
@nfeske: yes, that solved the problem. Commit 0364b05 updates the seccomp policy.

cproc, 2021-05-25:
The build errors with `KERNEL=hw` got resolved by configuring the armhf tool chain and the armhf_v7a platform  with `vfpv3` instead of `vfp`. I updated the commit accordingly. foc still reboots and we might not be able to fix this problem before the release, but at least the armhf tool chain could now already be useful for autopilot-testing hard-float with base-hw in addition to its original use-case for armhf Linux.

nfeske, 2021-05-25:
Given the trouble of armhf, I think we should better back away from this change. My reasoning is as follows:

* When we discussed armhf earlier, we came to the conclusion that we don't want to introduce another option besides arm_v7a but instead change our notion of arm_v7a to use hardfloat. Otherwise, the testing effort would go out of hand. Now it looks like the switch to armhf is much more risky than I anticipated (e.g., looking at the breakage of base-foc). So I feel that we should not pull through such a risky change a few days before the release.

* With the focus shifting away from arm v7 to arm v8, the switch to armhf does not appear as crucial to me, compared to one year ago. As far as I can see, only arm v7 is effected by this change.

* If armhf is desired for a specific use case, it is possible by the means of building a custom tool chain.

@cproc how do you think about it?

cproc, 2021-05-25:
Building Fiasco.OC with an armhf tool chain seems to be possible in general, as the default configured tool chain prefix is `arm-linux-gnueabihf-`, but I've also seen a `-mfloat-abi=soft` flag in a `Makeconf.arm` file and haven't checked yet if it gets applied or maybe overridden when building in the Genode build system. So, it might be an easy fix in the end, but could also need some time to figure out what exactly needs to be done. Given the little time left until the release, I agree that a full switch could be too risky at this time.

nfeske, 2022-05-25:
There is currently no plan for further work on this issue. In principle, we could consider armhf for the next tool chain update in 2023. But since 32-bit ARM becomes less important, there is no real incentive.

@trimpim would you agree to close this issue?

trimpim, 2022-05-27:
@nfeske I agree on closing this issue.

closed 2022-05-27 by trimpim